ros-navigation/navigation2 #3117: Robot randomly jumps around the map

wojciechmadry, 2022-08-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 Focal
- ROS2 Version:
  - Galactic
- Version or commit hash:
  - ros-galactic-navigation2	1.0.12-1focal.20220730.081413
- DDS implementation:
  - FastDDS

#### Steps to reproduce the issue
```
1. ros2 launch turtlebot3_bringup robot.launch.py use_sim_time:=False # Launch robot (Raspberry PI)
2. ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=False # Launch Navigation2 (PC)
3. ros2 launch slam_toolbox online_async_launch.py use_sim_time:=False # Launch SLAM (PC)
4. Start mapping the world
```

#### Expected behavior
Map the entire map without random jumps

#### Actual behavior
Robot doesn't jump if we use slam to scan some room and after that run navigation2 (We tried a few times):

[Mapped - streamable](https://streamable.com/lv7ni2)

Robot randomly jumps when we start navigation2 and slam_toolbox on the beginning and we try to map the room:

[Unmapped - streamable](https://streamable.com/yijo6a)

#### Additional information
We use physical Turtlebot3 (Jumps don't appear in simulation)
We checked multiple configurations like navigation and slam on Turtlebot or running parts of the nodes on PC (same network), and there were no differences.
We tried both navigation2 and turtlebot3_navigation2 and the behavior was the same.

All config files and launchers are below:

Navigation2 launcher:
```python
# Copyright 2019 Open Source Robotics Foundation, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# Author: Darby Lim

import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node

TURTLEBOT3_MODEL = os.environ['TURTLEBOT3_MODEL']

[... 644 lines cut ...]
      profile_acceleration_constant: 214.577

      # [rev/min2]
      # ref) http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#profile-acceleration
      profile_acceleration: 0.0

    sensors:
      bumper_1: false
      bumper_2: false

      illumination: false

      ir: false

      sonar: false

diff_drive_controller:
  ros__parameters:

    odometry:
      publish_tf: true
      use_imu: true
      frame_id: "odom"
      child_frame_id: "base_footprint"
```

arslogavitabrevis, 2022-08-17:
Both AMCL and slam_toolbox are giving the position inside the map (map frame to odom frame). When you start the navigation with the slam toolbox you have two node giving concurrently giving this position. I think this may explain the jump you see in the simulation. You should try to disable AMCL while mapping with custom launch file.

SteveMacenski, 2022-08-17:
@arslogavitabrevis  is correct

closed 2022-08-17 by SteveMacenski

wojciechmadry, 2022-08-19:
Hey @SteveMacenski, @arslogavitabrevis, 
Thanks a lot for the answers!
You are right that if AMCL works simultaneously with SLAM, the robot randomly jumps, but in our case, AMCL is not started.
This video present how it works when we enable AMCL:
[ Slam and AMCL working together](https://streamable.com/6oi6s8)
(At the end of this video there is the biggest jump, this is our problem)

We tested it with, firstly running only slam to map the office with slam_toolbox and then we turned off slam and loaded .yaml map to navigation stack:
[Navigation2 Jump without SLAM -> First example](https://streamable.com/xoijn3)
[Navigation2 Jump without SLAM ->  Second example](https://streamable.com/4h2fps)

In the case above I load the map to navigation using these commands:
```
ros2 launch turtlebot3_bringup robot.launch.py  use_sim_time:=False // This is run on turtlebot3
ros2 launch nav2_bringup bringup_launch.py use_sim_time:=False autostart:=True map:=path/map.yaml // This is run on PC in the same network
```

We also follow [nav2 with slam](https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html) and the result was the same:
[image]

If you did not encounter a similar problem maybe you could lead us, where start to searching for a solution.

arslogavitabrevis, 2022-08-22:
You launch the navigation including this file:
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([nav2_launch_file_dir, '/bringup_launch.py']),
            launch_arguments={
                'map': map_dir,
                'use_sim_time': use_sim_time,
                'params_file': param_dir}.items(),
        ),

The bringup_launch.py include the localisation_launch.py that launch the AMCL node. 

If you want you can do a 'ros2 node list' to check if they are both lauch. You may also get some interesting informations from the command 'ros2 topic info /tf -v' if you check for publisher.

akraus53, 2024-10-07:
Hi all, I'm having a similar problem in Gazebo with the simulated Turtlebot. Can it be that the 0xFDFDFDFD issue is the reason in my case or does this only apply to the real Turtlebot?

wojciechmadry, 2024-10-07:
Hi @akraus53, 
I think you have different problem.
Case with 0xfdfd... was strictly related with physical robot and incorrect reading from the driver. This problem was not occurs in simulation.
Maybe you have problem as Steve mention with simultaneously AMCL's running in slam and navigation - this can cause very similar problems