ros-navigation/navigation2 #3119: Replace config use_sim_time with SetUseSimTime in launch file

tonynajjar, 2022-08-18:
Small enhancement: you could replace all the use_sim_time in the config with the [SetUseSimTime](https://github.com/ros2/launch_ros/blob/humble/launch_ros/launch_ros/actions/set_use_sim_time.py) introduced in Humble. That would remove the RewrittenYaml logic from the launch file

One thing to note is that bt_navigator_navigate_through_poses_rclcpp_node won't get this parameter because it's created internally. Here is a workaround in bt_action_server_impl.h: 

```
  auto options = rclcpp::NodeOptions().arguments(
    {"--ros-args",
      "-r",
      std::string("__node:=") + std::string(node->get_name()) + client_node_name + "_rclcpp_node",
      "-p",
      "use_sim_time:=" + std::string(node->get_parameter("use_sim_time").as_bool() ? "true": "false"),
      "--"});
```

SteveMacenski, 2022-08-18:
Can you submit a PR :heart: ?

tonynajjar, 2022-08-21:
Here is a draft https://github.com/ros-planning/navigation2/pull/3131

closed 2022-08-25 by SteveMacenski