ros-navigation/navigation2 #3132: Collision Monitor polygons are frozen in place

MrOCW, 2022-08-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 docker container
- ROS2 Version:
  - Humble
- Version or commit hash:
  - 1421f17bd033b7c76affb655885a9f44904e1ecf
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Add collision monitor to bringup

#### Expected behavior
Polygons follow base_link
#### Actual behavior
Polygons are frozen in place
#### Additional information
[image]

MrOCW, 2022-08-22:
Apologies, the polygons do move along with base_link but only after navigation starts.

closed 2022-08-22 by MrOCW

marcusvinicius178, 2024-07-19:
Hi @MrOCW were you able to reproduce the collision monitor tutorial? I had similar issue as you and could not overcome yet: 

https://robotics.stackexchange.com/questions/112280/collision-monitor-in-nav2-not-working-properly-polygons-not-displayed-or-follow

I will be great on any assistance on this regard