ros-navigation/navigation2 #3133: Navigation aborts and does not output cmd_vel topic anymore with new poses

MrOCW, 2022-08-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 docker container
- ROS2 Version:
  - Humble
- Version or commit hash:
  - 1421f17bd033b7c76affb655885a9f44904e1ecf
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Not reproduceable as happens very randomly

#### Expected behavior
Sending a new waypoint should get the robot moving again
#### Actual behavior
No errors from any node but sending new waypoint just leads to
`[component_container_isolated-1] [WARN] [1661141666.820260355] [waypoint_follower]: [follow_waypoints] [ActionServer] Aborting handle.`

SteveMacenski, 2022-08-22:
This ticket needs much more detail.

SteveMacenski, 2022-08-30:
No response - closing

closed 2022-08-30 by SteveMacenski