ros-navigation/navigation2 #3135: nav2_bringup: Multi robot sim - camera rgb frame link undefined in model

ipa-kut, 2022-08-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humnle
- Version or commit hash:
  - 4c84fdde7a2955c9a4a7bdcf156db33e90d476a0
- DDS implementation:
  - Default in ROS2 Humble

#### Steps to reproduce issue

```
ros2 launch nav2_bringup multi_tb3_simulation_launch.py
```

#### Expected behavior

Multi robot simulation launches properly without errors

#### Actual behavior

Everything still works except for the following error messages noticed, which apparently can still be ignored:

```
[gazebo-1] [Err] [Model.cc:123] Error Code 23 Msg: FrameAttachedToGraph error, Non-LINK vertex with name [camera_rgb_joint] is disconnected; it should have 1 outgoing edge in MODEL attached_to graph.
[gazebo-1] [Err] [Model.cc:123] Error Code 23 Msg: Graph has __model__ scope but sink vertex named [camera_rgb_joint] does not have FrameType LINK or MODEL when starting from vertex with name [camera_rgb_joint].
[gazebo-1] [Err] [Model.cc:123] Error Code 26 Msg: PoseRelativeToGraph error, Vertex with name [camera_rgb_joint] is disconnected; it should have 1 incoming edge in MODEL relative_to graph.
[gazebo-1] [Err] [Model.cc:123] Error Code 26 Msg: PoseRelativeToGraph frame with name [camera_rgb_joint] is disconnected; its source vertex has name [camera_rgb_joint], but its source name should be __model__.
[controller_server-21] [INFO] [1661179723.048118251] [robot2.local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[gazebo-1] [Err] [Joint.cc:297] EXCEPTION: Couldn't Find Child Link[camera_rgb_frame]

```
I typically just remove this section of the model file for the joint from [here](https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/worlds/waffle.model#L457) and this resolves the error messages. If this is an acceptable solution, I can make PRs for this.

SteveMacenski, 2022-08-22:
Please! Also make sure to file it against `main` not `humble`. I can auto-backport to Humble after the PR is approved.

ipa-kut, 2022-08-31:
@SteveMacenski  Changed the branch filed against to main.   
Created the following PR to close this issue: https://github.com/ros-planning/navigation2/pull/3162

closed 2022-08-31 by SteveMacenski