ros-navigation/navigation2 #3142: Backup recovery fails

MrOCW, 2022-08-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 docker container
- ROS2 Version:
  - 
- Version or commit hash:
  - 1421f17bd033b7c76affb655885a9f44904e1ecf
- DDS implementation:
  - CycloneDDS

#### Expected behavior
Backs up successfully
#### Actual behavior
Tries to back up but fails and produce the errors below. Strangely there are no obstacles around the robot so it somehow always returns a Pose Goes Off Grid and a Collision Ahead..
Continues to the other round robin recovery behaviors

#### Additional information
nav2_1    | [behavior_server-11] [INFO] [1661241294.169048513] [behavior_server]: Attempting backup
nav2_1    | [behavior_server-11] [ERROR] [1661241294.172523600] [behavior_server]: Pose Goes Off Grid.
nav2_1    | [behavior_server-11] [WARN] [1661241294.172644726] [behavior_server]: Collision Ahead - Exiting DriveOnHeading
nav2_1    | [behavior_server-11] [WARN] [1661241294.172680408] [behavior_server]: backup failed
nav2_1    | [behavior_server-11] [WARN] [1661241294.172720666] [behavior_server]: [backup] [ActionServer] Aborting handle.

Wrote 2 custom behavior nodes:
IsObstacle just checks a distance ahead if there are obstacles then goes into the recovery subtree
IsCostmapWithinThreshold writes the direction & magntitide of spin to the blackboard
```
<?xml version="1.0"?>
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <RecoveryNode number_of_retries="6" name="NavigateRecovery">
      <PipelineSequence name="NavigateWithReplanning">
        <RateController hz="2.0">
          <RecoveryNode number_of_retries="1" name="ComputePathToPose">
            <Fallback>
              <ReactiveSequence>
                <Inverter>
                  <PathExpiringTimer seconds="3" path="{path}"/>
                </Inverter>
                <Inverter>
                  <GlobalUpdatedGoal/>
                </Inverter>
              </ReactiveSequence>
              <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
            </Fallback>
            <BackUp backup_dist="4.0" backup_speed="0.2"/>
          </RecoveryNode>
        </RateController>
        <Sequence name="Sequence">
          <Inverter>
            <IsObstacle cost_threshold="250" costmap_topic="local_costmap/costmap" distance="1.5"/>
          </Inverter>
          <RecoveryNode number_of_retries="1" name="FollowPath">
            <FollowPath path="{path}" controller_id="FollowPath"/>
            <ClearEntireCostmap name="ClearLocalCostmap-Context" service_name="local_costmap/clear_entirely_local_costmap"/>
          </RecoveryNode>
        </Sequence>
      </PipelineSequence>

      <ReactiveFallback name="RecoveryFallback">
        <GoalUpdated/>
        <RoundRobin name="RecoveryActions">
          <Sequence name="ClearingActions">
            <ClearEntireCostmap name="ClearLocalCostmap-Subtree" service_name="local_costmap/clear_entirely_local_costmap"/>
            <ClearCostmapAroundRobot name="ClearGlobalCostmap-Subtree" reset_distance="20" service_name="global_costmap/clear_around_global_costmap" server_timeout="10"/>
          </Sequence>
          <Sequence name="Sequence2">
            <IsCostmapWithinThreshold name="IsCostmapWithinThreshold" cost_threshold="700" costmap_topic="local_costmap/costmap" diff_threshold="30" distance="1.5" num_points="3" spin_dist="{spin_dist}"/>
            <BackUp backup_dist="4.0" backup_speed="0.2"/>
            <Spin spin_dist="{spin_dist}"/>
          </Sequence>
          <Wait wait_duration="2"/>
          <BackUp backup_dist="4.0" backup_speed="0.2"/>
        </RoundRobin>
      </ReactiveFallback>
    </RecoveryNode>
  </BehaviorTree>
</root>

```

I am using TEB controller and Theta Star planner

SteveMacenski, 2022-08-23:
> Pose Goes Off Grid.

That seems pretty clear to me, are you on the bounds of your costmap? That error is in the costmap collision checker indicating that it couldn't convert the global coordinates into costmap coordinates due to being out of bounds. 

> Why is DriveOnHeading being used when the tree does not use it?

Under the hood Backup and DriveOnHeading are nearly identical so they share portions of code. They're really only semantically different from a BT designer's perspective so we keep them separated to signify the differences between a recovery and a primitive action

MrOCW, 2022-08-23:
> are you on the bounds of your costmap?

Nope, the robot is in the middle of the local costmap(15x15) and no where near the bounds of the global costmap

SteveMacenski, 2022-08-23:
Can you visualize your local costmap while this happens - are you getting the topic and its updating properly? Are you using composition in your bring up so that all of the nodes are in a single process? Wondering if maybe some costmap updates aren't getting to the behavior server, which is unlikely unless you're not using composition + overloading your CPU.

MrOCW, 2022-08-24:
> Can you visualize your local costmap while this happens - are you getting the topic and its updating properly?

Yes it looks ok

> Are you using composition in your bring up so that all of the nodes are in a single process?

No I was not as it was easier to debug without it. I did look at the CPU usage, but it was quite low!

> Does this happen every time? Any common traits when it does happen?

Yes every time when it attempts a backup.

MrOCW, 2022-08-24:
@SteveMacenski I reverted to a previous commit of my project when the backup still works and slowly tried adding back whatever new stuff I had and, I found that after adding velocity smoother, the backup stopped working and started returning the errors above!

closed 2022-08-24 by MrOCW

daviddavo, 2023-01-04:
Shouldn't then the remapping be done also on navigation_launch.py? I also had this problem, but it seems to be the default behavior

https://github.com/ros-planning/navigation2/blob/58a651ce525a2a5146f2f1d8e27b6ac7b82e0635/nav2_bringup/launch/navigation_launch.py#L140-L149

Btw: Why are there two cmd_vel_nav publishers in the first place? I don't quite understand it

SteveMacenski, 2023-01-04:
@daviddavo Yes! I wasn't quite able to parse what @MrOCW meant in his last message but if there's an issue with topic remapping to support this, a PR would be much appreciated!

daviddavo, 2023-01-09:
I'm using Coppelia instead of Gazebo and I still haven't made it work. I can do a PR, but I haven't been able to test it yet.