ros-navigation/navigation2 #3149: Use SetParameter Launch API to set the yaml filename for map_server

SteveMacenski, 2022-08-25:
Would let us remove the RewrittenYaml remappings entirely and let us remove the requirement of a bogus `yaml_filename` entry in the yaml file to be overwritten. Now, it could just not be present at all and that would be fine. Just need to make sure that this is set only for the `map_server` node, which should be straight forward. 

See https://github.com/ros-planning/navigation2/pull/3131 for a motivating example and fixes https://github.com/ros-planning/navigation2/issues/3111 to be less quirky @superseppl. 

Note that we should also remove the `note` I added to the navigation.ros.org configuration for map server explaining that `yaml_filename` is strictly required.

stevedanomodolor, 2022-09-02:
@SteveMacenski I can try it this weekend

stevedanomodolor, 2022-09-04:
This might sound like a silly question but just to confirm. When you mean by

> Just need to make sure that this is set only for the `map_server` node, which should be straight forward.

If the set parameter is set for only the map_server node, what about other node that takes uses the params. For example the following launch file from the localization launch file 

```
load_nodes = GroupAction(
        condition=IfCondition(PythonExpression(['not ', use_composition])),
        actions=[
            SetParameter("use_sim_time", use_sim_time),
Node(
                package='nav2_map_server',
                executable='map_server',
                name='map_server',
                output='screen',
                respawn=use_respawn,
                respawn_delay=2.0,
                parameters=[configured_params],
                arguments=['--ros-args', '--log-level', log_level],
                remappings=remappings),
            Node(
                package='nav2_amcl',
                executable='amcl',
                name='amcl',
                output='screen',
                respawn=use_respawn,
                respawn_delay=2.0,
                parameters=[configured_params],
                arguments=['--ros-args', '--log-level', log_level],
                remappings=remappings),
            Node(
                package='nav2_lifecycle_manager',
                executable='lifecycle_manager',
                name='lifecycle_manager_localization',
                output='screen',
                arguments=['--ros-args', '--log-level', log_level],
                parameters=[{'autostart': autostart},
                            {'node_names': lifecycle_nodes}])
```
Similar to https://github.com/ros-planning/navigation2/pull/3131, would it not be also effective to set the yaml filename similarly as the  use_sim_time is set?

SteveMacenski, 2022-09-05:
Awesome!

closed 2022-11-07 by SteveMacenski