ros-navigation/navigation2 #3151: Simple Commander Costmap API

SteveMacenski, 2022-08-26:
Add a costmap API for `OccupancyGrid` messages coming into the Simple Commander API

Basically have the basic API from https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d.hpp + adding global frame ID and timestamp with a constructor that takes in the `OccupancyGrid`.

Potentially even a python collision checker too when given a footprint from the footprint topic + this new `CostmapPy` class

padhupradheep, 2022-08-26:
For someone who is going to implement it, we had something something similar, which can be found under https://github.com/neobotix/neo_nav2_py_costmap2D

(wish it was me, but unfortunately I don't have time until october to get into this)

stevedanomodolor, 2022-08-26:
I would love to try it, is t possible?

SteveMacenski, 2022-08-26:
Absolutely!

afifswaidan, 2022-08-27:
I would like to contribute as well, let me know what I can do!

igrak34, 2022-08-27:
Same here, let me know if you need some help with it :)

manishsaini0901, 2022-08-28:
I would love to contribute. Let me know if we can do it as a team. @SteveMacenski @afifswaidan @igrak34 @stevedanomodolor

afifswaidan, 2022-08-28:
> I would love to contribute. Let me know if we can do it as a team. @SteveMacenski @afifswaidan @igrak34 @stevedanomodolor 

I have no problem from my side, count me in.

stevedanomodolor, 2022-08-29:
> I would love to contribute. Let me know if we can do it as a team. @SteveMacenski @afifswaidan @igrak34 @stevedanomodolor

Sure me too, I don't mind

JacksonK9, 2022-08-29:
I would like to contribute. But, I am not so familiar with ROS2, but familiar with ROS1.
Can I try to contribute with as a team? @SteveMacenski @afifswaidan @igrak34 @stevedanomodolor

SteveMacenski, 2022-08-29:
Awesome! 

So the general things that need to be implemented are those in the files I linked to above, namely a class `PyCostmap2D` 

- A: Constructor for OccupancyGrid which stores the relevant information 
- A: getSizeInCellsX / getSizeInCellsY
- A: getSizeInMetersX / getSizeInMetersY
- A: getOriginX / getOriginY
- A: getResolution
- A: getGlobalFrameID
- A: getCostmapTimestamp
- B: getCost at map pixels and index
- B: setCost at map pixels
- B: mapToWorld
- B: worldToMap
- B: getIndex

That's 1 item and once that's done another person could contribute a costmap collision checker, where the object should take in this costmap object + a footprint to perform collision checking, perhaps using @padhupradheep 's line iterator: https://github.com/neobotix/neo_nav2_py_costmap2D/blob/main/neo_nav2_py_costmap2D/line_iterator.py analogous to https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/include/nav2_costmap_2d/footprint_collision_checker.hpp

I think there are a couple of natural roles here to help!
- Person A: Implement the PyCostmap2D class's first few functions marked in "A"
- Person B: Implement the methods for converting / getting information from the occupancy grid after the core API is done
- Person C: Implement and test a line iterator to march through the costmap given 2 points and check for collision
- Person D: Use that line iterator + costmap object + footprint input to check for collisions!

Who wants to get started with A and C? I think those can both be worked on in parallel

JacksonK9, 2022-08-30:
@SteveMacenski If I can do, I want to do the role of B.

stevedanomodolor, 2022-08-30:
I just created a draft pull request https://github.com/ros-planning/navigation2/pull/3159.  I should have this done by the end of tomorrow.

afifswaidan, 2022-08-30:
> Awesome!
>
> So the general things that need to be implemented are those in the files I linked to above, namely a class `PyCostmap2D`
>
> - A: Constructor for OccupancyGrid which stores the relevant [...]

Hello,

So I guess A is taken, you need now someone for C.
I can handle C if you want.

afifswaidan, 2022-08-30:
@SteveMacenski how do you think it's best to organize and kickoff the group work and track the progress?

stevedanomodolor, 2022-08-30:
We can use something like this, @SteveMacenski @afifswaidan 
## Task tracking

Author | Task assigned | Task Completed | Pull request Accepted| Task Dependency
--- | --- | --- | --- |--- 
@stevedanomodolor  |  A |  ❌ | ❌ | - 
--- |  B | ❌| ❌ | A
 --- |  C |  ❌| ❌ | -
 ---|  D | ❌| ❌ | A B C

manishsaini0901, 2022-08-30:
So, A, B, and C are taken?

manishsaini0901, 2022-08-30:
For D I think we have to collaborate somehow as it depends upon A,B and C.

afifswaidan, 2022-08-30:
> We can use something like this, @SteveMacenski @afifswaidan 
> ## Task tracking
> 
> Author | Task assigned | Task Completed | Pull request Accepted| Task Dependency
> --- | --- | --- | --- |--- 
> @stevedanomodolor  |  A |  ❌ | ❌ | - 
> --- |  B | ❌| ❌ | A
>  --- |  C |  ❌| ❌ | -
>  ---|  D | ❌| ❌ | A B C

Sounds good, I'll take C then, please add me to the table :)

stevedanomodolor, 2022-08-30:
@manishsaini0901  do you want take over the task?, Someone has to make the changes even though you might require inputs from authors of the other tasks

stevedanomodolor, 2022-08-30:
Updated on task
## Task tracking

Author | Task assigned | Task Completed | Pull request Accepted| Task Dependency
--- | --- | --- | --- |--- 
@stevedanomodolor  |  A |  :heavy_check_mark:   | ❌ | - 
@JacksonK9 |  B | ❌| ❌ | A
 @afifswaidan  |  C |  ❌| ❌ | -
 ---|  D | ❌| ❌ | A B C

JacksonK9, 2022-08-31:
@stevedanomodolor Can I start with your github repository? or Do I have to wait for the acception of Pull Request?

stevedanomodolor, 2022-08-31:
I think I would wait when my pull request is accepted. You can fork directly from navigation2 if my pull request is merged to have the current main. If we do the pull in the end of finishing the whole task, you might have to use my own fork

JacksonK9, 2022-08-31:
@stevedanomodolor Okay, I understand.

@SteveMacenski Which is better?
Can I start with official nav2 repository after the acception of Pull Request? or Just start with stevedanomodolor's nav2 repository?

stevedanomodolor, 2022-08-31:
@JacksonK9 if my pull is accepted you can use the official directly

SteveMacenski, 2022-08-31:
Task A done!

JacksonK9, 2022-09-01:
@SteveMacenski Thank you
maybe, I can finish the task B until the end of this week.

SteveMacenski, 2022-09-02:
https://github.com/ros-planning/navigation2/issues/3149 This is another good Python3 project if folks were interested!

stevedanomodolor, 2022-09-02:
@SteveMacenski I can try it this weekend,

afifswaidan, 2022-09-03:
@SteveMacenski for task C I already started implementing, but I need some more functions in the costmap_2d class, should I add these as well to the costmap_2d.py file that was committed 3 days ago or @stevedanomodolor is already working/committing them? Just to know if I should wait or go ahead and implement them to continue working on Task C.

stevedanomodolor, 2022-09-03:
That is been done by @JacksonK9 for task B i think, you might have to wait for him to finish his task. What do you think @SteveMacenski 

B: getCost at map pixels and index
B: setCost at map pixels
B: mapToWorld
B: worldToMap
B: getIndex

afifswaidan, 2022-09-04:
> That is been done by @JacksonK9 for task B i think. FOr me I am done with my task and has already been pushed.
> 
> B: getCost at map pixels and index
> B: setCost at map pixels
> B: mapToWorld
> B: worldToMap
> B: getIndex

Ah alright thanks @stevedanomodolor !
@JacksonK9 and @SteveMacenski any thoughts? Do you want me to wait for task B to be fully finished or should I implement what I need to proceed with C?

JacksonK9, 2022-09-04:
Updated on task: @stevedanomodolor  @afifswaidan @SteveMacenski 
## Task tracking

Author | Task assigned | Task Completed | Pull request Accepted| Task Dependency
--- | --- | --- | --- |--- 
@stevedanomodolor  |  A |  ✅    | ✅  | - 
@JacksonK9 |  B | :white_check_mark: | ❌ | A
 @afifswaidan  |  C |  ❌| ❌ | -
 ---|  D | ❌| ❌ | A B C

afifswaidan, 2022-09-04:
> Updated on task: @stevedanomodolor  @afifswaidan @SteveMacenski 
> ## Task tracking
> 
> Please give me the feedback of my work.
> 
> It is my first time to do the open source contribution work. 
> 
> Sorry for delay @afifswaidan 
> 
> Author | Task assigned | Task Completed | Pull request Accepted| Task Dependency
> --- | --- | --- | --- |--- 
> @stevedanomodolor  |  A |  ✅    | ✅  | - 
> @JacksonK9 |  B | :white_check_mark: | ❌ | A
>  @afifswaidan  |  C |  ❌| ❌ | -
>  ---|  D | ❌| ❌ | A B C

No worries, there's nothing to be sorry for. Let me know if you need any support then.

I'll wait for @SteveMacenski reply so we know how to proceed.

SteveMacenski, 2022-09-06:
Probably wait for it to be merged or at least approved so that we know the API isn't going to change for you to use

afifswaidan, 2022-09-06:
> Probably wait for it to be merged or at least approved so that we know the API isn't going to change for you to use 

Okay, sounds good, I'll wait for the updates and finalize task C. If anyone needs further support with the other tasks I'll be glad to assist.
Thanks!

SteveMacenski, 2022-09-12:
Ask B merged!

afifswaidan, 2022-09-12:
> Task B merged!

Great news! I'll handle task C at the weekend.
Thanks.

JacksonK9, 2022-09-13:
> Task B merged!

Updated on task: @stevedanomodolor  @afifswaidan @SteveMacenski 

It's been little long time to finish task B.

Thanks for your patience @afifswaidan @SteveMacenski 

And also, Thank you for your comment on my code @SteveMacenski 

## Task tracking
Author | Task assigned | Task Completed | Pull request Accepted| Task Dependency
--- | --- | --- | --- |--- 
@stevedanomodolor  |  A |  ✅    | ✅  | - 
@JacksonK9 |  B | :white_check_mark: | ❌ | A
 @afifswaidan  |  C |  ❌| ❌ | -
 ---|  D | ❌| ❌ | A B C

JacksonK9, 2022-09-13:
If I have some more things do with Python.

Please notify me 👍 

Thank you @SteveMacenski

afifswaidan, 2022-09-13:
> > Task B merged!
> 
> Updated on task: @stevedanomodolor  @afifswaidan @SteveMacenski 
> 
> It's been little long time to finish task B.
> 
> Thanks for your patience @afifswaidan @SteveMacenski 
> 
> And also, Thank you for your comment on my code @SteveMacenski 
> 
> ## Task tracking
> Author | Task assigned | Task Completed | Pull request Accepted| Task Dependency
> --- | --- | --- | --- |--- 
> @stevedanomodolor  |  A |  ✅    | ✅  | - 
> @JacksonK9 |  B | ✅  | ✅  | A
>  @afifswaidan  |  C |  ❌| ❌ | B
>  ---|  D | ❌| ❌ | A B C
> 
> 

No problem at all, well done and thanks for the effort!

awesomebytes, 2022-09-16:
I'm late to the conversation, but if anyone wants to take a look at an old costmap managing class for ROS1 as a reference:
https://github.com/awesomebytes/occupancy_grid_python

P.S.: I love the idea of having access via Python to as many things as possible :)

SteveMacenski, 2022-09-16:
Yeah! This should be virtually the same except without the internal costmap subscriber object, we assume a `OccupancyGrid` message is given. Though, it would not be difficult to wrap this base-class in another class with that subscription to have the same behavior.

SteveMacenski, 2022-10-31:
The line iterator is virtually done! Just removing some excess functions, I think we can get started on the collision checker using the line iterator + footprint in the costmap

SteveMacenski, 2022-11-08:
Essentially this, but in python, using the Costmap object + line iterator object created using this ticket: https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/footprint_collision_checker.cpp

Given a footprint as an API input, check it for collision.

afifswaidan, 2022-11-08:
> Essentially this, but in python, using the Costmap object + line iterator object created using this ticket: https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/footprint_collision_checker.cpp
> 
> Given a footprint as an API input, check it for collision.
> 

Alright everything is clear now, I can take it then.
Thanks for the reference!

brammmieee, 2022-12-01:
Hi there! I updated to the latest version of the simple commander API and noticed a mistake [https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#:~:text=rtn%20%3D%20self._getPathImpl(start%2C%20goal%2C%20planner_id%3D%27%27%2C%20use_start%3DFalse)](here!) Shall I make a PR or is this comment satisfactory? Thanks for all the great work btw!

brammmieee, 2022-12-01:
Hi there! I updated to the latest version of the API and noticed a bug in the [getpath](https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#:~:text=(start%2C%20goal%2C-,planner_id%3D%27%27%2C%20use_start%3DFalse,-)) method. Shall I create a PR to change this or is this comment enough? Thanks for the great work btw!

SteveMacenski, 2022-12-01:
I don't know what bug you mean, so a PR would be appreciated so I can see what changes you made and what issue it resolves!

SteveMacenski, 2022-12-15:
https://discourse.ros.org/t/nav2-costmap-collision-checking-python-api-added-to-nav2-simple-commander/28766