ros-navigation/navigation2 #3156: Changing behavior_tree from Simple Commander API causes navigation on cancelled goal

automech-rb, 2022-08-28:
## Bug report

**Required Info:**

- Operating System:
  - Docker 22.04 Ubuntu
- ROS2 Version:
  - rolling
- Version or commit hash:
  - latest 1d9d3c6
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

```
Use Simple Commander API and request any goal using:
-> goToPose(pose1, behavior_tree=’../my_navigate_to_pose_w_replanning_and_recovery.xml‘)
Here behavior_tree can be any tree (even same as default used by Nav2)

-> Then cancel the above navigation using cancelTask() API.

-> Next use goToPose(pose2) or use Rviz2 to request different goal using default bt tree.
```

#### Expected behavior
Navigate to new goal(pose2) according to default behaviour tree.

#### Actual behavior
Nav2 navigates to cancelled goal(pose1) instead of new requested goal. 
Gif is shown below as example.

#### Additional information

[image]
In this video I am using [follow_point.xml](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/follow_point.xml) as behavior (bug can be repoduced with any behaviour), and then using nav2 API and then cancelling it.
Then rviz2 for another goal request.

SteveMacenski, 2022-08-29:
Do you only see this through the simple commander API for the first request or with specific behavior trees? I regularly cancel and reissue goals in rviz2 for testing using the default BT and do not see that behavior.

automech-rb, 2022-08-30:
Hi @SteveMacenski 
This bug can be reproducer may time in a run repeatedly.
I first need to set `default_nav_to_pose_bt_xml` in param file to [navigate_w_..._if_path_becomes_invalid.xml](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml), as the Nav2 default bt tree in not able to reproduce the issue. 
Also, just using Rviz2 is not able to reproduce the issue, you need to use Nav2 python API with different behaviour tree to request a goal and then cancel it using API. Then use Rviz2 or python API again to request goal with default   [navigate_w_..._if_path_becomes_invalid.xml](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml) as stated in my report.

Here are the logs:
```
[bt_navigator-16] [INFO] [1661818536.533826771] [bt_navigator]: Begin navigating from current location to (-3.50, -6.92)
[controller_server-12] [INFO] [1661818536.715852307] [controller_server]: Received a goal, begin computing control effort.
[controller_server-12] [INFO] [1661818537.816026861] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO] [1661818538.816038295] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO] [1661818539.816025915] [controller_server]: Passing new path to controller.
[bt_navigator-16] [INFO] [1661818540.864537583] [bt_navigator]: Goal canceled
[bt_navigator-16] [WARN] [1661818540.864587410] [bt_navigator]: [navigate_to_pose] [ActionServer] Client requested to cancel the goal. Cancelling.
[controller_server-12] [INFO] [1661818540.916024917] [controller_server]: Goal was canceled. Stopping the robot.
[controller_server-12] [WARN] [1661818540.916074971] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[controller_server-12] [WARN] [1661818540.916251866] [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[bt_navigator-16] [INFO] [1661818549.518423350] [bt_navigator]: Begin navigating from current location to (3.90, 2.00)
[controller_server-12] [INFO] [1661818549.521891848] [controller_server]: Received a goal, begin computing control effort.
[controller_server-12] [INFO] [1661818567.722652652] [controller_server]: Reached the goal!
[bt_navigator-16] [INFO] [1661818567.758608174] [bt_navigator]: Goal succeeded
```

Here, the new requested goal  x ,y :: (3.90, 2.00) is not respected (even though it is logged correctly by bt_navigator), but instead navigates to old cancelled goal.  

Also, I am using rviz tool provided by nav2 but the issue can be reproduced just using python API. Just don't specify behaviour tree in 2nd goal request (after cancelling 1st goal) , as stated in my bug report.

SteveMacenski, 2022-08-30:
I see that we share the same `blackboard_` in the BT Action Server even if we change bebehavior trees themselves. We could destroy the blackboard and recreate it if BT's change (that seems sensible; move [this block](https://github.com/ros-planning/navigation2/blob/7703c816a11368fb1a48663dc92e4e93affb637e/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L111-L117) into a new function called on configuration + changing BTs). But to me that doesn't really solve the core issue of how this could be happening in the first place -- then secondarily cleaning up the BT for switching.

The correct `Begin navigating from current location to ...` pose tells me that its getting the request and setting it in the blackboard - so it seems likely that the error isn't in the BT navigator if it only happens with a specific behavior tree(s), but rather in how the tree is running.

 The 2 nodes that happen before computing the path to the pose are `GloballyUpdatedGoalCondition` and `IsPathValidCondition` which are both unique to these sets of BTs so that tracks for me as a good place to start looking.

If I were to make a guess based on a few minutes of looking @jwallace42 but without actually tracking down the error itself to validate; I think what's happening is that [this part of the BT](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml#L14-L19) under `ReactiveSequence` is returning so that we don't compute a path via `ComputePathToPose` so its using the old path on the blackboard. 

@automech-rb this only happens when the second request uses the `if_path_becomes_invalid.xml` BTs, correct?

automech-rb, 2022-08-30:
Hello @SteveMacenski 

Yeah, the problem occurs when using `if_path_becomes_invalid.xml`. But this is not the only condition. The xml file also needs to be specified in 2nd goal using API (I updated my script in above comment as an example). If I normally use the API without using xml file, everything works perfectly.

automech-rb, 2022-08-31:
@SteveMacenski Yeah, this only occurs when using `if_path_becomes_invalid`. Also need to specify bt xml file in `goToPose` API. I updated the python script as an example. If I don't specify xml file in API, everything works as expected.

As you both commented earlier, I feel this is more of issue of how  blackboard_ is handled when using xml file.

SteveMacenski, 2022-08-31:
> Also need to specify bt xml file in goToPose API. I updated the python script as an example. If I don't specify xml file in API, everything works as expected.

This is where we're really missing each other and need a clear statement of what specifically works and what specifically doesn't. Are you saying that if you just `goToPose` even using the `if_path_becomes_invalid.xml` as the default and cancel between requests, its fine if its not specified in the action request? So this only happens when both using `if_path_becomes_invalid.xml` **and** also specifying it in the request?

automech-rb, 2022-09-01:
Hello @SteveMacenski 
Sorry for the confusion. Let me explain it in more clearly:

### **A. This error does not occur**:
Set default behavior tree `default_nav_to_pose_bt_xml` as [navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml) in param file.
Use Simple Commander API and request any goal using:
-> goToPose(pose1)
-> cancelTask()
-> Next use goToPose(pose2)

### **B. This error occurs**:
Case 1 (original bug report method): 
Set default behavior tree `default_nav_to_pose_bt_xml` as [navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml) in param file.
Use Simple Commander API and request any goal using:
-> goToPose(pose1, ’navigate_to_pose_w_replanning_and_recovery.xml‘)
Here behavior_tree file in API can be any tree (even same as default used by Nav2 but using file)
-> Then cancel the above navigation using cancelTask() API call.
-> Next use goToPose(pose2)

Case 2 (script method):
Don't set  `default_nav_to_pose_bt_xml`  in param file (leave it as Nav2 default).
Use Simple Commander API and request any goal using:
-> goToPose(pose1)
-> Then cancel the above navigation using cancelTask() API call.
-> Next use goToPose(pose2, 'navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml')

In both error cases, I am specifying behavior file in python API, which led to my conclusion that using xml file in python API causes this error.

SteveMacenski, 2022-09-01:
https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L143-L145

The goal message is populated each time - so I don't think its from the Python API unless something very odd is happening - it seems more likely with the BT XML handling or something in BT navigator - since I think you said it happens via rviz too.

jwallace42, 2022-09-07:
@automech-rb 

Can edit the bringup_launch.py file by switching the log level from info to debug. 

Then launch rqt_console by 
`ros2 run rqt_console rqt_console` there is a red icon to start recording the log messages. 

Once it is starting to record, run the simulation and reproduce the error. Then stop the recording, save the file and set it to me so I can take a look :). 

The other thing you should try so we can eliminate the nav2_simple_commander as an issue is sending the goals manually though the command line. 

`josh@josh-desktop:~/nav2_ws/src/navigation2$ ros2 action send_goal /navigate_to_pose nav2_msgs/action/NavigateToPose "pose:
  header:1.0
    stamp:0.5
      sec: 0
      nanosec: 0
    frame_id: 'map'
  pose:: 0.0
    position:
      x: 1.0
      y: -0.5
      z: 0.0: ''" 
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
behavior_tree: ''" 
`

If you can reproduce the error then we definitely know that it is something below the nav2_simple commander.

jwallace42, 2022-09-07:
Or you can send me your branch :)

automech-rb, 2022-09-07:
Hello @jwallace42 
I am not able to share the files due to company policies. But i think you can recreate the scenario easily.

I tested the pr but still leads to same error behaviour. I am also adding the log debug file as you stated.
## Setup: 
Case 2 (script method):
Don't set default_nav_to_pose_bt_xml in param file (leave it as Nav2 default).
Use Simple Commander API and request any goal using:
-> goToPose(pose1)
-> Then cancel the above navigation using cancelTask() API call.
-> Next use goToPose(pose2, 'navigate_w_recovery_and_replanning_only_if_path_becomes_invalid.xml')

#### Python script and log file

[files.zip](https://github.com/ros-planning/navigation2/files/9522236/files.zip)

In `test.py`, AGV in the end goes to pose1 instead of pose 2.

SteveMacenski, 2022-09-08:
Not surprised, that PR does something in the same direction as the previous work just resetting the path between runs, but that's not the root cause of why things aren't passing correctly around.

jwallace42, 2022-09-08:
Yeah, 

I though halt() would be called when the behavior tree completed which was a misunderstanding. 

Pretty much, what is happening is the path gets planned. When the task is cancelled the computePathToPose is already completed so the blackboard still contains the old path.  

So, when ever a behavior tree ever gets loaded we need to make sure the path is empty. The simplest thing (which I just put in my pr and works) is to reset the path every time the behavior tree is loaded. This is a bad long term solution as there could be more params that would need to be reset between runs. 

To clean this up, we would just clear the blackboard on reloading but this causes another issue. There are parameters for the blackboard that get passed down from the bt_navigator to be loaded by the navigator.hpp. As a result, we can't clear all the parameters.

SteveMacenski, 2022-09-08:
>  so the blackboard still contains the old path ... when ever a behavior tree ever gets loaded we need to make sure the path is empty

Shouldn't a new path be being computed in that BT either way before the controller is called? I'd like to understand the exact reason why a new path isn't available already in that BT when FollowPath BT node is invoked.  That to me seems more like the bug -- because even if the path is reset on the blackboard, that means the FollowPath is still being called on an empty path.

automech-rb, 2022-09-09:
@jwallace42 I tested the PR, it is working now as expected!

SteveMacenski, 2022-09-09:
Got it! Why didn't this cause problems before though when calling the same try as the default multiple times in a row?

jwallace42, 2022-09-09:
So, 

first time through planning gets triggered by is path valid. Then, when we send a action to the behavior tree again the tree does not get reloaded so the globalUpdatedGoal works correctly and causes us to replan.

SteveMacenski, 2022-09-09:
Ah, I understand, thanks!

SteveMacenski, 2022-09-09:
PR is approved, just needs updates for test failures

SteveMacenski, 2022-09-09:
Merged!

closed 2022-09-09 by SteveMacenski