ros-navigation/navigation2 #3157: Nav2-velocity-smoother not found as binary

austin-InDro, 2022-08-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 AMD64
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
```
ros-humble-nav2-amcl    1.1.0-2jammy.20220808.152642
ros-humble-nav2-behavior-tree   1.1.0-2jammy.20220808.152432
ros-humble-nav2-behaviors       1.1.0-2jammy.20220808.162406
ros-humble-nav2-bringup 1.1.0-2jammy.20220808.173547
ros-humble-nav2-bt-navigator    1.1.0-2jammy.20220808.162459
ros-humble-nav2-common  1.1.0-2jammy.20220729.152646
ros-humble-nav2-controller      1.1.0-2jammy.20220808.162526
ros-humble-nav2-core    1.1.0-2jammy.20220808.161443
ros-humble-nav2-costmap-2d      1.1.0-2jammy.20220808.155624
ros-humble-nav2-dwb-controller  1.1.0-2jammy.20220808.163911
ros-humble-nav2-lifecycle-manager       1.1.0-2jammy.20220808.155251
ros-humble-nav2-map-server      1.1.0-2jammy.20220808.152709
ros-humble-nav2-msgs    1.1.0-2jammy.20220729.155054
ros-humble-nav2-navfn-planner   1.1.0-2jammy.20220808.163301
ros-humble-nav2-planner 1.1.0-2jammy.20220808.163324
ros-humble-nav2-regulated-pure-pursuit-controller       1.1.0-2jammy.20220808.163430
ros-humble-nav2-rotation-shim-controller        1.1.0-2jammy.20220808.163810
ros-humble-nav2-rviz-plugins    1.1.0-2jammy.20220808.172240
ros-humble-nav2-simple-commander        1.1.0-2jammy.20220729.160105
ros-humble-nav2-smac-planner    1.1.0-2jammy.20220808.163513
ros-humble-nav2-smoother        1.1.0-2jammy.20220808.161609
ros-humble-nav2-theta-star-planner      1.1.0-2jammy.20220808.163422
ros-humble-nav2-util    1.1.0-2jammy.20220808.151531
ros-humble-nav2-voxel-grid      1.1.0-2jammy.20220729.155236
ros-humble-nav2-waypoint-follower       1.1.0-2jammy.20220808.163435
```
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue
```
apt-get install ros-humble-nav2-velocity-smoother
```

#### Expected behavior
Installs package 
#### Actual behavior
Cannot find package
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-nav2-velocity-smoother
```
#### Additional information
These are being run in a Docker container (however, doesn't matter). Also tried `apt search ros-humble-nav2 | grep velocity` on my host machine with no luck

SteveMacenski, 2022-08-29:
It will be in the next humble sync.

closed 2022-08-29 by SteveMacenski

AlexeyMerzlyakov, 2022-08-29:
Confirming the issue: there is no "ros-humble-nav2-velocity-smoother" found at Jammy on docker, or at ros2 distribution [packages list](http://packages.ros.org/ros2/ubuntu/dists/jammy/main/binary-amd64/Packages) (although `nav2_velocity_smoother/` exists in `humble` branch).