ros-navigation/navigation2 #3160: nav2_gazebo_spawner outputs an error when spawning

dohlenaugur, 2022-08-30:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04 focal
- ROS2 Version:
  Foxy
#### Steps to reproduce issue
Execute ros2 launch nav2_bringup multi_tb3_simulation_launch.py

#### Expected behavior
The nav2_gazebo_spawner should spawn without errors.
#### Actual behavior
When launched, the nav2_gazebo_spawner will spawn the robot, but output:

[nav2_gazebo_spawner-2] [INFO] [1661870132.538805784] [entity_spawner]: ...connected!
[nav2_gazebo_spawner-2] [INFO] [1661870132.540188805] [entity_spawner]: spawning `robot1` on namespace `robot1` at 0.0, 0.5, 0.01
[nav2_gazebo_spawner-2] [INFO] [1661870132.546656062] [entity_spawner]: Sending service request to `/spawn_entity`
[nav2_gazebo_spawner-2] Traceback (most recent call last):
[nav2_gazebo_spawner-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 216, in spin_until_future_complete
[nav2_gazebo_spawner-2]     executor.add_node(node)
[nav2_gazebo_spawner-2] AttributeError: 'float' object has no attribute 'add_node'
[nav2_gazebo_spawner-2] 
[nav2_gazebo_spawner-2] During handling of the above exception, another exception occurred:
[nav2_gazebo_spawner-2] 
[nav2_gazebo_spawner-2] Traceback (most recent call last):
[nav2_gazebo_spawner-2]   File "/opt/ros/foxy/lib/nav2_gazebo_spawner/nav2_gazebo_spawner", line 11, in <module>
[nav2_gazebo_spawner-2]     load_entry_point('nav2-gazebo-spawner==0.3.0', 'console_scripts', 'nav2_gazebo_spawner')()
[nav2_gazebo_spawner-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/nav2_gazebo_spawner/nav2_gazebo_spawner.py", line 102, in main
[nav2_gazebo_spawner-2]     rclpy.spin_until_future_complete(node, future, args.timeout)
[nav2_gazebo_spawner-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 219, in spin_until_future_complete
[nav2_gazebo_spawner-2]     executor.remove_node(node)
[nav2_gazebo_spawner-2] AttributeError: 'float' object has no attribute 'remove_node'

#### Additional information
I tried to use the nav2_gazebo_spawner without the multi robot launch file, but got the same error. When I perform a similar request to the spawn service, I get no error

SteveMacenski, 2022-08-30:
@dohlenaugur try https://github.com/ros-planning/navigation2/pull/3161. I think when I backported something a couple of years ago that Foxy API in Python3 didn't get updated to support the feature yet

dohlenaugur, 2022-08-31:
This seems to remove the error. Thank you a lot.

closed 2022-08-31 by dohlenaugur