ros-navigation/navigation2 #3163: bt_navigator

BATAxjh, 2022-09-01:
ros-foxy-

ros2 launch   navigation_launch.py  

[lifecycle_manager-13] [INFO] [1662003008.932584225] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-11] [INFO] [1662003008.934698601] [bt_navigator]: Configuring
[bt_navigator-11] [ERROR] [1662003009.187668684] []: Caught exception in callback for transition 10
[bt_navigator-11] [ERROR] [1662003009.190816521] []: Original error: Could not load library: libnav2_transform_availablemap_condition_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-11] [WARN] [1662003009.194177208] []: Error occurred while doing error handling.
[bt_navigator-11] [FATAL] [1662003009.197183521] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-13] [ERROR] [1662003009.202547305] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-13] [ERROR] [1662003009.202713494] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[map_server-5] [DEBUG] [map_io]: Read map /home/sunrise/robot_robot/install/nav2_bringup/share/nav2_bringup/maps/careRobot.pgm: 1231 X 1106 map @ 0.01 m/cell

why   libnav2_transform_availablemap_condition_bt_node.so: cannot open shared object file: No such file or directory

BATAxjh, 2022-09-01:
in params_file  “nav2_transform_availablemap_condition_bt_node”    change  “ nav2_transform_available_condition_bt_node”    ok!!
 maybe The versions are different

AlexeyMerzlyakov, 2022-09-01:
After install locally Foxy on VM, I'll try to reproduce the problem locally and will share the results.

BATAxjh, 2022-09-02:
Well, maybe I typed it in accidentally，sorry

Daviesss, 2023-03-11:
> Well, maybe I typed it in accidentally，sorry

Hello, were you able to resolve the issue. I also have the same problem.