ros-navigation/navigation2 #3164: multi_tb3_simulation_launch.py not launching as intended

dohlenaugur, 2022-09-01:
## Bug report
- Operating System:
  Ubuntu 20.04 focal
- ROS2 Version:
  Foxy

#### Steps to reproduce issue
Execute  ros2 launch nav2_bringup multi_tb3_simulation_launch.py
#### Expected behavior
Launch turtlebot3 world with 2 turtlebots and nav2 should be able to navigate
#### Actual behavior
Everything launches, but in console there is the error:
[rviz2-15] [INFO] [1662036057.859719256] [robot2.rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-15] [INFO] [1662036057.891850261] [robot2.rviz2]: Message Filter dropping message: frame 'base_scan' at time 38,617 for reason 'Unknown'
[planner_server-21] [INFO] [1662036058.080355069] [robot2.global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist

$ ros2 topic list
/clock
/parameter_events
/robot1/global_costmap/costmap
/robot1/global_costmap/costmap_updates
/robot1/initialpose
/robot1/local_costmap/costmap
/robot1/local_costmap/costmap_updates
/robot1/local_costmap/published_footprint
/robot1/local_plan
/robot1/map
/robot1/map_updates
/robot1/mobile_base/sensors/bumper_pointcloud
/robot1/particlecloud
/robot1/plan
/robot1/scan
/robot1/tf
/robot1/tf_static
/robot1/waypoints
/rosout

#### Additional information
If I restart the launch file, the namespace sometimes switches to robot2, but navigation is not possible.
Both robots were spawned and there are no errors in the console.

AlexeyMerzlyakov, 2022-09-01:
After install locally Foxy on VM, I'll try to reproduce the problem locally and will share the results.

AlexeyMerzlyakov, 2022-09-02:
After running the multi-simulation on Foxy, I've noted that both robots are being created and visualized in Gazebo in all runs:
```
[leha@leha-VM Documents]$ ros2 node list
/gazebo
/robot1/amcl
/robot1/amcl_rclcpp_node
/robot1/bt_navigator
...
/robot1/turtlebot3_imu
/robot1/turtlebot3_joint_state
/robot1/turtlebot3_laserscan
/robot1/waypoint_follower
/robot2/amcl
/robot2/amcl_rclcpp_node
/robot2/bt_navigator
...
/robot2/turtlebot3_imu
/robot2/turtlebot3_joint_state
/robot2/turtlebot3_laserscan
/robot2/waypoint_follower
```

@dohlenaugur, could you please provide the full log of `ros2 launch nav2_bringup multi_tb3_simulation_launch.py 2>&1 | tee run.log` run for more details?
Also, which DDS implementation are you using? Did you tried CycloneDDS as well?

dohlenaugur, 2022-09-02:
[run.log](https://github.com/ros-planning/navigation2/files/9475911/run.log)

I have not changed any DDS settings, so I assume it is eProsima’s Fast DDS. I have not tried anything else yet.

AlexeyMerzlyakov, 2022-09-02:
@dohlenaugur, thank you for the operative respond on this.
Confirming the problem: from the log I could not see that the nodes are not going into "Configuring" stage, e.g. for `map_server`, one can find it only goes to `Creating` stage, but never moves forward to `Configuring`.

I've also noted the same effect locally, both for Foxy binary and Rolling build from sources, when running `multi_tb3_simulation_launch.py`. For `tb3_simulation_launch.py` everything is going flawlessly. The effect confirmed on both CycloneDDS and FastRTPS, so that seems not to be well-known FastDDS problem.
It seems, that for some reasons, lifecycle managers do not switch nodes' states into `Configure` state for multi TB3 simulation.

dohlenaugur, 2022-09-02:
I have experienced with bringup_launch.py, that if I type in 'use_namespace': 'true', I get:

Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 4 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 106, in visit
    if self.__condition is None or self.__condition.evaluate(context):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/condition.py", line 44, in evaluate
    return self._predicate(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/conditions/if_condition.py", line 43, in _predicate_func
    return evaluate_condition_expression(context, self.__predicate_expression)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/conditions/evaluate_condition_expression_impl.py", line 41, in evaluate_condition_expression
    expanded_expression = perform_substitutions(context, expression)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 197, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/substitutions/python_expression.py", line 72, in perform
    return str(eval(perform_substitutions(context, self.expression), {}, math.__dict__))
  File "<string>", line 1, in <module>
NameError: name 'true' is not defined
Task exception was never retrieved
future: <Task finished name='Task-16' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 443, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
[INFO] [sync_slam_toolbox_node-1]: process started with pid [35528]
[INFO] [map_saver_server-2]: process started with pid [35530]
[INFO] [lifecycle_manager-3]: process started with pid [35532]
[INFO] [map_saver_server-2]: sending signal 'SIGINT' to process[map_saver_server-2]
[INFO] [sync_slam_toolbox_node-1]: sending signal 'SIGINT' to process[sync_slam_toolbox_node-1]
[ERROR] [map_saver_server-2]: process has died [pid 35530, exit code -2, cmd '/opt/ros/foxy/lib/nav2_map_server/map_saver_server --ros-args -r __ns:=/robot1 --params-file /tmp/tmp87wrg3_0'].
[ERROR] [sync_slam_toolbox_node-1]: process has died [pid 35528, exit code -2, cmd '/opt/ros/foxy/lib/slam_toolbox/sync_slam_toolbox_node --ros-args -r __node:=slam_toolbox -r __ns:=/robot1 --params-file /opt/ros/foxy/share/slam_toolbox/config/mapper_params_online_sync.yaml --params-file /tmp/launch_params_iaxu9gez'].
[lifecycle_manager-3] [INFO] [1662123700.351001107] [robot1.lifecycle_manager_slam]: Creating
[... cut ...]
[ERROR] [lifecycle_manager-3]: process[lifecycle_manager-3] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [lifecycle_manager-3]: sending signal 'SIGTERM' to process[lifecycle_manager-3]
[ERROR] [lifecycle_manager-3]: process has died [pid 35532, exit code -15, cmd '/opt/ros/foxy/lib/nav2_lifecycle_manager/lifecycle_manager --ros-args -r __node:=lifecycle_manager_slam -r __node:=lifecycl

With 'use_namespace': 'True' it will launch without errors. Is this related?

dohlenaugur, 2022-09-05:
Also with bringup_launch.py, if I launch it like this:

IncludeLaunchDescription(
            PythonLaunchDescriptionSource(os.path.join(get_package_share_directory("nav2_bringup"), 'launch', 'bringup_launch.py')),
            launch_arguments={
                'namespace': namespace,
                'use_namespace': 'True',
                'use_sim_time': 'True',
                'slam': 'True',
                'map': '',
                'params_file':os.path.join(ess_path, 'launch/slam_and_navigation', 'nav2_params.yaml'),
                'default_bt_xml_filename':os.path.join(get_package_share_directory('nav2_bt_navigator'), 'behavior_trees', 'navigate_w_replanning_and_recovery.xml'),
                'autostart': 'True',
            }.items()),

The slam toolbox does not seem to start. I can not see any of its services or nodes.

AlexeyMerzlyakov, 2022-09-09:
The problem is related to that `autostart` parameter is being disabled by default in `multi_tb3_simulation_launch.py`. This worked since very beginning this file was added in https://github.com/ros-planning/navigation2/commit/0fa1a95e3fa528573445cef7f9779646e82412a7#diff-9fe2fa6b8bb2ac393c70324f1f31d596fc56f8740cdb857c26c059331a281753R124 with disabled value. I might suggest to set its value to the `True` by default, as utilized in `tb3_simulation_launch.py`. @SteveMacenski , @orduno, any ideas, why autostart was disabled for multi-robots case?

As a workaround for now, I might suggest to run multi-robot simulation with intentionally enabled `autostart`:
```
ros2 launch nav2_bringup multi_tb3_simulation_launch.py autostart:=True
```

SteveMacenski, 2022-09-09:
Autostart was actually default `False` everywhere and I think that was a change I made when I took over the project because I was sick of the tickets about how things weren't working :laughing: 

We could have autostart as true, though for the multirobot case it might be nice to transition each up separately to distribute the discovery traffic

choobacca-bacca, 2022-09-13:
With the workaround, now everything launches however, another issue arises where we are unable to issue navigation goals to the robots. The error message that pops up is:
```
[rviz2-4] Start navigation
[rviz2-4] [ERROR] [1663042098.784683521] [robot1._]: navigate_to_pose action server is not available. Is the initial pose set?
```
This is despite the initial pose is set up already, as seen below in the screenshot

[image]

SteveMacenski, 2022-09-13:
You may not use the rviz tool for multiple robots. It hardcodes the server IDs and namespaces for the typical default setup. The rviz panel is only for getting started / demos.

choobacca-bacca, 2022-09-13:
Ok understood, thank you for replying, will send the goal via the command line instead.

closed 2022-09-13 by SteveMacenski