ros-navigation/navigation2 #3167: time wasted in _graph.emplace

mangoschorle, 2022-09-02:
https://github.com/ros-planning/navigation2/blob/634d2e3d9b6bde5558c90fe6583d52b3ed66cf55/nav2_smac_planner/src/a_star.cpp#L122
can be improved. 
emplace_back will always construct the value even if it is already in the graph. So if it is in the graph, it is created and immediately afterwards destroyed. (see https://en.cppreference.com/w/cpp/container/unordered_map/emplace)

We could spare unnecessary constructions doing sth like this

 
```
  auto iter = graph_.find(index);
  if (iter == graph_.end()) {
    return &(std::get<0>(graph_.emplace(std::piecewise_construct, std::forward_as_tuple(index),
                                        std::forward_as_tuple(index /*initializer list of node t or NodeT(index)*/)))
                 ->second);
  }
  return &(iter->second);
```

Another thing I don't understand is why  AStar reserves _graph.reserve(100000);
Seems a bit random to me. Why not reserving max_iterations?

SteveMacenski, 2022-09-02:
Please submit a PR, sounds good to me!

> Why not reserving max_iterations?

Frankly, the `max_iterations` didn't exist at the time that was written to optimize performance. But also `max_iterations` isn't the right thing because the graph will contain far more than just the number of iterations of the planner. The iterations of the planner are the number of **visited** nodes in the graph, but theoretically up to `num_prims` number of nodes are **queued** into the graph at each visitation.

mangoschorle, 2022-09-05:
Will do.

SteveMacenski, 2022-09-06:
I’d also make sure in a test that it actually is faster. It wouldn’t surprise me if the immediate destruction due to a pre-existing node was faster than the find implementation- even though either way the hash table’s bucket would need to be found. I suppose it depends on how find is implemented and how the compiler optimizes it. I wouldn’t necessarily assume this is faster so it should be an empirical decision based on benchmarking. 

On the max iterations- something to mention is that when the graph is cleared, the memory block is still allocated. So if there’s a much larger graph size after the first plan, it’ll retain that capacity. This initial reserve is mostly just for the 1st planning instance to give it a reasonable starting size rather than incurring multiple copying while the container is resized

mangoschorle, 2022-09-08:
On a side node before digging into this. Each of the nodes (hybrid, lattice, ..) has a reset() method. This isn't used currently or is it?

mangoschorle, 2022-09-09:
I did some analysis using gperftools. I had around 6% improvement. Quite interesting to see that the unordered_map business takes up 30%+ of the planning times.

Old version emplace
[image]

Find and emplace
[image]

mangoschorle, 2022-09-09:
Now this is super interesting. After seeing how many time was taken up by the unordered_map I went ahead and just for fun replaced it with the first google hit I could find that has some stars on github (https://github.com/martinus/robin-hood-hashing). I didn't look into licensing & so on ... Just for the sake of getting an idea whether one could improve here.

34 % runtime drop.

Time units planning:

**std::unordered_map:**
find&emplace 473.945 time units
emplace (current version) 504.593 time units

**robin_hood::unordered_map:**
find&emplace 313.878 time units
emplace (current version) 339.624 time units

[image]

SteveMacenski, 2022-09-09:
> I did some analysis using gperftools. I had around 6% improvement. Quite interesting to see that the unordered_map business takes up 30%+ of the planning times.

Indeed, this is why I designed the A* implementation the way I did to absolutely remove any lookups in the graph beyond insertion, by using the returned pointers and storing those in `NodeBasic` so that later access was lookup-free. 

That looks like an impressive speed up, definitely worth considering adding! 

> I didn't look into licensing & so on

I just did, this would be totally compatible to include the headers in this project. Though, the headers are kill its test coverage metrics :/ I found https://packages.debian.org/buster/robin-map-dev which are binaries for https://github.com/Tessil/robin-map which claims to implement the same robin hood hashing. Maybe worth a try? If that works, I'd be happy to submit the rosdep keys to the central index so we can move Smac over to it if it gets us a good boost! There's also this library: https://github.com/martinus/unordered_dense same guy, just newer.

mangoschorle, 2022-09-12:
I tried out the unordered_dense map follow-up. It segfaulted but I didn't find the time to further look into this thing.
I suggest to change the emplace to -> find, emplace. In any case 6% free lunch.
Concerning the hash-map, I presonally would also opt for a better map such as round-robin. Although I think that the benefits of it would only really kick in on non-trivial planning requests.

SteveMacenski, 2022-09-12:
Sounds good - can you open the PR?

> Concerning the hash-map, I presonally would also opt for a better map such as round-robin.

I totally agree, this is useful. If you could throw me a branch or a draft PR with that, I can see how I can integrate it as cleanly as possible w.r.t. the dependencies and test coverage metrics.

SteveMacenski, 2022-09-15:
https://github.com/ros-planning/navigation2/pull/3195 for the checking if existing before emplacing.

I wasn't sure why you did the `piecewise_construct` and `forward_as_tuple` stuff here, can you explain your thinking or if my solution is equally as good?

mangoschorle, 2022-09-16:
I just timed both solutions: mine 688031 units yours 689687. That's 0.25% difference. Given noise I would say they are just equally good.

Now for the round robin hashing it really is dead simple. I just copied the header [robin_hood.h](https://github.com/martinus/robin-hood-hashing/blob/master/src/include/robin_hood.h) into my code base and replaced
`typedef std::unordered_map<unsigned int, NodeT> Graph;`

with

`typedef robin_hood::unordered_map<unsigned int, NodeT> Graph;`

other than that its all completely compatible.

I am really sorry to contribute like this and not just hand in a pull request. We do have the possibility to contribute to OS in the form of actual code, but it requires a clearing process, which I am currently familiarizing with. We are looking into this at our end.

SteveMacenski, 2022-09-16:
Understood, thanks for as much as you've done, its been very helpful! Smac Planners are already pretty fast and I'm very happy now that they straight up blow the others out of the water with these changes :wink: 

I'll play around with the robin_hood stuff this afternoon

SteveMacenski, 2022-09-16:
I included the header in the repo `robin_hood.h` and it fails to plan (just hangs) and the server eventually crashes using that header file and replacing the `std` with the `robin_hood` namespace in the graph type. Sure you didn't do anything else? 

Branch: https://github.com/ros-planning/navigation2/tree/robin_hood

SteveMacenski, 2022-09-16:
https://github.com/martinus/unordered_dense/blob/main/include/ankerl/unordered_dense.h

However the other one I pointed to seems to work fine if I include that instead and replace the line with

```
  typedef ankerl::unordered_dense::map<unsigned int, NodeT> Graph;
```

I don't see a significant difference in compute times -- but I only did some extremely quick tests, I'd need to do more to make a real call if that helped or not (the variance was high, but certainly didn't seem like a 30% drop consistently).

mangoschorle, 2022-09-18:
robin_hood works fine on this end. Our largest environments are 600m*700m and all unit tests succeed. My code base in general has diverged quite a bit form the nav2 version (due to some restriction and practices) but in the end, I am emplacing nodes just as you do.

We are compiling g++ (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0.

You are absolutely right this is not good. But it sort of surprises me. In the end it was a naive pick on my side. As said I took the first thing with 1000+ stars, that's it.

> Concerning the hash-map, I presonally would also opt for a better map such as round-robin. Although I think that the benefits of it would only really kick in on non-trivial planning requests.

I am experiencing these 30% drops consistently in all of my _trickier_ planning requests. I don't know if there is much gain before the first rehashing.

SteveMacenski, 2022-09-19:
Got it -- well I would be curious as to the changes you've made if its diverged that much if there are improvements there to the algorithm that can be made here as well! 

`unordered_node_map` does indeed do it - thanks. I'll do some more appropriate benchmarking this afternoon, but I do now see a 15-35% reduction from 2 particular quick test cases averaged over 5 runs. This is worth investigating for real benchmark improvement estimates, but I think ultimately this will be merged even if the low-end estimates are true. 

Thanks for bringing this up!

SteveMacenski, 2022-09-19:
On a set of 100 random plans over a 100x100m map, I see a 25% total decrease in planning time without any change in other metrics, so I think this is definitely both good to be merged and a shockly large total planning improvement for a trivial change!

SteveMacenski, 2022-09-20:
From my deeper experiments, I see between 10-15% on a trivial-ish sandbox environment as part of our benchmarking suite https://github.com/ros-planning/navigation2/tree/main/tools/planner_benchmarking. Far from the 30% reported, but also may very well be that high for longer, more complex paths. Either way 10-15% improvement for a change in data structure is well worth it on its own right (or even half of that!). 

Only issue I'm trying to deal with now is how to exclude the robin hood header file from our testing and linting suite which is reporting a ton or errors and lack of coverage.

SteveMacenski, 2022-09-20:
Ended up just actually linting the robin hood file by hand -- waiting on CI to turn over successfully and will merge. Closing! 

Thanks @mangoschorle for the great work and mentioning this, huge impact for such a small change :smile:

SteveMacenski, 2022-09-22:
I tested these changes on some extremely large maps and see the same 30% numbers you’re seeing! Just giving external confirmation from another application.

SteveMacenski, 2022-09-22:
I uploaded some instructions in the comment at the top of the file for setting up the config files so that things are setup the way the script expects for calling the planners. If you want to do a subset of those, you'll have to remove the planners your not interested in from the `planners` list. For maps, we also in the script specify a map in the variable `map_path`. If you want to use something else, replace that line. 

For launching things, just run Nav2 as you normally would and then run this script in another terminal. Obviously you don't need gazebo/the other servers/etc but that's the lowest overhead way. I have some launch files I use for just launching the planner server / map server / faking out TF frames / and the lifecycle manager to active everything, but I haven't found a good place to dump that yet for public consumption. It has some hardcoded content that are not fit for general use (maps, mostly).

SteveMacenski, 2022-09-22:
On another topic, last night something popped in my head around dinner time, I remembered looking at the partial GDB result you showed above and glanced past the part where the 2D distance heuristic is taking up a bunch of time. Not sure why 2 weeks later my brain decided to bring that to the forefront of my mind, but we do use `hypotf` in that heuristic which is known to be heavier than just doing $\sqrt(x*x + y*y)$ by quite a bit if repeated in the ~tens of thousands of times. I'll test that later this week and maybe also get a few percentage points back.