ros-navigation/navigation2 #3171: Collision monitor not detecting collision with pure rotation in approach model

automech-rb, 2022-09-05:
## Bug report

**Required Info:**

- Operating System:
  - Docker 22.04 Ubuntu
- ROS2 Version:
  - rolling
- Version or commit hash:
  - 7813feefd4ad4d80064af82b4ac659f52a68283e
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

I am using a rectangular robot with two laser scanner. Config as follows:
```yaml
collision_monitor:
  ros__parameters:
    base_frame_id: "base_footprint"
    odom_frame_id: "odom"
    cmd_vel_in_topic: "cmd_vel_raw"
    cmd_vel_out_topic: "cmd_vel"
    transform_tolerance: 0.5
    source_timeout: 5.0
    stop_pub_timeout: 2.0
    polygons: ["FootprintApproach"]
    FootprintApproach:
      type: "polygon"
      action_type: "approach"
      footprint_topic: "/local_costmap/published_footprint"
      time_before_collision: 2.0
      simulation_time_step: 0.02
      max_points: 0
      visualize: True
    observation_sources: ["scan_front", "scan_back"]
    scan_front:
      type: "scan"
      topic: "/scan/front"
    scan_back:
      type: "scan"
      topic: "/scan/back"
```
I am publishing 0.3 rad/sec twist to cmd_vel_raw topic in both gif below.

#### Expected behavior
As `time_before_collision` is 2.0 sec, it should detect collision for obstacles under 34 degree in rotation wrt current pose. It should slow down to avoid collsion.
#### Actual behavior
Collision is not detected (in the log), and robot continued to rotate and collide with obstacle (In gif, it is a medium sized block)
In first gif below, first I try to collide head on, but the collision is detected and robot slows down. As next I try to rotate in place, but robot collides with block (log don't show any collision warning).
[image]

In this gif, same behavior but with different angle of approach.
[image]

#### Additional information
Approach model works perfectly in head on collision but not when rotating in place. I went through the source code but can not find anything strange which might be causing this. I tried longer `time_before_collision` and using more finer time step, but still same result in rotation.

AlexeyMerzlyakov, 2022-09-05:
I've tried to reproduce locally the result with similar square footprint, but there was no problem appeared on my side: robot's rotation is slowing down while approaching the obstacle detected by laser scanner.
@automech-rb, are front and back laser scanners in your configuration have 360 degrees FoV? If laser scanner have the restricted field of view, it might miss side obstacles during rotation, while at linear movement it does not.

Another problem might be huge time difference (more than `source_timeout`) between laser scanner and CM node; or where the laser scanner is being published in another time, like using/non-using simulation time while running under Gazebo.

Could you please provide the log CM launch to check are there any abnormalities or not?

automech-rb, 2022-09-05:
Hello @AlexeyMerzlyakov 
Thanks for quick reply.

I am adding a video of my demo:

https://user-images.githubusercontent.com/24780015/188480499-ce10615b-452c-49a4-81e2-5f522f9c6926.mp4

I get your point on side obstacles but from the rviz2 screen, I feel like these points should be detected for collision.
I have two laser (one on front right and other diagonally opposite) which are both covered with local costmap footprint (can be seen in rviz2 screen above).

I am using Gazebo sim for testing, and collision monitor node is currently using clock topic from gazebo.

Here are the logs:
```
cat collision_monitor_461_1662392113924.log 
[INFO] [1662392113.941850481] [collision_monitor]: 
        collision_monitor lifecycle node launched. 
        Waiting on external lifecycle transitions to activate
        See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1662392113.988723107] [collision_monitor]: Configuring
[INFO] [1662392113.993117403] [collision_monitor]: [PolygonApproach]: Creating Polygon
[INFO] [1662392113.994998286] [collision_monitor]: [scan_front]: Creating Scan
[INFO] [1662392113.999424054] [collision_monitor]: Activating
[INFO] [1662392113.999497100] [collision_monitor]: Creating bond (collision_monitor) to lifecycle manager.
[INFO] [1662392286.670966866] [collision_monitor]: Robot to approach for 2.000000 seconds away from collision
[INFO] [1662392290.670891394] [collision_monitor]: Robot to continue normal operation
[INFO] [1662392320.460334719] [collision_monitor]: Robot to approach for 2.000000 seconds away from collision
[INFO] [1662392320.660346000] [collision_monitor]: Robot to continue normal operation
[INFO] [1662392373.481029375] [collision_monitor]: Robot to approach for 2.000000 seconds away from collision
[INFO] [1662392376.491102420] [collision_monitor]: Robot to continue normal operation
[INFO] [1662392381.279321552] [rclcpp]: signal_handler(signum=2)
```

AlexeyMerzlyakov, 2022-09-06:
@automech-rb, thank you for the problem details and video provided! I've reproduced the problem on my side. This issue appeared due to incorrect velocities comparison in case when linear speed is equal to 0. Work on the fix is in progress. I'll keep you updated with latest news.

automech-rb, 2022-09-06:
looking forward for the fix @AlexeyMerzlyakov 
I also ended up writing a new collision monitor node. I feel that current nav2 collision monitor can be have a performance improvement by skipping sources points which are far ( > max speed * sim_time) and also transforming the polygon instead of all the sensor source data. What do you think ?

SteveMacenski, 2022-09-06:
I'd love to see a proposal / PR!

AlexeyMerzlyakov, 2022-09-07:
PR #3177 should fix the problem.

Regarding the restriction optimization: yes this will work. Since the robot could be rotated, we should count its geometry. From the first approximation, it seems the maximum cut-off radius should be `cmd_vel_in * time_before_collision + shape_diameter` where the `shape_diameter = max(dist(poly_[i], poly_[j]) : i,j polygon iterators)` for polygons and `shape_diameter = 2 * radius_` for circles.
The restriction optimization routine might be added to the `process()` right after getting the data from sources.

automech-rb, 2022-09-07:
I tested the pr, works as expected. Thanks!

Also,  I think its efficient to filter data from `sources` instead of adding all points first to `std::vector<Point> & data`. Maybe somewhere here:

https://github.com/ros-planning/navigation2/blob/42d0bbd026f84fb6447982f4448ea5c51f6fdebb/nav2_collision_monitor/src/range.cpp#L108

SteveMacenski, 2022-09-08:
We made a small change to it, can you test again?

automech-rb, 2022-09-09:
Did some quick tests on simulator. Working as expected with latest commits.

closed 2022-09-09 by SteveMacenski