ros-navigation/navigation2 #3179: nav2_bringup tb3_simulation_launch.py does not launch, invalid tf between map and odom

choobacca-bacca, 2022-09-08:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- ROS2 Version:
  - Galactic Binaries
- Version or commit hash:
```
ros-galactic-nav2-amcl	1.0.12-1focal.20220730.031414
ros-galactic-nav2-behavior-tree	1.0.12-1focal.20220730.031414
ros-galactic-nav2-bringup	1.0.12-1focal.20220730.081550
ros-galactic-nav2-bt-navigator	1.0.12-1focal.20220730.034950
ros-galactic-nav2-common	1.0.12-1focal.20220730.020404
ros-galactic-nav2-controller	1.0.12-1focal.20220730.035022
ros-galactic-nav2-core	1.0.12-1focal.20220730.034047
ros-galactic-nav2-costmap-2d	1.0.12-1focal.20220730.032116
ros-galactic-nav2-dwb-controller	1.0.12-1focal.20220730.072836
ros-galactic-nav2-gazebo-spawner	1.0.12-1focal.20220730.003012
ros-galactic-nav2-lifecycle-manager	1.0.12-1focal.20220730.031454
ros-galactic-nav2-map-server	1.0.12-1focal.20220730.031510
ros-galactic-nav2-msgs	1.0.12-1focal.20220730.021355
ros-galactic-nav2-navfn-planner	1.0.12-1focal.20220730.034842
ros-galactic-nav2-planner	1.0.12-1focal.20220730.034904
ros-galactic-nav2-recoveries	1.0.12-1focal.20220730.035128
ros-galactic-nav2-regulated-pure-pursuit-controller	1.0.12-1focal.20220730.035131
ros-galactic-nav2-rotation-shim-controller	1.0.12-1focal.20220730.035831
ros-galactic-nav2-rviz-plugins	1.0.12-1focal.20220730.075921
ros-galactic-nav2-smac-planner	1.0.12-1focal.20220730.034906
ros-galactic-nav2-theta-star-planner	1.0.12-1focal.20220730.034933
ros-galactic-nav2-util	1.0.12-1focal.20220730.030758
ros-galactic-nav2-voxel-grid	1.0.12-1focal.20220730.023307
ros-galactic-nav2-waypoint-follower	1.0.12-1focal.20220730.035204
```

- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
```
source /opt/ros/galactic/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/galactic/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False
```

#### Expected behavior
tb3_simulation launches with nav stack. Able to provide nav_goals for tb3 to move to.

#### Actual behavior
Gazebo simulation launches with TB3 in.
Rviz launches with map displayed. However, there is no tf between map to tb3.

```
[controller_server-8] [INFO] [1662631106.512542668] [local_costmap.local_costmap]: start
[controller_server-8] [INFO] [1662631106.512969245] [controller_server]: Creating bond (controller_server) to lifecycle manager.
[lifecycle_manager-13] [INFO] [1662631106.615111674] [lifecycle_manager_navigation]: Server controller_server connected with bond.
[lifecycle_manager-13] [INFO] [1662631106.615157438] [lifecycle_manager_navigation]: Activating planner_server
[planner_server-9] [INFO] [1662631106.615344608] [planner_server]: Activating
[planner_server-9] [INFO] [1662631106.615369299] [global_costmap.global_costmap]: Activating
[planner_server-9] [INFO] [1662631106.615379404] [global_costmap.global_costmap]: Checking transform
[planner_server-9] [INFO] [1662631106.615403557] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[amcl-6] [INFO] [1662631106.644153788] [amcl]: createLaserObject
[planner_server-9] [INFO] [1662631107.115458679] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[planner_server-9] [INFO] [1662631107.615484588] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[planner_server-9] [INFO] [1662631108.115467293] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-4] [INFO] [1662631108.520637352] [rviz]: Message Filter dropping message: frame 'odom' at time 0.056 for reason 'discarding message because the queue is full'
[planner_server-9] [INFO] [1662631108.615451553] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-4] [INFO] [1662631108.680960137] [rviz]: Message Filter dropping message: frame 'base_scan' at time 0.200 for reason 'discarding message because the queue is full'
[amcl-6] [WARN] [1662631108.685797347] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[rviz2-4] [INFO] [1662631108.744790011] [rviz]: Message Filter dropping message: frame 'odom' at time 0.236 for reason 'discarding message because the queue is full'
[rviz2-4] [INFO] [1662631108.904950477] [rviz]: Message Filter dropping message: frame 'base_scan' at time 0.400 for reason 'discarding message because the queue is full'
[rviz2-4] [INFO] [1662631108.941629054] [rviz]: Message Filter dropping message: frame 'odom' at time 0.446 for reason 'discarding message because the queue is full'
[planner_server-9] [INFO] [1662631109.115481154] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist

```

SteveMacenski, 2022-09-08:
Please read the getting started guide. If you still have questions please ask how to ROS answers

closed 2022-09-08 by SteveMacenski

SongQ-Young, 2022-11-21:
sudo apt install ros-<ros2-distro>-turtlebot3*

kikass13, 2023-01-06:
still broken