ros-navigation/navigation2 #3180: Sending a BT for a goal different from the default one makes BT crash

bernatgaston, 2022-09-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.1
- ROS2 Version:
  - Humble
- Version or commit hash:
ros-humble-navigation2	1.1.0-2jammy.20220808.172821

- DDS implementation:
  - cycloneDDS

#### Steps to reproduce issue
```
1. Set a default behaviour tree to the nav2
2. Send a goal with the behavior_tree option of the NavigateToPose set to a BT (different from the default one)

```

#### Expected behavior
Use alterative BT for a specific goal

#### Actual behavior
Crash

#### Additional information

After calling NaviagtionToPose:

[ros2-14] [ros2-1] [bt_navigator-7] terminate called without an active exception
[ros2-14] [me00_navigation-2] [ERROR] [1662652147.222046948] [me00_1.Navigation]: No service to get_state for bt_navigator
[ros2-14] [ros2-1] [ERROR] [bt_navigator-7]: process has died [pid 103619, exit code -6, cmd '/opt/ros/humble/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator -r __ns:=/me00_1 --params-file /tmp/tmp6xe56kat -r /tf:=tf -r /tf_static:=tf_static -r /map:=slam_map -r cmd_vel:=Navigation/cmd_vel -r /slam_toolbox/clear_changes:=slam_toolbox/clear_changes -r /slam_toolbox/dynamic_map:=slam_toolbox/dynamic_map -r /slam_toolbox/manual_loop_closure:=slam_toolbox/manual_loop_closure -r /slam_toolbox/pause_new_measurements:=slam_toolbox/pause_new_measurements -r /slam_toolbox/save_map:=slam_toolbox/save_map -r /slam_toolbox/serialize_map:=slam_toolbox/serialize_map -r /slam_toolbox/toggle_interactive_mode:=slam_toolbox/toggle_interactive_mode'].

SteveMacenski, 2022-09-08:
What behavior tree? Could very well be your tree is invalid. 

Also please get a trace back, you can see a tutorial how on our website navigation.ros.org.

bernatgaston, 2022-09-09:
My fault. It seems it works when the same experiment is done with basic BTs. Thanks and sorry for that

closed 2022-09-09 by bernatgaston