ros-navigation/navigation2 #3185: Use different costmap layers for different controller/planner plugins

tonynajjar, 2022-09-12:
## Feature request

#### Feature description
Currently all planner plugins use the same costmap created in `planner_server.cpp`. I have two planner plugins and I want them to use different costmap layers (I guess this means different costmaps altogether). Conceptually, is this something that should be allowed?

#### Implementation considerations

AlexeyMerzlyakov, 2022-09-12:
From the first glance, you could make two global planners, configure both and use different set of costmaps, one per each planner. The only restriction might be, if there are two planners are allowed or not allowed to run simultaneously. Anyway, I think, this is rather a good topic for a ROS answers, than feature request.

tonynajjar, 2022-09-12:
> you could make two global planners

With "global planner" you mean planner_server node or planner plugin?
If you mean planner plugin, as far as I know, we can't configure which costmap each planner uses
If you mean planner_server, I thought about that but considered it as a pretty hacky solution; I'd need to launch the planners in two different namespaces for the costmap topics not to conflict

> Anyway, I think, this is rather a good topic for a ROS answers, than feature request.

Is it? I'm not asking how to achieve that, I'm requesting a cleaner way of enabling this. However if you and @SteveMacenski recommend launching two planner servers as the way to go then I can work with that.

SteveMacenski, 2022-09-12:
At the point you have a different set of costmap plugins for different algorithms, it probably makes sense for you to actually just have 2+ `planner_server` instances in general. That way you can maintain separate costmap configurations with different subsets of planners. And this is trivial to do, just namespace them at launch and update your BT to call the action server with the appropriate namespace (the BT action nodes have an internal port that can be overrided from its default).

tonynajjar, 2022-09-12:
Thanks a lot for the detailed explanation, I'll go with a second planner server. I'll close this now as there is nothing more to look into.

closed 2022-09-12 by tonynajjar