ros-navigation/navigation2 #3186: Robot stuck forever once in costmap filter

MrOCW, 2022-09-13:
## Feature request
Allow robot to backup or spin even if in local costmap filter
#### Feature description
Robot might overshoot the controller (TEB in my case) and have a wider turning radius than intended of the controller. I have a costmap filter on curbs and the robot occasionally drives close to them. When the footprint touches the costmap filter, it is stuck forever until I teleop it away from the area and FollowPath continues as per normal. Backup returns collision so it doesnt drive backwards.

#### Implementation considerations
Maybe deactivate the costmap filter around the robot temporarily similar to footprint clearing?

AlexeyMerzlyakov, 2022-09-13:
The effect appears if enable the KeepoutFilter on local costamp. The keepout zone is being treated here as an obstacle and DriveOnHeading (parent of Backup) behavior won't allow robot to exit accidently entered area. However, I am not sure, that the keepout zones could be clearable for that. They have been marked as always non-clearable layers, in order to not allow robot to do something inside a keepout zone, due to safety reasons: only operator (using teleop or something similar) could move robot out from such zone. It is not difficult to make a keepout zone to be clearable e.g. by parameter. However, I am not sure, whether we really need to change this behavior, because of intentional concept of safety, e.g. while using massive production robots in the facilities.

In your case, where the robot is occasionally go inside a local keepout zone could be eliminated e.g. by introducing an additional inflation layer around it, like I've mentioned in the https://github.com/ros-planning/navigation2/issues/3102#issuecomment-1225494691. Doesn't it resolve the situation?

MrOCW, 2022-09-13:
looks great! let me give that a try and get back to you after further testing with that config thanks! :)

SteveMacenski, 2022-09-14:
It would not be sensible to just set the layer as clearable, because it really isn't. You'd clear it and immediately repopulate it the next iteration which would not, in effect, change anything but take more CPU time. 

Adding inflation seems like the move or in stuck conditions being able to temporarily toggle that layer on/off which is possible with the `enabled` parameter.

MrOCW, 2022-09-15:
got it thanks!

closed 2022-09-15 by MrOCW