ros-navigation/navigation2 #3191: Galactic branch's Spin.Action missing duration definition

madgrizzle, 2022-09-15:
## Bug report
- Operating System:
Ubuntu 20.04
- ROS2 Version:
Galactic
- Version or commit hash:
ros-galactic-navigation2  1.08-1focal.20220203.051427
- DDS implementation:
FastRTPS

#### Steps to reproduce issue
When using nav2_simple_commander spin command, a crash occurs and error in line 187 of robot_navigator.py is reported that the "Spin Goal" object has no attribute 'time_allowance'.   

#### Additional information
robot_navigator.py imports Spin from nav2_msgs.action and attempts to set the time_allowance, but Spin.action's goal definition in the galactic branch only defines float32 target_yaw and does not include builtin_interfaces/Duration time_allowance found in the definition of the main branch. 

https://github.com/ros-planning/navigation2/blob/3df00a74b79b4581599a65d3dbf60a9568cc5b5b/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L186

**galactic branch of Spin.action:**
https://github.com/ros-planning/navigation2/blob/3df00a74b79b4581599a65d3dbf60a9568cc5b5b/nav2_msgs/action/Spin.action#L1-L7

**main branch of Spin.action:**
https://github.com/ros-planning/navigation2/blob/34ecee20a1913a71bc345b0438a0081b73159458/nav2_msgs/action/Spin.action#L1-L9

SteveMacenski, 2022-09-15:
We cannot change the message definition after the release of a distribution, since that is not ABI compatible for existing users.

It is an oversight that that was backported, so instead of changing the message that portion of the simple API should just be removed

SteveMacenski, 2022-09-15:
https://github.com/ros-planning/navigation2/pull/3193 Can you verify this resolves your issue?

madgrizzle, 2022-09-15:
understood.  thanks.  I've been working from the release package and not building from source, but will give it a go and let you know.

SteveMacenski, 2022-09-15:
Understandable - sorry about that. We don't have unit tests set up in Python3 (yet) so that isn't covered. Interpreted languages :eyeroll: amiright?!

madgrizzle, 2022-09-15:
hmmm.. is this related?  I got this error message right after the robot physically spun. 
[image]

madgrizzle, 2022-09-15:
Seems to work, my error message related to how I 'spun' the node.  Thanks for the quick fix, now I can move forward with my mid-life crisis robot build.

closed 2022-09-15 by madgrizzle