ros-navigation/navigation2 #3196: Feature: Subscribe to polygons in collision monitor

tonynajjar, 2022-09-15:
Since the collision monitor is supposed to replace a safety scanner, I think it would make sense to emulate field set switching by subscribing to polygons.

Typically fieldsets would change according to the spee; one could write a separate node that subscribes to speed and publishes which fieldset/polygon should be active

SteveMacenski, 2022-09-15:
> is supposed to replace a safety scanner

It is **not** supposed to replace a safety scanner, to be clear. It is a collision monitor which is not certified as a replacement for any safety equipment. 

I'm not entirely clear what you're asking for. The polygons used are published that can be subscribed to. Are you asking for the ability to dynamically change the polygons?

tonynajjar, 2022-09-15:
Gotcha, wrong wording, I did see the discussion about that on Discourse.

Anyway, yes I mean to dynamically change polygons

SteveMacenski, 2022-09-15:
Sure, yeah, I think that sounds reasonable. Your choices would be via dynamic parameters, topics, or services. What do you think there?

tonynajjar, 2022-09-15:
> Edit: or do you mean disabling/enabling several different pre-established polygons? Services there would make alot of sense.

Mmm, no I was actually thinking new unconfigured polygons for greater flexibility.

> Your choices would be via dynamic parameters, topics, or services

Yeah good question. I would think topics because of the potentially high frequency of change (e.g if the switching is dependent on speed and steering angle). As far as I know topics would offer the least latency.

vinnnyr, 2022-09-16:
+1 for topics. We too have a custom variant of this feature living elsewhere in the system, and like stated @tonynajjar  stated here:

> Typically fieldsets would change according to speed/steering angle.

we change the "zones" based on the current speed and current steering angle of our robot, maybe at a rate near 10hz. I don't know if it makes sense hitting a service at quite that rate.

SteveMacenski, 2022-09-16:
Topics seems reasonable to me :+1:

AlexeyMerzlyakov, 2022-09-19:
Good. The major thing here was agreed:
The current functionality of polygons in the Collision Monitor to be enhanced by subscribing to the topics:
the similar way as `APPROACH` model utilizes. Current functionality with strictly-defined polygon will be intact.
This could be reached e.g. by adding a new `polygon_topic` parameter. If this parameter is not set, using `points` array with polygon vertexes. Otherwise - subscribing to the polygon topic.

I'll could add this functionality in next CM update.

closed 2022-11-17 by SteveMacenski

tonynajjar, 2022-11-18:
Thank you very much!

doisyg, 2022-12-05:
Just to state that we just tested this new feature (mainly @kaichie) and that is exactly what we were looking for.

mbed92, 2023-07-26:
Okay, does that functionality still work in Humble? I need that, but I can not get it to work. param `polygon_topic` or`polygon_sub_topic` do not exist in collision_monitor.

mbed92, 2023-07-26:
Does that functionality work on Humble? I need something like that to dynamically change polygons, but I cannot get it to work - params `polygon_sub_topic' (and `polygon_topic`) simply do not exist. I'm using the newest release 1.1.8.