ros-navigation/navigation2 #3200: [RPP] Parameter to enable/disable collision detection

fantalukas, 2022-09-19:
----
## Feature request

Hey, I would like to propose a new parameter for RPP. The parameter would enable/disable collision detection by RPP.

#### Feature description

In our application, we are using RPP controller together with its obstacle detection. This obstacle detection was our main too, but we were limited by its implementation. For example, the detection distance is limited to carrot distance and there is no threshold for number of points that must be detected inside the footprint of the robot.
I am aware that this is completely wanted because in general there is no need for RPP obstacles detection to be more robust as the planner server is able to create a collision-free path.

However, we are using our fixed paths so we heavily rely on obstacle detection on controller level.
Since the nav2_collision_monitor is implemented, we started using this more robust module to detect obstacles and recently found out that the obstacle detection by RPP is redundant. However, it is not possible to disable the RPP obstacle detection completely.

Moreover, when we somehow get points inside the robot (it is caused by lidars and wrong reflection of lidar's rays from some surfaces) the RPP detects these points as an obstacle. It is obviously correct behavior but unwanted in our case.
Unfortunately, I am not able to solve this problem in RPP collision detection. I tried to set parameter _max_allowed_time_to_collision_up_to_carrot_  to minimum but the the obstacle is always detected inside the robot at least once.

My solution, that works perfectly and that I suggest here in this issue, is to add a parameter to enable/disable obstacle detection by RPP.

Maybe I missed something in collision detection by RPP that could solve the problem more effectively.
If so, please let me know.

#### Implementation considerations

Here is the commit in my forked repository of navigation2  that implements this parameter: https://github.com/fantalukas/navigation2/commit/f6ce77fde9f67cab693db4e60acce34b5b5cd72c

SteveMacenski, 2022-09-19:
> when we somehow get points inside the robot

The clear under footprint parameter should take care of that for you

But either way, sure a parameter here could make sense. Please submit a PR + a PR for documentation to https://navigation.ros.org/configuration/packages/configuring-regulated-pp.html to add the parameter to the configuration guide and I'll merge

fantalukas, 2022-09-20:
Great, I am really glad that the feature seems to be useful. I will submit PR + PR for documentation as soon as possible.

Thank you for proposing another solution for my problem and for very quick response.

closed 2022-09-21 by SteveMacenski