ros-navigation/navigation2 #3205: Controller server loop-rate distorts controling times

SteveMacenski, 2022-09-20:
We found in development of MPPI and while I was benchmarking controllers for another project that the controller server will report that it missed the loop rate when in fact the controller is well capable of running _much_ faster than the rates reported.

An example, DWB can run at 250hz, yet running at 30hz at times will cause the error on occasion when each iteration takes a mere 3ms, typically.

Look into why this is the case and come with a resolution so that we can use the algorithms' full capabilities. I find that if I simply remove the loop sleep that we get it back, so maybe find another way to throttle the control loop? 

@AlexeyMerzlyakov this would be a great medium sized, high impact project if you were interested

afifswaidan, 2023-01-06:
Hello,
Can I get more details about which scripts are having this issue so I can also have a look?

SteveMacenski, 2023-01-06:
nothing overly fancy. I was just noticing DWB complaining about missing its deadline when pushing to 30-50hz on my laptop and grew suspicious of that fact when I was working on a much heavier controller (#3350) at faster rates. I added prints in the controller server of a timer before and after the actual trajectory planner plugins times and noticed that it was running fast enough but the controller server reported that it couldn't keep up with the loop time.

SteveMacenski, 2023-10-28:
I had a great conversation with Jan from Bosch at ROSCon that gave me a good idea on this! We can set the thread priority for the controller server's algorithm execution to be elevated relative to the rest of the stack so that we hit these timers more accurately. We don't mind delaying other systems a few milliseconds (global planning, etc) nearly as much as this thread.

SteveMacenski, 2023-10-28:
https://github.com/ros-planning/navigation2/pull/3914  resolves

closed 2023-10-28 by SteveMacenski