ros-navigation/navigation2 #3208: Depth Image Costmap Plugin

vinnnyr, 2022-09-21:
## Depth Image Costmap Plugin

#### Feature description
Now that there are a few more depth cameras on the market, as well as popular monocular to depth models published, it would be nice to have a depth image populate a costmap as a plugin.

From what I understand, the current workflow is convert a depth image to a pointcloud2, and then use an existing costmap plugin to convert to a  pointcloud2 , and then use one of the existing costmap plugins to populate the costmap. 
It would be nice if didn't have to do the conversion, and have a costmap plugin subscribe to the depth image directly. 

#### Implementation considerations
Not sure.

SteveMacenski, 2022-09-21:
>  the current workflow is convert a depth image to a pointcloud2, and then use an existing costmap plugin to convert to a pointcloud2 , and then use one of the existing costmap plugins to populate the costmap.

Why 2 pointcloud conversions? You should only need to convert your depth image into a pointcloud2 and then publish that for the costmap layer to process.

There are many types of embedding of depth information in image formats, so while we could support it here, there would need to be more information give about how to interpret the data. Further, its long been established that depth cameras publish a depth image as well as a pointcloud, so this typically isn't really an issue.

Either way, someone will need to convert the depth image into a pointcloud2 - its more modular to have it done outside of the costmap layer ('small, sharp tools' and all that jazz)

vinnnyr, 2022-09-21:
> From what I understand, the current workflow is convert a depth image to a pointcloud2, and then use an existing costmap plugin to convert to a pointcloud2 , and then use one of the existing costmap plugins to populate the costmap.

Ah definitely a typo. 

Should read:

> the current workflow is convert a depth image to a pointcloud2, and then use one of the existing costmap plugins to populate the costmap.

SteveMacenski, 2022-09-21:
So tl;dr of my opinion here: I think the current state of things are reasonable and the major motivating reason for a new layer would be if we wanted to treat that data differently.

vinnnyr, 2022-09-21:
ok you bring up a fair counter argument.

I guess I was just trying to take into account the expense to re-publish a pointcloud where the depth image contains the same data as the point cloud, but maybe I should try this workflow for a while to make sure this is something I actually want before prematurely jumping to the solution.

SteveMacenski, 2022-09-21:
Keep in mind if you have them as components in the same process, its pretty low overhead. But really, your sensor driver should be producing this. The NN is a different story, true. But if that's your model exposed to ROS, then you could always publish the pointcloud as well.

closed 2022-09-21 by SteveMacenski