ros-navigation/navigation2 #3209: Nav2 build failure: Colcon failied to build at Nav2_util

AshinTheAnvil, 2022-09-21:
**Hi I am facing an issue with nav2 building using colcon build. I use a VMware machine to run ubuntu 22(Jammy) with Ros 2 Rolling, installed using Binary install. I face an issue with nav2_util at build stage. The following error throws up;**

`error: ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = lifecycle_msgs::srv::GetState; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<lifecycle_msgs::srv::GetState> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’ is deprecated: use rclcpp::QoS instead of rmw_qos_profile_t [-Werror=deprecated-declarations]

47 | client_ = node_->create_client<servicet>( | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^ 48 | service_name, | ~~~~~~~~~~~~~
49 | rmw_qos_profile_services_default, | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
50 | callback_group_); | ~~~~~~~~~~~~~~~~
In file included from /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/node.hpp:1594, from /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28, from /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/executors.hpp:22, from /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/rclcpp.hpp:155, from /home/ashin/nav2_ws1/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20, from /home/ashin/nav2_ws1/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24, from /home/ashin/nav2_ws1/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15: /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/node_impl.hpp:157:1: note: declared here 157 | Node::create_client( | ^~~~ cc1plus: all warnings being treated as errors

gmake[2]: * [src/CMakeFiles/nav2_util_core.dir/build.make:104: src/CMakeFiles/nav2_util_core.dir/lifecycle_service_client.cpp.o] Error 1

gmake[2]: * Waiting for unfinished jobs....

gmake[1]: * [CMakeFiles/Makefile2:210: src/CMakeFiles/nav2_util_core.dir/all] Error 2

gmake[1]: * Waiting for unfinished jobs....

gmake: [Makefile:146: all] Error 2

Failed <<< nav2_util [13.7s, exited with code 2]

Summary: 6 packages finished [17.0s] 1 package failed: nav2_util 2 packages had stderr output: nav2_simple_commander nav2_util 30 packages not processed`

_How Can resolve this issue? please help me with this process as I am a beginner in Ros2. Thank you all_

SteveMacenski, 2022-09-21:
I don't see any issues in CI or my daily builds locally - are you sure your rolling install is up to date? 

This is probably better suited for ROS Answers.

SteveMacenski, 2022-09-21:
Whoops, wrong button

AshinTheAnvil, 2022-09-21:
> 

Okay thanks for your help.

SteveMacenski, 2022-09-21:
I did a clean build locally just this morning and didn't see any issues. It looks like you built rolling locally, its likely that that's not in sync with the current state, perhaps?

AshinTheAnvil, 2022-09-21:
> I did a clean build locally just this morning and didn't see any issues. It looks like you built rolling locally, its likely that that's not in sync with the current state, perhaps?

Thanks for your help Steve.
I will try to reinstall and do as you had suggested.

closed 2022-09-21 by AshinTheAnvil