ros-navigation/navigation2 #3212: nav2_utils build error

muttistefano, 2022-09-22:
Hi all,
i am building ros rolling from source on ubuntu 18.
I bumped into this building error and i cannot tell if im using a wrong branch or what.
Currently i am on the "main" branch of nav2.
The error is the following:

```
In file included from /home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24:0,
                 from /home/pauli/ros2_rolling/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/service_client.hpp: In instantiation of ‘nav2_util::ServiceClient<ServiceT>::ServiceClient(const string&, const SharedPtr&) [with ServiceT = lifecycle_msgs::srv::ChangeState; std::__cxx11::string = std::__cxx11::basic_string<char>; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>]’:
/home/pauli/ros2_rolling/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:36:55:   required from here
/home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:47:13: error: ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = lifecycle_msgs::srv::ChangeState; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<lifecycle_msgs::srv::ChangeState> >; std::__cxx11::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’ is deprecated: use rclcpp::QoS instead of rmw_qos_profile_t [-Werror=deprecated-declarations]
     client_ = node_->create_client<ServiceT>(
In file included from /home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/node.hpp:1594:0,
                 from /home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/executors.hpp:22,
                 from /home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /home/pauli/ros2_rolling/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/node_impl.hpp:157:1: note: declared here
 Node::create_client(
 ^~~~
In file included from /home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24:0,
                 from /home/pauli/ros2_rolling/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/service_client.hpp: In instantiation of ‘nav2_util::ServiceClient<ServiceT>::ServiceClient(const string&, const SharedPtr&) [with ServiceT = lifecycle_msgs::srv::GetState; std::__cxx11::string = std::__cxx11::basic_string<char>; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>]’:
/home/pauli/ros2_rolling/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:36:55:   required from here
/home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:47:13: error: ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = lifecycle_msgs::srv::GetState; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<lifecycle_msgs::srv::GetState> >; std::__cxx11::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’ is deprecated: use rclcpp::QoS instead of rmw_qos_profile_t [-Werror=deprecated-declarations]
     client_ = node_->create_client<ServiceT>(
In file included from /home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/node.hpp:1594:0,
                 from /home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/executors.hpp:22,
                 from /home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /home/pauli/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /home/pauli/ros2_rolling/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/home/pauli/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp/node_impl.hpp:157:1: note: declared here
 Node::create_client(

```

Thanks

vinnnyr, 2022-09-22:
Similar ros-answers question I just saw this morning. Seems to be the same issue :thinking: https://answers.ros.org/question/406790/nav2-build-failure-colcon-failied-to-build-at-nav2_util/

SteveMacenski, 2022-09-22:
This seems odd - I'm working on rolling right now and I see no issues on 22.04 where current rolling is targeted. I just did an update and clean rebuild of utils and had no issues. Are you sure that your local rolling build is up to date?

muttistefano, 2022-09-22:
everything seems up to date.
Im not on 22.04, im on ubuntu 18

SteveMacenski, 2022-09-22:
Rolling/main on Nav2 targets 22.04, we offer no support for out of supported distributions. 

With that said, if Rolling built fine, then I don't see why anything in Nav2 would give you a problem. When did you do your rolling source build?

AshinTheAnvil, 2022-09-22:
Hi even Faced the same issue the other day.
StillI couldnot resolve it..

SteveMacenski, 2022-09-22:
Did you try my suggestion?

AshinTheAnvil, 2022-09-22:
> Did you try my suggestion?

I am about to start with it and surely will get back to you .

AshinTheAnvil, 2022-09-22:
> Did you try my suggestion?
_Hi Steve,
Now I face a different error as follows,_

`/home/ashin/nav2_ws/src/navigation2/nav2_lifecycle_manager/src/lifecycle_manager.cpp:68:58: error: ‘typename rclcpp::Service<ServiceT>::SharedPtr rclcpp::Node::create_service(const string&, CallbackT&&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = std_srvs::srv::Trigger; CallbackT = std::_Bind<void (nav2_lifecycle_manager::LifecycleManager::*(nav2_lifecycle_manager::LifecycleManager*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>))(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >)>; typename rclcpp::Service<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Service<std_srvs::srv::Trigger> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’ is deprecated: use rclcpp::QoS instead of rmw_qos_profile_t [-Werror=deprecated-declarations]`

   68 |   is_active_srv_ = create_service<std_srvs::srv::Trigger>(
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
   69 |     get_name() + std::string("/is_active"),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~               
   70 |     std::bind(&LifecycleManager::isActiveCallback, this, _1, _2, _3),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   71 |     rmw_qos_profile_services_default,
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                     
   72 |     callback_group_);
      |     ~~~~~~~~~~~~~~~~                                      
In file included from /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/node.hpp:1594,
                 from /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/executors.hpp:22,
                 from /home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/ashin/nav2_ws/install/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /home/ashin/nav2_ws/install/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /home/ashin/nav2_ws/src/navigation2/nav2_lifecycle_manager/include/nav2_lifecycle_manager/lifecycle_manager.hpp:26,
                 from /home/ashin/nav2_ws/src/navigation2/nav2_lifecycle_manager/src/lifecycle_manager.cpp:16:
/home/ashin/ros2_rolling/ros2-linux/include/rclcpp/rclcpp/node_impl.hpp:190:1: note: declared here
  190 | Node::create_service(
     
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/nav2_lifecycle_manager_core.dir/build.make:76: CMakeFiles/nav2_lifecycle_manager_core.dir/src/lifecycle_manager.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:176: CMakeFiles/nav2_lifecycle_manager_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

Failed   <<< nav2_lifecycle_manager [23.5s, exited with code 2]
Aborted  <<< nav2_behavior_tree [1min 5s]                                                                                                                                      
Aborted  <<< nav2_velocity_smoother [1min 1s]                                                                                        
Aborted  <<< nav2_amcl [1min 8s]                                           

Summary: 9 packages finished [1min 13s]
  1 package failed: nav2_lifecycle_manager
  3 packages aborted: nav2_amcl nav2_behavior_tree nav2_velocity_smoother
  3 packages had stderr output: nav2_behavior_tree nav2_lifecycle_manager nav2_simple_commander
  24 packages not processed

SteveMacenski, 2022-09-22:
Please format your posts, this is rather difficult to read.  I did it for you this time, but please do in the future.

This is precisely the same error, just on another line, change that `rmw_qos_profile_services_default` to `rclcpp::ServicesQoS()`

muttistefano, 2022-09-23:
Hi, and thanks @SteveMacenski for the suggestions.
Substituting L49 in the service_client.hpp with rclcpp::ServicesQoS() actually worked for me.
I had to perform the same substitution in the file : lifecycle_manager.cpp too, in the nav2_lifecycle_manager package, and in the file nav2_behavior_tree/bt_service_node.hpp 
I still cannot tell where the initial error came from.
Thanks

AshinTheAnvil, 2022-09-23:
Hi @SteveMacenski  , As @muttistefano said it works if lines in those 3 different files are changed.
Now the workspace is built successfully.
Still I do not know how the problem originated.
Thanks.

AshinTheAnvil, 2022-09-29:
Hi @SteveMacenski  Should I or @muttistefano  submit a PR ?

muttistefano, 2022-09-29:
I can submit it, I just need a while.
Do the changes affect other ubuntu distributions or else?

AshinTheAnvil, 2022-09-29:
> I can submit it, I just need a while. Do the changes affect other ubuntu distributions or else?
> […](#)

Hi @muttistefano 
Please go ahead.
I faced similar issue working on Ubuntu 22.

muttistefano, 2022-09-29:
Hi, 
i tried to build the nav2 with the proposed changes on Ubuntu22 - rolling and it doesn't work.

```
/tmp/pd/src/navigation2/nav2_util/include/nav2_util/service_client.hpp: In instantiation of ‘nav2_util::ServiceClient<ServiceT>::ServiceClient(const string&, const SharedPtr&) [with ServiceT = lifecycle_msgs::srv::ChangeState; std::string = std::__cxx11::basic_string<char>; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>]’:
/tmp/pd/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:35:3:   required from here
/tmp/pd/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:49:15: error: cannot convert ‘rclcpp::ServicesQoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   49 |       rclcpp::ServicesQoS(),
      |               ^~~~~~~~~~~~~
      |               |
      |               rclcpp::ServicesQoS
In file included from /opt/ros/rolling/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /tmp/pd/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /tmp/pd/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /tmp/pd/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/opt/ros/rolling/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = lifecycle_msgs::srv::ChangeState; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<lifecycle_msgs::srv::ChangeState> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
```
At this point, i suppose the initial problem was related to Ubuntu 18 .
Thanks

AshinTheAnvil, 2022-09-29:
> Hi, i tried to build the nav2 with the proposed changes on Ubuntu22 - rolling and it doesn't work.
>
> ```
> /tmp/pd/src/navigation2/nav2_util/include/nav2_util/service_client.hpp: In instantiation of [...]

It worked for me. I did as we had discussed a while ago. I did the same steps.

SteveMacenski, 2022-09-29:
I get the same error actually in my install using Rolling binaries on 22.04

SteveMacenski, 2022-09-29:
Either way, I think at this point if you both have something that's working, we can close this if there's no change that needs to be made to the main-supported branch/distribution/OS

AshinTheAnvil, 2022-09-29:
Yes sure.
Thanks Steve

closed 2022-09-29 by SteveMacenski