ros-navigation/navigation2 #3221: Running multple mission with one BehaviorTreeEngine

stevedanomodolor, 2022-09-23:
Just out of curiosity, is it possible to run multiple trees with the same BehaviorTreeEngine in parallel without one affecting the other?

SteveMacenski, 2022-09-23:
Assuming that none of the nodes in the 2 trees running simultaneously conflict with each other, no, I don't see a cause for an issue. For example, you can't have both the trees calling the same controller server giving conflicting commands, else obviously only one of them is going to be processed correctly.

stevedanomodolor, 2022-09-23:
For the run function in https://github.com/ros-planning/navigation2/blob/80909a1cb18e4d66930f7fee9198358bfcd200db/nav2_behavior_tree/src/behavior_tree_engine.cpp#L37, I should be able to call it multiple times with different trees and not have issues, on the condition as you said that they should not be conflicting nodes in the tree?

SteveMacenski, 2022-09-23:
Multiple instances, yes.

stevedanomodolor, 2022-09-23:
Thank you @SteveMacenski

closed 2022-09-23 by stevedanomodolor