ros-navigation/navigation2 #3224: Waypoint Follower has problems with loops in path

wilcobonestroo, 2022-09-26:
## Bug report

- Operating System: Ubuntu 20.04.1 with WSL2 on Windows 10 enterprise
- ROS2 Version: Humble binaries
- Version or commit hash: ros-humble-navigation2  1.1.2-1jammy.20220909.044252
- DDS implementation: Cyclone DDS

#### Steps to reproduce issue

I started the default setup using the following commands:
```
source /opt/ros/humble/setup.bash
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False
```
I gave the initial pose estimation. Then I clicked `Waypoint / Nav through Poses` button. I entered some waypoints that include a loop or crossing. When the robot gets at the intersection, it starts rotating.

[image]

#### Expected behavior
The robot should follow the path, going through waypoint 1, 2, etc.

#### Actual behavior
The robot gets stuck at the intersection. The console shows the following errors:

```
[component_container_isolated-6] [INFO] [1664204403.282001500] [controller_server]: Passing new path to controller.
[component_container_isolated-6] [INFO] [1664204406.282359100] [controller_server]: Passing new path to controller.
[component_container_isolated-6] [INFO] [1664204409.332028400] [controller_server]: Passing new path to controller.
[component_container_isolated-6] [ERROR] [1664204409.432006700] [controller_server]: Failed to make progress
[component_container_isolated-6] [WARN] [1664204409.432204900] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[component_container_isolated-6] [INFO] [1664204409.448672000] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[component_container_isolated-6] [INFO] [1664204409.451335100] [controller_server]: Received a goal, begin computing control effort.
[component_container_isolated-6] [WARN] [1664204409.522955600] [controller_server]: Control loop missed its desired rate of 20.0000Hz
[component_container_isolated-6] [INFO] [1664204412.358360200] [controller_server]: Passing new path to controller.
[component_container_isolated-6] [INFO] [1664204415.408384000] [controller_server]: Passing new path to controller.
[component_container_isolated-6] [INFO] [1664204418.458356800] [controller_server]: Passing new path to controller.
[component_container_isolated-6] [INFO] [1664204421.458438800] [controller_server]: Passing new path to controller.
[component_container_isolated-6] [ERROR] [1664204421.558496500] [controller_server]: Failed to make progress
[component_container_isolated-6] [WARN] [1664204421.558721700] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[component_container_isolated-6] [INFO] [1664204421.578224000] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[component_container_isolated-6] [INFO] [1664204421.578585500] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[component_container_isolated-6] [WARN] [1664204421.803100300] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 4.4701 Hz
[component_container_isolated-6] [INFO] [1664204421.824858300] [controller_server]: Received a goal, begin computing control effort.
[component_container_isolated-6] [INFO] [1664204424.875638700] [controller_server]: Passing new path to controller.
```
Sometimes I also get these.
```
[component_container_isolated-6] [WARN] [1664203791.993742800] [controller_server]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.040393900] [DWBLocalPlanner]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.040527500] [DWBLocalPlanner]: 1.00: BaseObstacle/Trajectory Hits Obstacle.
[component_container_isolated-6] [WARN] [1664203792.041014400] [controller_server]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.090633200] [DWBLocalPlanner]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.092851800] [DWBLocalPlanner]: 1.00: BaseObstacle/Trajectory Hits Obstacle.
[component_container_isolated-6] [WARN] [1664203792.094528000] [controller_server]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.096045500] [controller_server]: Controller patience exceeded
[component_container_isolated-6] [WARN] [1664203792.096126300] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[component_container_isolated-6] [INFO] [1664203792.115866700] [behavior_server]: Attempting backup
[component_container_isolated-6] [WARN] [1664203792.116672200] [behavior_server]: Collision Ahead - Exiting DriveOnHeading
[component_container_isolated-6] [WARN] [1664203792.116730700] [behavior_server]: backup failed
[component_container_isolated-6] [WARN] [1664203792.116753700] [behavior_server]: [backup] [ActionServer] Aborting handle.
[component_container_isolated-6] [INFO] [1664203792.141993900] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[component_container_isolated-6] [INFO] [1664203792.142767400] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[component_container_isolated-6] [WARN] [1664203792.799159500] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 1.5282 Hz
[component_container_isolated-6] [INFO] [1664203792.818036800] [controller_server]: Received a goal, begin computing control effort.
[component_container_isolated-6] [ERROR] [1664203792.825989200] [DWBLocalPlanner]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.826280000] [DWBLocalPlanner]: 1.00: BaseObstacle/Trajectory Hits Obstacle.
[component_container_isolated-6] [WARN] [1664203792.826748200] [controller_server]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.875194100] [DWBLocalPlanner]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.875318400] [DWBLocalPlanner]: 1.00: BaseObstacle/Trajectory Hits Obstacle.
[component_container_isolated-6] [WARN] [1664203792.875847900] [controller_server]: No valid trajectories out of 419!
[component_container_isolated-6] [ERROR] [1664203792.924911100] [DWBLocalPlanner]: No valid trajectories out of 419!
```

It keeps doing this for a while and then gives up.

#### Additional information

wilcobonestroo, 2022-09-27:
It is indeed the same issue as #2622. I will have a try with RPP.

closed 2022-09-27 by wilcobonestroo

SteveMacenski, 2022-09-27:
Well it should work with DWB too, so if you find it works with RPP and not DWB, that's worth opening the ticket again for and seeing what we missed when we ported the work from RPP to DWB.

hg5j9k6a, 2024-05-18:
I'm not sure is it good to leave comment here.

I'm currently facing the same issue of intersecting path, and I'm using released branch of 1.1.12 , RPP controller and Smac Hybrid-A* Planner.

most of the time, robot will suddenly track the end part of path as video below:
https://github.com/ros-navigation/navigation2/assets/48769492/a6524222-f813-472b-9063-1c90365fd599

due to the RPP controller is using function: `nav2_util::geometry_utils::min_by` to find the closest pose on the path, i tried to modify some code from /geometry_utils.hpp

the result is good to me, and here is the result video:
https://github.com/ros-navigation/navigation2/assets/48769492/3f0e7d4a-eebb-46c9-9d78-c1914de26a19
https://github.com/ros-navigation/navigation2/assets/48769492/d7e1110b-7e5b-4fcf-8a0b-15c2db2b3942

and the code i changed is shown here:
[image]
[image]

the idea is came from #2622 from @SteveMacenski , and the few assumption i made is:
1. the RPP controller should be robust enough to handle a small error of start pose from path if the error is within resolution value.
2. based on grid map, if intersection occurred in same grid , then we choose the old pose we found before.
3. RPP controller erases the passed parts of path.

however, in case of given a static path and robot position off the path, where robot is closer to reset part of path, it will change the tracking path to the reset part of path as video below:

https://github.com/ros-navigation/navigation2/assets/48769492/d403c756-b8a2-473a-86fc-51c490c622c4

i believe this result is acceptable and if robot off the path, it should replan the path.

hg5j9k6a, 2024-05-21:
Thank you for your reply, it inspired me to solve my problem in a different way.

As you mention: 
> That distance of path tracking error could be small/large depending on the algorithm or the choice of parameters.
I really agree with that, because my assumption is based on perfect controller with no error, and it will be hard to find a reasonable value when it comes error..

But after you point to `max_robot_pose_search_dist`, i start wondering if there was any good method to find a boundary of path that interested me. Then i tried to find a path without fallback for [closest_pose_upper_bound](https://github.com/ros-navigation/navigation2/blob/bf291d704dbb1c2572575dd0620b12f816407233/nav2_regulated_pure_pursuit_controller/src/path_handler.cpp#L67)

Here is the thing i tried:
I created a parameter `robot_pose_search_fallback_tolerance` for checking upper bound of path that won't have fallback part, and  `robot_pose_search_fallback_tolerance` should greater than or equal to xy_tolerance.
and added a new function of `first_after_integrated_distance_without_fallback`:
[image]

The result as below is even better than the one i shown before.
https://github.com/ros-navigation/navigation2/assets/48769492/c16888f5-eb94-4e24-96d0-a9c8936cd7f7

If you have any other concern or suggestion please let me know.

SteveMacenski, 2024-05-21:
I don't fully understand the fallback bit, why is that necessary if you set the max search distance appropriately for your application?

hg5j9k6a, 2024-05-21:
yes, you're right!!!
If I set the maximum search distance appropriately, the behavior is what I expect and I don't need to modify anything.
`max_robot_pose_search_dist` solved my problem. 
thanks for your help.