ros-navigation/navigation2 #3225: Request to support additional sensors in collision monitor

automech-rb, 2022-09-29:
## Feature request

#### Feature description

Add support for RGBD sensor (`sensor_msgs/Image`) for collision monitor.

#### Implementation considerations

Similar to sensor described at : https://navigation.ros.org/setup_guides/sensors/setup_sensors.html

SteveMacenski, 2022-09-29:
A depth image can be easily converted into a PointCloud2 for use via a depth image proc pipeline instance.

automech-rb, 2022-09-29:
My mindset was to not increase one more topic as `sensor_msgs/Image` type is supported in Costmap2D. But what you say makes sense too.

Closing the issue.

closed 2022-09-29 by automech-rb

SteveMacenski, 2022-09-29:
Images are not supporting costmap 2D. That section is going over types of gazebo sensors for simulation