ros-navigation/navigation2 #3230: Tb3  is out of bounds of the costmap when navigating and mapping.

lmendyk, 2022-10-05:
I’m following the tutorial ( SLAM) Navigating While Mapping¶ for physical new turtlebot3 with new lidar LDS-02.  Initially I can move the robot using rviz Navigation plug-in – the map seems to be create correctly. But after while, the robot disappers from  the current position and is located at different “island” which is a replica of the previous piece of map but place in different part on the rviz screen.

The logs which I think may be related to the problem are:
 
[planner_server-2] [WARN] [1664956617.007039032] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1664956617.010995992] [planner_server]: Cannot create a plan: the robot's start position is off the global costmap. Planning will always fail, are you sure the robot has been properly localized?

…
[controller_server-1] [WARN] [1664956617.066782169] [local_costmap.local_costmap]: Sensor origin at (-556.94, -304.65 0.13) is out of map bounds (-553.96, -301.67, 0.91) to (11.70, -22.00, 0.00). The costmap cannot raytrace for it.

[planner_server-2] [WARN] [1664956617.207334840] [global_costmap.global_costmap]: Sensor origin at (10.60, -21.62) is out of map bounds (-3.31, -3.17) to (10.87, 6.65). The costmap cannot raytrace for it.
[planner_server-2] [WARN] [1664956617.211693706] [planner_server]: Cannot create a plan: the robot's start position is off the global costmap. Planning will always fail, are you sure the robot has been properly localized?

I have used commands for starting the navigation stack as in the tutorial.  I guess that it may be caused by some localization drifting.  Could you point me if I can use some parameters  tuning to add some tolerance to the sensor data? Or it is due to wrong turtebot3 configuration provided by the vendor?  Or some problem with costmap itself?

SteveMacenski, 2022-10-05:
Please send us a screen cast video, I'm not sure exactly what you're describing to give an answer

lmendyk, 2022-10-05:
This is the screen cast.
[Screencast from 05.10.2022 23_54_23.webm](https://user-images.githubusercontent.com/62011538/194171927-0f5a4eb8-1e90-4a58-b671-bdb4a999c8cf.webm)

SteveMacenski, 2022-10-05:
Yeah, that shouldn't be happening. That's an issue with the SLAM Library, please open a ticket in the repository of the SLAM system you're using explaining how you installed it / version.

closed 2022-10-05 by SteveMacenski