ros-navigation/navigation2 #3241: Local map is empty, global map not updating any new obstacles

AgentRegi, 2022-10-12:
Hello dear nav2 team,
I have an issue with the getting started setup (https://navigation.ros.org/getting_started/index.html) for ros humble running on Ubuntu 22.04.
When I start the launch file (ros2 launch nav2_bringup tb3_simulation_launch.py) my local map stays empty.
It is published but all grid cells are 0, I have this grey rectangle around the robot in rivz.
Even when I insert new obstacles in gazebo the  global map is not updated accordingly.

The laser scan in rviz looks fine and the localization works too (even slam_toolbox is doing what it should with slam:=True). I can navigate but only based on static map plugin in the global map (I guess).
The global and local costmap also subscribing to the /scan topic (ros2 node info (e.g.) /local_costmap/local_costmap) 
I also checked various local costmap parameters and from the outside everything looks alright. 

I guess it has something to do with the scan topic. I have the feeling the sensor callback is not executed. 
Before I take a closer look at the code (my first step would be check the scan callbacks of the costmaps), I would like to ask if somebody else had this issue. 

It is probably something very dumb from my side but I can't wrap my head around it. Because from checking everything that I know from ROS1 there should be no problem. Could it be an rviz visualization issue.

Best
Regi

automech-rb, 2022-10-12:
Check https://github.com/ros-planning/navigation2/issues/3014

AgentRegi, 2022-10-12:
Hi,
thank you. But I am still not sure how to fix should I follow this discussion https://github.com/ros2/rmw_fastrtps/pull/619.

automech-rb, 2022-10-12:
Sorry for my short answer. You can try changing your DDS to cycloneDDS.

AgentRegi, 2022-10-12:
Aha ok. What I did is building the https://github.com/ros2/rmw_fastrtps package humble branch in a ros ws together with nav2 repo. Now the costmap update works. And the costmaps are as they used to be. 
Thanks I am happy now.

closed 2022-10-12 by SteveMacenski

TheConstructAi, 2022-10-21:
Is this the official solution? Its strange for me that even in the demo example this is the case.
Has there been a fix of that?

SteveMacenski, 2022-10-26:
For the moment, Fast-DDS is not usable with Nav2 (and probably other common packages) until they fix and release the patches to these issues. Instead, use Cyclone DDS.