ros-navigation/navigation2 #3243: ClearEntireCostmap plugin sends request to server but does not clear the costmap

kamakshijain, 2022-10-12:
Hello, I am trying to use ClearEntireCostmap plugin (and similar costmap manipulation plugins) for both globalcostmap and localcostmap. I however, am able to send the request and get a response however, I don't see the costmap clearing.
I also, tried to send command line service call and was able to get a response but again costmap did not clear.

Please see the attached picture. 
Picture contains, navigation logs, cmd line service call to global and local costmap and rviz.

## Bug report
[image]

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic 
- Version or commit hash:
  - release version 1.0.12
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue
1. Use nav galactic 1.0.12
2. Behavior tree : using `ClearEntireCostmap name="ClearGobalCostmap-Subtree" service_name="global_costmap/clear_entirely_global_costmap"` 
3. or `ros2 service call /global_costmap/clear_entirely_global_costmap nav2_msgs/srv/ClearEntireCostmap "{request: {}}"` or 
`ros2 service call /local_costmap/clear_entirely_local_costmap nav2_msgs/srv/ClearEntireCostmap "{request: {}}"`

#### Expected behavior
1.  local and global costmap should have cleared. Basically working as though in my config yaml file costmap was disabled.
#### Actual behavior
1. costmap is still there. No change.
#### Additional information

----
## Feature request
Ability to enable and disable costmaps at run time.
#### Feature description

Ability to disable or enable costmaps (local and global) after it has been set by config yaml file.
#### Implementation considerations

SteveMacenski, 2022-10-12:
Your example looks to only have the static and inflation layer. The static layer is not clearable, since it is not going to change by deleting all its data and repopulating (e.g. there is no live data being processed from it to "clear"). The inflation layer then has no action if there are no changes.

So I think that it is probably working, you're just not using any layers that clearing is meaningful with

kamakshijain, 2022-10-12:
Sharing 2 different configurations I tried. 
1st modified for our testing
2nd sample configuration provided in the turtorial (https://navigation.ros.org/configuration/packages/configuring-costmaps.html)
Am I miss it something?

Trial 1:
```
local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: True
      plugin_names: ["obstacle_layer", "inflation_layer"]
      plugin_types: ["nav2_costmap_2d::ObstacleLayer", "nav2_costmap_2d::InflationLayer"]
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
      robot_radius: 0.22
      plugins: ["voxel_layer", "inflation_layer"]
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: True
        publish_voxel_map: True
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        mark_threshold: 0
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
      static_layer:
        map_subscribe_transient_local: True
      always_send_full_costmap: True
      observation_sources: scan
      scan:
        topic: /scan
        max_obstacle_height: 2.0
        clearing: True
        marking: True
  local_costmap_client:
    ros__parameters:
      use_sim_time: True
  local_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: True

global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 1.0
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_link
      use_sim_time: True
      robot_radius: 0.22
      resolution: 0.05
      plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
      always_send_full_costmap: True
      observation_sources: scan
      scan:
        topic: /scan
        max_obstacle_height: 2.0
        clearing: True
        marking: True
  global_costmap_client:
    ros__parameters:
      use_sim_time: True
  global_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: True
```

Trail 2: default config given in tutorial

```
global_costmap:
  global_costmap:
    ros__parameters:
      footprint_padding: 0.03
      update_frequency: 1.0
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_link
      use_sim_time: True
      robot_radius: 0.22 # radius set and used, so no footprint points
      resolution: 0.05
      plugins: ["static_layer", "obstacle_layer", "voxel_layer", "inflation_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        combination_method: 1
        scan:
          topic: /scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          inf_is_valid: false
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: True
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        publish_voxel_map: True
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        unknown_threshold: 15
        mark_threshold: 0
        observation_sources: pointcloud
        combination_method: 1
        pointcloud:  # no frame set, uses frame from message
          topic: /intel_realsense_r200_depth/points
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          clearing: True
          marking: True
          data_type: "PointCloud2"
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
        enabled: true
        subscribe_to_updates: true
        transform_tolerance: 0.1
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        enabled: true
        inflation_radius: 0.55
        cost_scaling_factor: 1.0
        inflate_unknown: false
        inflate_around_unknown: true
      always_send_full_costmap: True

local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: True
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
```

SteveMacenski, 2022-10-12:
Are you sure its not being cleared then simply repopulated with the current data?

SteveMacenski, 2022-10-13:
I jsut did a quick test on `rolling / main` and things are working fine. I localized the robot into an incorrect spot a few times so I created alot of scatter in the costmap. Then I called the service `/global_costmap/clear_entirely_global_costmap` service using the ros2 CLI and I can see that the scatter is removed.

Lannister-Xiaolin, 2022-10-16:
I had met the same problem, in ros2 galactive version,  **clear_entirely_global_costmap** not working for both local cost map and global costmap!!

[image]

SteveMacenski, 2022-10-31:
Galactic is EOL starting tomorrow https://docs.ros.org/en/galactic/Releases.html. Happy to merge a patch if someone finds an issue, but its not going to receive developer attention from the maintainers. Fortunately, this does not appear to be a problem on the current branches, I'd recommend upgrading ROS distributions since Galactic is no longer supported as of tomorrow.

closed 2022-10-31 by SteveMacenski