ros-navigation/navigation2 #3247: ros-humble: compille error for nav2_simple_commander in ubuntu22.04

xianglunkai, 2022-10-15:
--- stderr: nav2_simple_commander                                    
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 0.1.43ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
---

SteveMacenski, 2022-10-15:
I believe this is general ROS2 problem with all ROS2 python packages. Please correct me if I’m wrong, then I’ll definitely look at fixing this. But I thought I had looked into it and found that all python packages are giving this error on first build.

xianglunkai, 2022-10-16:
Txs! 
However, i did not prompt any errors when compling under galatic-ubuntu20.04 Focal, so i suspect it is related to the python version in ubuntu22.04

jwallace42, 2022-10-20:
Just wanted to point out that this issue doesn't actually stop you from running the nav2_simple_commander on ubuntu 22.04.

closed 2022-10-26 by SteveMacenski