ros-navigation/navigation2 #3249: Console screen jumps back to "normal" screen after way point following or navigating through poses.

jwallace42, 2022-10-19:
## Feature request

#### Feature description

Stay on the navigate through poses screen after success or failure. This prevents the user from having to click back into that screen ever time they want to navigate through poses or to waypoints. 

#### Implementation considerations

padhupradheep, 2022-10-21:
The easiest way to deal with this issue is that, we need to transition to `accumulating` after the `accumated_wp_` state becomes inactive. 

https://github.com/ros-planning/navigation2/blob/a50d45f1c275fdb557a8b0e4f89754887800f67a/nav2_rviz_plugins/src/nav2_panel.cpp#L426

Currently, this state usually goes to inactive, when the waypoint following has been completed, or if the goal status from the waypoint follower is anything other than accepted or executing. Same applies for nav_through_poses. 
Also, in this PR, I make it inactive, when there are no waypoints accumulated or when there is an invalid loop value. 

`accumulating` state is the state where you accumulate goals for either waypoint following or nav through poses.

Also the user wanted to navigate and work with multiple waypoints. So landing back to the accumulating state after finishing the initial task with the waypoint following or nav_through_poses would be not a bad idea. I guess this point also might justify the change.

jwallace42, 2022-10-21:
@padhupradheep 

Sounds good! If you open a pr I can review :).

SteveMacenski, 2022-10-31:
This seems sensible. Is it anyone's plan to submit a PR on it? Else, my "keep tickets actionable" twitch makes me want to close this. Its a subjective opinion of behavior - which I think is reasonable - but also not really critical or needing to be changed if no one's committing to its implementation.

padhupradheep, 2022-10-31:
Will be taken care in #3165 😃

SteveMacenski, 2022-10-31:
OK - linked

SteveMacenski, 2022-11-03:
I'm going to close this since its a nice to have + implemented and roll it in under the ticket for the looping waypoint following ticket that #3165 implements

closed 2022-11-03 by SteveMacenski