ros-navigation/navigation2 #3253: Runtime costmap plugin enabling/disabling doesn't work

andriimaistruk, 2022-10-21:
Hi, I faced an issue that changing at runtime the value of obstacle_layer.enabled from true to false doesn't work, meaning the scan data is still projected on the costmap. Does this work at all or there is some issue with doing it via "ros2 param set /local_costmap/local_costmap obstacle_layer.enabled false"?
Also, could you also take a look at https://answers.ros.org/question/403196/smac-collision-checking-for-non-circle-footprint-type/.
Thanks!

padhupradheep, 2022-10-21:
If you are looking for the dynamic parameter reset in a distro pre-humble, unfortunately, we do not support that. With Galactic edging to EOL very soon, I guess this feature will not be supported for older distros.

andriimaistruk, 2022-10-25:
For runtime change of the `obstacle_layer.enabled` value from true to false to make the scan disappear from costmap, I should move to humble, correct?

padhupradheep, 2022-10-25:
Yes!

closed 2022-10-25 by andriimaistruk