ros-navigation/navigation2 #3257: Add success/failure information to the `NavigateToPose` action result

sea-bass, 2022-10-25:
## Feature request

#### Feature description
Right now, the `NavigateToPose` action only has an empty result saying whether the action was complete. It would be great to expand this to at least include the success/failure of the action.

My use case is wrapping calls to the Nav2 stack as part of a task execution framework that should not only determine when navigation is complete, but also whether the robot successfully reached its target pose when the action was complete.

As a workaround, we can look at the robot pose when the action is complete, and compare it with the original goal.

jwallace42, 2022-10-25:
This should be really simple to implement. We can just pass through the result of the behavior tree here: 

https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L231 and change the action message result to a bool. 

Out of curiosity, are you using the nav2 simple commander?

sea-bass, 2022-10-25:
> 

> This should be really simple to implement. We can just pass through the result of the behavior tree [here](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L231) and change the action message result to a bool.

Oh, yeah, I may try this out in a personal fork to see how it does.

> Out of curiosity, are you using the nav2 simple commander? It already wraps action calls into a [python api](https://github.com/ros-planning/navigation2/tree/main/nav2_simple_commander)

I'm not using the simple commander directly -- we're doing our own custom thing based off [`pyrobosim`](https://github.com/sea-bass/pyrobosim) that basically does the same thing.

With this modification, both my solution and the simple navigator could similarly be changed to be more expressive... so I like this. From [these lines of code](https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L160-L161), for example.

Current:
```python
        self.result_future = self.goal_handle.get_result_async()
        return True
```

Possible:
```
        self.result_future = self.goal_handle.get_result_async()
        return self.result_future.result().success
```

jwallace42, 2022-10-25:
> Thank you for the quick and encouraging response!

np! It is a good idea. I will put a pr out for this change in later tonight :). Pretty busy with some other stuff during the day

> Oh, yeah, I may try this out in a personal fork to see how it does.
Cool! Let me know how it goes :). 

> With this modification, both my solution and the simple navigator could similarly be changed to be more expressive... so I like this. From [these lines of [...]

I will add that in too.

jwallace42, 2022-10-26:
So, 

I completely forgot that the action server has a result code :). I would take a look at https://docs.ros.org/en/foxy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Cpp.html. 

`  void result_callback(const GoalHandleFibonacci::WrappedResult & result)
  {
    switch (result.code) {
      case rclcpp_action::ResultCode::SUCCEEDED:
        break;
      case rclcpp_action::ResultCode::ABORTED:
        RCLCPP_ERROR(this->get_logger(), "Goal was aborted");
        return;
      case rclcpp_action::ResultCode::CANCELED:
        RCLCPP_ERROR(this->get_logger(), "Goal was canceled");
        return;
      default:
        RCLCPP_ERROR(this->get_logger(), "Unknown result code");
        return;
    }`

I would also take a look at the python implementation [here](https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/demo_inspection.py). 

It has exactly what you are looking for :).

sea-bass, 2022-10-26:
Yeah -- That's exactly what I need and it works, thank you!

Sneaky that all actions in ROS2 now have that status baked in separately from the actual result message definition.

closed 2022-10-26 by sea-bass