ros-navigation/navigation2 #3258: Calling halt on all nodes does not fully initialize a behavior tree.

tgreier, 2022-10-25:
## Bug report
Changes introduced in BehaviorTree_v3_cpp 3.6.1 (https://github.com/BehaviorTree/BehaviorTree.CPP/pull/329) have caused behavior trees to not be fully re-initialized when re-used.

I am using a custom navigation behavior tree that uses a SequenceStar. When the navigation is canceled, the assumption is that all BT nodes are halted and initialized.

The pull request cited above has changed behavior such that SequenceStar->halt no longer sets child idx to 0. 
I understand the reason for the change. But this leaves no method to completely re-initialize a BT without destroying it and rebuilding it.

[This code in bt_navigator](https://github.com/ros-planning/navigation2/blob/ca482808a7a7c52ce01ae3c662dc2b980968fc16/nav2_bt_navigator/src/bt_navigator.cpp#L156) blocks a behavior tree from being recreated if it is already loaded. Therefore, the reused behavior tree is not starting from a fully initialized state.

Is there a work-around for this issue?
 
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - FOXY

#### Steps to reproduce issue
Re-use a navigation behavior tree with a SequenceStar that was canceled.

#### Expected behavior
Behavior Tree executes as if it were starting from a fully initialized state.

#### Actual behavior
SequenceStar nodes start from where they left off.

#### Additional information

jwallace42, 2022-10-26:
So, 

this appears to be a implementation issue with your behavior tree rather than a issue with nav2. The behavior tree guys also responded to your question https://github.com/BehaviorTree/BehaviorTree.CPP/pull/329. :)

jwallace42, 2022-10-26:
Ah, alight! 

Just looked through the the [tree class](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/v3.8/include/behaviortree_cpp_v3/bt_factory.h) it doesn't seem like there is anything that _resets_ the tree which I am not really surprised by. 

Maybe the BT guys have another way, otherwise we will have to reload the tree each time.

jwallace42, 2022-10-31:
@tgreier there doesn't seem to be any updates from the BT guys. If you want to put in a pr removing the check to reload that would be great :).

SteveMacenski, 2022-11-01:
I commented on that closed PR, I agree with you, that PR was not correct. Lets see what they say.

jwallace42, 2022-12-01:
Still no response from the BT guys :(.

SteveMacenski, 2022-12-01:
https://github.com/BehaviorTree/BehaviorTree.CPP/pull/329 was reverted! It should be fixed next release of BT.CPP

closed 2022-12-01 by SteveMacenski