ros-navigation/navigation2 #3271: Codecov metrics for Assisted Teleop / Drive on Heading behaviors not being picked up

SteveMacenski, 2022-11-07:
CC @jwallace42 

This is especially odd since the other plugins are getting their metrics reported correctly (wait, spin, backup) and backup uses alot of drive on heading as part of its code. I can't seem to find any reason for this after a couple of hours of initial debugging.

padhupradheep, 2022-11-08:
Just a small hunch, I think by default `assisted_teleop` is not included as part of the default behavior tree nor in any of the BT's that we have. Could that be a cause ? :thinking:

SteveMacenski, 2022-11-08:
No, the system tests call the behavior directly

SteveMacenski, 2022-12-01:
I was playing around this afternoon in https://github.com/ros-planning/navigation2/pull/3305. Forward declaring template class did not help (I suspected something odd was happening with the template type patterns), nor did matching the public/private functions + destructor file location (not that I thought that would help, but it was one of the very things different about the 2 failing plugins from the others). 

The files look to me to be exactly the same otherwise. As well with the system tests, only difference I can find is drive on heading uses `goal.time_allowance.sec = test_params.time_allowance;` in test cases. I'm stumped at the moment. I notice the tests aren't failing cleanly which could potentially be part of it? But doesn't explain why some work and some don't, they all appear to have that issue. Next up to diagnose the ghost in the system is to start renaming stuff and moving files around to see if I can isolate it to something in cmake / linking potentially (e.g. swap the test code or plugin code to see which it follows - issue is in system tests or in the code itself // swap order of build / link to see if its related to something in CI with build order and caching).

mmattb, 2023-01-06:
Thanks to Steve for pointing this issue out to me on the discussion forums. I'd love to help out, since I've touched code coverage mechanics before in other contexts. Can somebody point me to some materials I can use to get bootstrapped on this? e.g. what package and what tests are involved, how do I install and run them, what does your workflow for gathering the code cov info look like?

SteveMacenski, 2023-01-06:
Hi,

Oh my god, what a savior! Let me start with some context and resources, then I'll answer your specifics.

So this all started when we noticed when we moved to 22.04/Humble (and Rolling's cycle to 22.04 too) that our coverage dropped significantly due to missing coverage with what seemed to be most (if not all) of the Nav2 System Tests. The unit testing in each package seemed to be fine. We didn't really understand why. I thrashed around a bit and found that if I tested the stack with `use_composition=false` instead, that we got the vast majority of our test coverage for the system tests back. Weird, but I moved on thinking that it wasn't worth the time to find root cause if we had a solution and this isn't a functional problem for users, just in testing metrics (though, I feel the dark technical debt monster waiting to bite me). Might also be something of interest for you to look into if you're curious. I think it might (?) have something to do with a non-clean exit of the tests due to elevated killing of the processes after the test completes successfully, but I never actually got to testing it. Though the fact that we're not cleanly exiting in composition is a problem in its own right regardless (which might also be a nice contribution!). 

After that, I noticed that basically the only thing I could easily identify now as missing was in the behavior server, namely the assisted teleop and drive on heading plugins. We have system tests which cover them but for some reason some or many of the functions that are being called from them aren't getting picked up. I tried a bunch of stuff to see but after 3 days I ran out of time and had to move on to complete my 2022 goals and my plan of action for continuing is above to try a few desperate last-ditch things to see if I could identify where the issue lies (in the behavior package, or the system tests).

Resources I've been using:
- These are the behavior system tests https://github.com/ros-planning/navigation2/tree/main/nav2_system_tests/src/behaviors
- The behaviors source code https://github.com/ros-planning/navigation2/tree/main/nav2_behaviors
- Our codecov entry, which builds all PRs and branches on the main repo https://app.codecov.io/gh/ros-planning/navigation2
- The codecov showing the issue https://app.codecov.io/gh/ros-planning/navigation2/tree/main/nav2_behaviors/plugins

> Can somebody point me to some materials I can use to get bootstrapped on this?

In addition to the resources above, just the usual build / install instructions for Nav2/ROS2. Install Rolling binaries on 22.04 + Nav2 from source in a workspace `colcon build`. I gather code coverage results from pushing branches to the git server and running circleCI or opening a draft PR while I play around. I believe you can run codecov locally but I didn't bother to try to figure out all that since CI didn't run much slower and gave me CPU time to do other things in parallel. If you want to do this too, you can open a draft PR (which I'll just unsubscribe to until you tell me to look at it) to push and play at your pleasure. Or I can give you tentative access to push branches if you don't want to make as much noise. 

These are some instructions for executing individual tests and generating a report using lcov https://github.com/ros-planning/navigation.ros.org/issues/88#issue-710546771 - though as I said, I never actually tried using that for this particular issue (yet) so I don't actually know what that would report is any different.

mmattb, 2023-01-07:
Ah I see. I've seen a bunch of things which can look like this. Metrics not flushing atexit seemed like the most common one, which indeed could indicate something strange with the process unwind/atexit/exit. Another had to do with weird dynamic linking details. Another is the code coverage tool not adapting to certain optimizations, macros, or code gen, but this code on its face looks too vanilla to be running afoul of those.

I'm working on getting the tests running on my local machine. My intuition is to grep the symbols in the binary and the codecov microdata to bisect the problem. I might end up tinkering with the shutdown/killing process since that does seems like a good starting point. When somebody doesn't have control over a strange shutdown sequence, they may sometimes conditionally compile in manual flushes in some places. That's a big hammer, but in our case it could give us a way to diagnose the problem. [StackOverflow example](https://stackoverflow.com/questions/30929205/gcov-cross-profiling-gcov-flush-does-not-flush-coverage-data-for-shared-lib).

Who can help me get to the bottom of why the tests just spin forever waiting for the initial pose to be received? Any hints?

I was able to invoke the unit tests using the tutorial, and to get code cov data from those (great tutorial!).  It seems that the system tests are invoked using some Python. I stripped the command out so I can run the test directly, but It just spins on the initial pose. Help?

```
/usr/bin/python3.10 -u /opt/ros/rolling/share/ament_cmake_test/cmake/run_test.py /home/mbryan/Projects/ros/navigation2/build/nav2_system_tests/test_results/nav2_system_tests/test_drive_on_heading_recovery.xml --package-name nav2_system_tests --generate-result-on-success --env TEST_MAP=/home/mbryan/Projects/ros/navigation2/src/nav2_system_tests/maps/map_circular.yaml TEST_EXECUTABLE=/home/mbryan/Projects/ros/navigation2/build/nav2_system_tests/src/behaviors/drive_on_heading/test_drive_on_heading_behavior_node TEST_WORLD=/home/mbryan/Projects/ros/navigation2/src/nav2_system_tests/worlds/turtlebot3_ros2_demo.world GAZEBO_MODEL_PATH=/home/mbryan/Projects/ros/navigation2/src/nav2_system_tests/models BT_NAVIGATOR_XML=navigate_to_pose_w_replanning_and_recovery.xml --command /home/mbryan/Projects/ros/navigation2/src/nav2_system_tests/src/behaviors/drive_on_heading/test_drive_on_heading_behavior_launch.py

...

[test_drive_on_heading_behavior_node-15] [WARN] [1673084120.542461901] [DriveOnHeading_behavior_test]: Initial pose not received
[test_drive_on_heading_behavior_node-15] [INFO] [1673084120.542633978] [DriveOnHeading_behavior_test]: Sent initial pose
[controller_server-7] [INFO] [1673084120.593469320] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[test_drive_on_heading_behavior_node-15] [WARN] [1673084120.643018619] [DriveOnHeading_behavior_test]: Initial pose not received
[test_drive_on_heading_behavior_node-15] [INFO] [1673084120.643188216] [DriveOnHeading_behavior_test]: Sent initial pose
[test_drive_on_heading_behavior_node-15] [WARN] [1673084120.743676705] [DriveOnHeading_behavior_test]: Initial pose not received
[test_drive_on_heading_behavior_node-15] [INFO] [1673084120.743861768] [DriveOnHeading_behavior_test]: Sent initial pose

...
```

I'll be on a train all day tomorrow; will try to pick this up later.

SteveMacenski, 2023-01-09:
Something to note too looking at the code is that the DriveOnHeading and the BackUp behaviors both share a ton of code (BackUp is derived from DriveOnHeading). Of the duplicated code that exists, its mostly to enforce protections on the backup version, so doing an apples-to-apples comparison of them is possible and as far as I can tell are the exact same. Yet one provides the results and the other is missing some. 

> Who can help me get to the bottom of why the tests just spin forever around setting the initial pose? Any hints?

What do you mean? The setting of initial pose can take a few times to get together in the tests but it should eventually. Looking at the CI logs, you'll see alot of those messages but it always gets there. I think its just while things are setting up (and/or using the default DDS vendor which may or may not be having some discovery issues) and lifecycle nodes are activating. Its been awhile since I seriously thought about that since it works well enough and reliably for tests in my experience. 

But that may not be what you're seeing if its actually spinning forever. Which tests are you trying to run? 

>  It seems that the system tests are invoked using some Python

Yes, many are based on launch files, with a test script and a setup script (tester and the nav stack bringup). The launch API has a test API so that once the tester exits, it takes down the setup as well (e.g. nav stack). We use that for the system tests to launch the nav stack or a part of it we're interested in alongside a tester.py script.

> Timed out waiting for transform from base_link to odom to become available

That sounds like gazebo might not be running

mmattb, 2023-01-10:
I was able to recreate the issue today, finally. Took some set up time :) I can reproduce the same basic cov numbers you linked me to, using:

`colcon test --packages-select nav2_system_tests`

Adding sleeps to TearDownTestCase in the relevant system tests didn't seem to give a chance for gcov to flush.  Not sure if it flushes periodically or at process exit only. Next I'll try a manual __gcov_flush if it's straightfoward, which will bisect the problem nicely.

SteveMacenski, 2023-01-10:
Great! I can't say how much I appreciate your work here. This is much, much deeper into gcov/lcov than I would have been able to go

mmattb, 2023-01-13:
Returning to this today...  I think I've effectively proven that the gcov dump process, triggered atexit by __gcov_exit(), is not running when it should.  So Steve's theory seems correct. Using onRun() as a reference, I can add __gcov_exit there, which effectively guarantees we see the onRun call in progress (see screenshot below). Using log lines trivially proves the code is still being covered.

I don't really understand the ros2/launch or nav2 launch stuff yet; I just glanced at it briefly. Likewise with ROS node signal handlers etc. That's clearly where my focus should go next.

@SteveMacenski has anybody tried to bisect code changes on the ROS or Nav2 sides to see if this is due to a code change, or is the behavior simply different on 22.04? (I don't want to set up a 20.04 VM to find out)

[image]

mmattb, 2023-01-13:
Uncaught exception in the tear down of the behavior node causing us to drop the process on the floor?

`  0 [43.062857] (nav2_system_tests) StdoutLine: {'line': b'9: [behavior_server-10] [INFO] [1673598507.589703428] [behavior_server]: Destroying\n'}
  1 [43.062996] (nav2_system_tests) StdoutLine: {'line': b"9: [behavior_server-10] terminate called after throwing an instance of 'rclcpp::exceptio    ns::RCLError'\n"}
  2 [43.063104] (nav2_system_tests) StdoutLine: {'line': b"9: [behavior_server-10]   what():  Couldn't initialize rcl timer handle: the given conte    xt is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/guard_condition.c:67\n"}
`

mmattb, 2023-01-13:
Also, could somebody link me to the test logs for one of the bunk code cov runs, and also for the last-known-good run?  Specifically I want events.log that I see spit out in my sandbox runs, i.e. log/test_<timestamp>/events.log .  That will let me self-help looking for differences between the runs.

SteveMacenski, 2023-01-13:
> can you give me the hash of the last-known-good build?

For the assisted teleop file you show, there has never been a 'good build' where the coverage was displayed properly, to my knowledge. I believe the first time I noticed, I added some prints to show that the functions were being called so I didn't hold it against the PR contributor since it wasn't something they did. 

> Uncaught exception in the tear down of the behavior server causing us to drop the process on the floor? if the log lines are telling the truth, we are hitting an exception during shutdown, which could perhaps explain why we don't get a gcov dump with all the goods

Semi-related question for my person education: I've used GDB to grab tracebacks (my tutorial https://navigation.ros.org/tutorials/docs/get_backtrace.html) on crashes but haven't used it to try to debug shutdown failures before. Is there anything semantically different going on there or the same old? 
 
On topic: It is interesting to note as I did in the longer comment that when I switch from non-composed bringup (e.g. 1 server 1 process) to composed bringup (e.g. all servers in 1 process) we did get a ton of our coverage results back. So I'm curious to know if the exception that's being raised on shutdown is unique to the Behavior Server where we see that issue. Also, why it would be that some of the behavior server tests give us results if others don't? 

> Need to spelunk this Bond code

Bond basically creates a set of publishers and subscriptions between 2 servers to make sure each other are still alive by passing heartbeats. I only use it single-directionally that the Lifecycle Manager is checking if the `<server>` is still responding. If not, it might be in a deadlock condition or crashed, so then the Lifecycle Manager will bring down all servers under its control for safety. 

In a given CI run in Circle you can go to the `artifacts` tab and it provides all of the files we cache ([example from a nightly job yesterday](https://app.circleci.com/pipelines/github/ros-planning/navigation2/8641/workflows/4c448a4b-11d2-49cd-bae3-173c0ee78819/jobs/28678/artifacts)). I see an `events.log`, but I'm not sure thats what you're looking for. We use this script in CI to run the code coverage collection https://github.com/ros-planning/navigation2/blob/main/tools/code_coverage_report.bash and it should be outputting a file `total_coverage.info`.

@ruffsl why would that file be missing? We run the artifacts collection stage before we run the Codecov collection so is that the single reason why or should we also move that directory to be something that the Artifacts are picking up?

mmattb, 2023-01-14:
Typing on my phone, so apologies for the brevity. I'll look back at this in a few days.

> Semi-related question for my person education: I've used GDB to grab tracebacks when the cause is non-obvious in my development (my tutorial https://navigation.ros.org/tutorials/docs/get_backtrace.html) on crashes but haven't used it to try to debug shutdown failures before. Is there anything semantically different going on there or the same old? It honestly did not occur to me to apply the tool that way.

You can probably use 'break exit' or 'break _exit' or some such, then step forward from there. You can breakpoint signal handlers too in that way, which is what you'd want here.
The backtrace I gave you above came from boost stacktrace, which I stuck in the Timer code. I ran the test without gdb then. Handy trick I found out!

> On topic: It is interesting to note as I did in the longer comment that when I switch from non-composed bringup (e.g. 1 server 1 process) to composed bringup (e.g. all servers in 1 process) we did get a ton of our coverage results back. So I'm curious to know if the exception that's being raised on shutdown is unique to the Behavior Server where we see that issue. If its unique to this server, then that means there's something we can do in the code to fix it. Also, why it would be that some of the behavior server tests give us results if others don't? Might be another clue.

Yeah that is interesting. If my hypothesis is correct, then a.) We need to not be killing the global context before users of it have a chance to clean up; that would be a ROS bug I think; or b.) We need to ensure the bond code isnt making things during destruction, which may or may not be a Nav2 side fix. Whether or not this is our code cov issue, it certainly doesn't seem intended or correct? But I'm new here so im not sure!

mmattb, 2023-01-15:
Yeah, so this isn't the fix, but it gets us nearly full coverage on the assisted_teleop test. What it demonstrates is that that Bond destruction is the source of the problem. This test just means we drop the Bond on the ground and never break it properly.

It looks like on the ROS side there is a "preshutdown callback" which can be registered on the global Context, which gives us an opportunity to break bonds properly before things start going sideways. That could be the real fix. Thank you to the authors for that mechanism; they added it in 2021. https://github.com/ros2/rclcpp/issues/1706

```
diff --git a/nav2_util/src/lifecycle_node.cpp b/nav2_util/src/lifecycle_node.cpp
index 9868a38bbbe6cd29520162941623ec93702e3d08..ab24e103e5f92fd57354f48744913355aac31a08 100644
--- a/nav2_util/src/lifecycle_node.cpp
+++ b/nav2_util/src/lifecycle_node.cpp
@@ -41,11 +41,12 @@ LifecycleNode::LifecycleNode(
 LifecycleNode::~LifecycleNode()
 {
   RCLCPP_INFO(get_logger(), "Destroying");
   // In case this lifecycle node wasn't properly shut down, do it here
   if (get_current_state().id() ==
     lifecycle_msgs::msg::State::PRIMARY_STATE_ACTIVE)
   {
+    destroyBond();
     on_deactivate(get_current_state());
     on_cleanup(get_current_state());
   }
 }
@@ -67,9 +68,10 @@ void LifecycleNode::createBond()
 void LifecycleNode::destroyBond()
 {
   RCLCPP_INFO(get_logger(), "Destroying bond (%s) to lifecycle manager.", this->get_name());
 
   if (bond_) {
-    bond_.reset();
+    //bond_.reset();
+    bond_.release();
   }
 }
```

[image]

mmattb, 2023-01-16:
[image]

There we go. I haven't fixed all of the Exceptions, which means there is still some code cov we aren't seeing. Looks like it's going to be a case of whack-a-mole unfortunately.

You can see the commit for this first portion of the fix here. I'll put a PR up after I whack the other instances I've seen, and after writing some unit tests, but before doing anything else with the tests to catch these.

https://github.com/mmattb/navigation2/commit/4bda9f8b274e4231b1abf719e4e5e359bdbb2003

ruffsl, 2023-01-16:
> @ruffsl why would that file be missing? We run the artifacts collection stage before we run the Codecov collection so is that the single reason why or should we also move that directory to be something that the Artifacts are picking up?

The collection and upload of code coverage are steps that are invoked after the workspace caching and artifact upload (of only logs and test result files) steps are invoked. This was done so that code coverage could be handled separately depending upon if the job was a release build or a debug build. The coverage info file can still be downloaded from the job's codecov report page, as that is where it's uploaded (and tagged by respective rmw). I didn't see the need to duplicate the archival of duplicate report files across separate hosting platforms.

mmattb, 2023-01-18:
Great points @SteveMacenski.  I will say: some of these are stochastic bugs for whatever reason.  I see some of them every other run. It would also not shock me if gcov flushed on events other than just exit, s.t. we see some coverage sometimes.

Now I've whacked four of these shutdown bugs, which is making me want to check back in with you all to make sure I'm not just chasing ghosts in my local clone here...  I won't be shocked if you tell me that ROS and Nav2 never really prioritized getting clean shutdowns, and so we get these code cov issues as a result of that. I also won't be shocked if you tell me "Matthew, you aren't going to be able to fix all the shutdown issues" :)

```
/path/to/navigation2/install/nav2_costmap_2d/lib/nav2_costmap_2d/nav2_costmap_2d_cloud --ros-args -r __node:=costmap_2d_cloud -r voxel_grid:=local_costmap/voxel_grid
```
When I ran this, then used ctrl+c, I got a seg fault (stack below). Applying this change, the seg fault goes away:

```
diff --git a/nav2_costmap_2d/src/costmap_2d_cloud.cpp b/nav2_costmap_2d/src/costmap_2d_cloud.cpp
index 8c40709b..488fad88 100644
--- a/nav2_costmap_2d/src/costmap_2d_cloud.cpp
+++ b/nav2_costmap_2d/src/costmap_2d_cloud.cpp
@@ -225,6 +225,12 @@ int main(int argc, char ** argv)
     "voxel_grid", rclcpp::SystemDefaultsQoS(), voxelCallback);
 
   rclcpp::spin(g_node->get_node_base_interface());
+
+  // Needed for shutdown() to not puke
+  g_node.reset();
+  pub_marked.reset();
+  pub_unknown.reset();
+
   rclcpp::shutdown();
 
   return 0;
```

Could somebody else confirm that behavior, i.e. that they see the segfault on shutdown, and don't with my change?

I see at least one more of these issues to whack.  If you all think I'm not chasing ghosts here, I can keep chasing the others.

```
Thread 1 "nav2_costmap_2d" received signal SIGSEGV, Segmentation fault.
___pthread_mutex_lock (mutex=0x7e0) at ./nptl/pthread_mutex_lock.c:80
80	./nptl/pthread_mutex_lock.c: No such file or directory.
(gdb) bt
#0  ___pthread_mutex_lock (mutex=0x7e0) at ./nptl/pthread_mutex_lock.c:80
#1  0x00007ffff624e99e in eprosima::fastdds::dds::DomainParticipantImpl::find_type(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const ()
   from /home/mbryan/Projects/ros/ros2_rolling/install/fastrtps/lib/libfastrtps.so.2.9
#2  0x00007ffff6254f36 in eprosima::fastdds::dds::DomainParticipantImpl::unregister_type(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /home/mbryan/Projects/ros/ros2_rolling/install/fastrtps/lib/libfastrtps.so.2.9
#3  0x00007ffff690c910 in rmw_fastrtps_shared_cpp::remove_topic_and_type(CustomParticipantInfo const*, eprosima::fastdds::dds::TopicDescription const*, eprosima::fastdds::dds::TypeSupport const&) () from /home/mbryan/Projects/ros/ros2_rolling/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
#4  0x00007ffff68e5436 in rmw_fastrtps_shared_cpp::__rmw_destroy_service(char const*, rmw_node_s*, rmw_service_s*) ()
   from /home/mbryan/Projects/ros/ros2_rolling/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
#5  0x00007ffff69bcec1 in rmw_destroy_service () from /home/mbryan/Projects/ros/ros2_rolling/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
#6  0x00007ffff6c63a7f in rmw_destroy_service () from /home/mbryan/Projects/ros/ros2_rolling/install/rmw_implementation/lib/librmw_implementation.so
#7  0x00007ffff71ab617 in rcl_service_fini () from /home/mbryan/Projects/ros/ros2_rolling/install/rcl/lib/librcl.so
#8  0x00007ffff7cc52a7 in rclcpp::Service<rcl_interfaces::srv::ListParameters>::Service(std::shared_ptr<rcl_node_s>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::AnyServiceCallback<rcl_interfaces::srv::ListParameters>, rcl_service_options_s&)::{lambda(rcl_service_s*)#1}::operator()(rcl_service_s*) const ()
   from /home/mbryan/Projects/ros/ros2_rolling/install/rclcpp/lib/librclcpp.so
#9  0x00007ffff7cdadf4 in std::_Sp_counted_deleter<rcl_service_s*, rclcpp::Service<rcl_interfaces::srv::ListParameters>::Service(std::shared_ptr<rcl_node_s>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::AnyServiceCallback<rcl_interfaces::srv::ListParameters>, rcl_service_options_s&)::{lambda(rcl_service_s*)#1}, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() () from /home/mbryan/Projects/ros/ros2_rolling/install/rclcpp/lib/librclcpp.so
#10 0x0000555555592592 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
#11 0x00007ffff7b4de35 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() () from /home/mbryan/Projects/ros/ros2_rolling/install/rclcpp/lib/librclcpp.so
#12 0x00007ffff7bbac4c in std::__shared_ptr<rcl_service_s, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() () from /home/mbryan/Projects/ros/ros2_rolling/install/rclcpp/lib/librclcpp.so
#13 0x00007ffff7bbac6c in std::shared_ptr<rcl_service_s>::~shared_ptr() () from /home/mbryan/Projects/ros/ros2_rolling/install/rclcpp/lib/librclcpp.so
#14 0x00007ffff7cc2290 in rclcpp::ServiceBase::~ServiceBase() () from /home/mbryan/Projects/ros/ros2_rolling/install/rclcpp/lib/librclcpp.so
#15 0x00007ffff7cdc0fa in rclcpp::Service<rcl_interfaces::srv::ListParameters>::~Service() () from /home/mbryan/Projects/ros/ros2_rolling/install/rclcpp/lib/librclcpp.so
[... cut ...]
#42 0x00007ffff6ce3610 in __GI_exit (status=<optimized out>) at ./stdlib/exit.c:143
#43 0x00007ffff6cc7d97 in __libc_start_call_main (main=main@entry=0x555555580860 <main>, argc=argc@entry=6, argv=argv@entry=0x7ffffffec158) at ../sysdeps/nptl/libc_start_call_main.h:74
#44 0x00007ffff6cc7e40 in __libc_start_main_impl (main=0x555555580860 <main>, argc=6, argv=0x7ffffffec158, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7ffffffec148) at ../csu/libc-start.c:392
#45 0x00005555555812d5 in _start ()
```

mmattb, 2023-01-18:
>>> We should be calling that in all of the task server nodes [on deactivation](https://github.com/ros-planning/navigation2/blob/2e652abb9c09c77fafd6acd83651080888960784/nav2_behaviors/src/behavior_server.cpp#L209-L222) but maybe these experiments aren't bringing the lifecycle back down to a safe state? If so, then that destruction isn't called so adding it to the LifecycleNode's destructor seems like a good logical option so that its called either way.

That's exactly the idea, yeah, except that the destructor fires too late (after rclcpp::shutdown()) in the case that we are letting the dtor fire at scope end. So: we need to leverage preshutdown callbacks to do the cleanup where onDeactivate isn't being called (which is presumably some of these SIGINT cases).

Feel free to add me as some sort of Collaborator with restricted permissions if you want to assign this to me!

mmattb, 2023-01-18:
```
  0 [1169.514017] (nav2_system_tests) StdoutLine: {'line': b"27: [ERROR] [component_container_isolated-4]: process[component_cont    ainer_isolated-4] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'\n"}
  1 [1169.515275] (nav2_system_tests) StdoutLine: {'line': b"27: [INFO] [component_container_isolated-4]: sending signal 'SIGKILL    ' to process[component_container_isolated-4]\n"}
```

With composition=True I seem to hit https://github.com/ros2/rclcpp/issues/2083.  So it seems we have at least two issues, and the composition change worked around one of them.

SteveMacenski, 2023-01-18:
```
steve@mo:~/Documents/nav2_ws$ ros2 run nav2_costmap_2d nav2_costmap_2d_cloud 
[WARN] [1674074046.227370863] [rclcpp]: A zero depth with KEEP_LAST doesn't make sense; no data could be stored.This will be interpreted as SYSTEM_DEFAULT
[WARN] [1674074046.227459748] [rclcpp]: A zero depth with KEEP_LAST doesn't make sense; no data could be stored.This will be interpreted as SYSTEM_DEFAULT
^C[INFO] [1674074058.038445433] [rclcpp]: signal_handler(signum=2)

```

Appears to exit cleanly for me when I just run it on its own. Were you throwing data through it or just launching in isolation?

mmattb, 2023-01-19:
@SteveMacenski I wasn't throwing data at it, but I was launching it with the line I showed in https://github.com/ros-planning/navigation2/issues/3271#issuecomment-1386685834

The line I show is how the test apprars to launch it, rather than ros run.

SteveMacenski, 2023-01-19:
No, that exits cleanly for me as well as `ros2 run`

mmattb, 2023-01-19:
Weird! Thanks for trying that. I'm not sure what is different in my case.  "We aren't shipping your machine".

I'll propose a PR with my current set of fixes, and see if the tests pass and code cov looks good through CI. Do I just make a PR to main, and CI will run automatically?

SteveMacenski, 2023-01-19:
Yes, if you open a PR, it'll trigger the jobs and you can see the CI results in CircleCI's pop ups. We have some tests failing due to issues in upstream libraries so the codecov prompt doesn't auto populate when test failures happen, but you can see it on https://codecov.io/gh/ros-planning/navigation2 selecting your PR branch on the drop down

mmattb, 2023-01-20:
Looks like in the PR automation the code cov drastically improved on the plugins, as intended. This theta star stuff got worse, but I see that the theta star tests failed, and not only on this PR.

I still see one occasional failure in my sandbox which causes code cov loss, but it's stochastic unfortunately.  Consider the PR WiP.

mmattb, 2023-01-21:
I thought so too, but it looks like the waypoint follower miss does happen sometimes. It's not consistent:

https://app.codecov.io/gh/ros-planning/navigation2/pull/3374

Likewise with oscillation.cpp:

https://app.codecov.io/gh/ros-planning/navigation2/pull/3372

Clearly the fixes I have here help, but are not complete. Shall I open issues for these other ones?

SteveMacenski, 2023-02-01:
Thanks to @mmattb for some really awesome debugging and fixes that took us all the way to rclcpp with some issues!

closed 2023-02-01 by SteveMacenski

mmattb, 2023-02-02:
> Thanks to @mmattb for some really awesome debugging and fixes that took us all the way to rclcpp with some issues!

It was a pleasure Steve! Thanks for letting me jump in.