ros-navigation/navigation2 #3277: [DWB] Question about StandardTrajectoryGenerator::generateTrajectory in ROS2

oym1994, 2022-11-10:
dwb_msgs::msg::Trajectory2D StandardTrajectoryGenerator::generateTrajectory(
  const geometry_msgs::msg::Pose2D & start_pose,
  const nav_2d_msgs::msg::Twist2D & start_vel,
  const nav_2d_msgs::msg::Twist2D & cmd_vel)
{
  dwb_msgs::msg::Trajectory2D traj;
  traj.velocity = cmd_vel;
  //  simulate the trajectory
  geometry_msgs::msg::Pose2D pose = start_pose;
  nav_2d_msgs::msg::Twist2D vel = start_vel;
  double running_time = 0.0;
  std::vector<double> steps = getTimeSteps(cmd_vel);
  traj.poses.push_back(start_pose);
  for (double dt : steps) {
    //  calculate velocities
    vel = computeNewVelocity(cmd_vel, vel, dt);

    //  update the position of the robot using the velocities passed in
    pose = computeNewPosition(pose, vel, dt);

    traj.poses.push_back(pose);
    traj.time_offsets.push_back(rclcpp::Duration::from_seconds(running_time));
    running_time += dt;
  }  //  end for simulation steps

  if (include_last_point_) {
    traj.poses.push_back(pose);
    traj.time_offsets.push_back(rclcpp::Duration::from_seconds(running_time));
  }

  return traj;
}

Here why set traj.velocity = cmd_vel,  instead set traj.velocity to be the first sampling velocity in the for loop?  The experiment shows that the result velocity may be against the acceleration limit.

SteveMacenski, 2022-11-10:
In general, don't copy paste code, just give a link to it in the github interface for viewing.

I don't see any issue with this. Its starting from the initial (eg current) velocity and only using the `traj.velocity` / `cmd_vel` for defining the maximum velocity to achieve over the course of the trajectory using the proper acceleration limits established in `computeNewVelocity`. The `traj.velocity` is mostly used for storage for scoring and recovering out the actual cmd_vel that generated the trajectory that was rolled out in the loop you provide the code for.

oym1994, 2022-11-11:
Thank for your kind reply.
It will not influence the trajectory, just influence the velocity result. In my experiment, if the robot start to move, it may give
a large velocity at the beginning, which may conflict with the acceleration limits. Then I changed to use the first velocity of the trajectory, then it becomes OK. While by using "traj.velocity = cmd_vel;", it equals to use the first velocity of the trajectory.

SteveMacenski, 2022-11-11:
Cmd_vel is the velocity represented by the trajectory to be scored. We can see where the current speed is used to integrate towards the desired cmd_vel in the code. So I think you need to show some proof of that claim based on the acceleration parameters set by the user

oym1994, 2022-11-15:
Thank u again!
Below is a comparison:
**Original logic where the target velocity is utilized:**
[image]

**Let the cmd_vel equal to the first velocity of the sampling trajectory:**
[image]

some related configurations:
max_speed_xy: 1.5
max_vel_x: 1.5
max_vel_y: 0.0
max_vel_x_backwards: 0.2
acc_lim_x: 3.0
acc_lim_y: 0.0
acc_lim_theta: 3.0
sim_time: 2.0

From the velocity and cmd_vel curve, we can see that with the original logic,  the velocity would increase rapidly(especially at the beginning), which may make the acceleration to exceed the configuration.

SteveMacenski, 2022-11-15:
Can you show numerical evidence that it exceeds acceleration limits? I again looked at the code and again see no error.

oym1994, 2022-11-16:
At the beginning, the linear speed is 0（red curve）, and the output cmd_vel.linear.x(light green curve) is about 1.3m/s, which means it want to increase the speed from 0 to 1.3 at once.  The same to the cmd_vel.angular.z and angular speed (blue and purple).
From the code logic, the target velocity is the final velocity during the sim_time. It is valid to go to the target velocity in sim_time(such as 2s). But it may be invalid in control duration dt( such as 0.1 s).

SteveMacenski, 2022-11-16:
I see the first cycle achieving ~0.3m/s (guestimating on your graph image) in the first cycle while the odometry is reading 0. You mention before that your acceleration is set to 3 and your time dt is 0.1. `v = v0 + at` tells me that the max achievable is 0.3m/s, which seems to check out. 

Then it stays flat at that same ~0.3m/s (which is probably exactly 0.3m/s if you had the direct data) as long as the odometry is at 0m/s and then a little bit after for the controller to catch up (that delay in cmd_vel at 1347) for which then the commanded velocity spikes complimentary. Lets say that the first peak just after we move from ~0.3m/s is ~0.65m/s and the odometry was reading ~0.45m/s.

`v = v0 + at` says that we can achieve `0.45m/s + 0.1 * 3` = 0.95, for which the ~0.65 m/s is well within.

Again, even analyzing your data, this all looks absolutely correct to me.

oym1994, 2022-11-17:
I think the graph u mentioned is the second one. I have modified a logic here as below. Using the original logic, u can get the first graph, where u can see rapid linear and rotation speed acceleration.

```cpp
dwb_msgs::msg::Trajectory2D StandardTrajectoryGenerator::generateTrajectory(
  const geometry_msgs::msg::Pose2D & start_pose,
  const nav_2d_msgs::msg::Twist2D & start_vel,
  const nav_2d_msgs::msg::Twist2D & cmd_vel)
{
  dwb_msgs::msg::Trajectory2D traj;
  **traj.velocity = cmd_vel;**     ---delete this
  //  simulate the trajectory
  geometry_msgs::msg::Pose2D pose = start_pose;
  nav_2d_msgs::msg::Twist2D vel = start_vel;
  double running_time = 0.0;
  std::vector<double> steps = getTimeSteps(cmd_vel);
  traj.poses.push_back(start_pose);
  bool getFirstVel = false; ---add this
  for (double dt : steps) {
    //  calculate velocities
    vel = computeNewVelocity(cmd_vel, vel, dt);
    if(!getFirstVel ){  ---add this
      getFirstVel  = false;
      traj.velocity = vel;
    }
    //  update the position of the robot using the velocities passed in
    pose = computeNewPosition(pose, vel, dt);

    traj.poses.push_back(pose);
    traj.time_offsets.push_back(rclcpp::Duration::from_seconds(running_time));
    running_time += dt;
  }  //  end for simulation steps

  if (include_last_point_) {
    traj.poses.push_back(pose);
    traj.time_offsets.push_back(rclcpp::Duration::from_seconds(running_time));
  }

  return traj;
}
```

SteveMacenski, 2022-11-17:
`traj.velocity` **must** be the value of the `cmd_vel` that is requested to be scored and generated because its recovered [here](https://github.com/ros-planning/navigation2/blob/987fe905ec58a97ec85c485a1733ead585a5bc7f/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp#L323) to send that trajectory to the base after scoring. The role of the function `generateTrajectory()` is to take in a proposed command velocity that should already be in the dynamic window of feasible velocities from the current velocity (via the [trajectory generator itself](https://github.com/ros-planning/navigation2/blob/987fe905ec58a97ec85c485a1733ead585a5bc7f/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp#L370)) and roll it out into poses by dynamic limits (e.g. `computeNewVelocity()`). It is not the role of this function to change this velocity. 

Its understood that in the immediate time-step its not guaranteed to be drivable, that's why `computeNewVelocity()` rolls out using the acceleration limits to converge to the requested `cmd_vel` over the course of projecting the future trajectory that would be generated if that command was sent for that time period. The trajectory generator using the iterators finds the max/min possible velocities achievable over the simulation time set (default 1.7s) to generate the trajectories over in the dynamic window. So they're all achievable within that simulation time.

oym1994, 2022-11-21:
Thanks for your patient and kind response. Hope this analysis could really help!

oym1994, 2022-12-08:
Sorry for late response. What kind of evidence do u need?  May the second curve chart prove that?

SteveMacenski, 2022-12-08:
Yeah, running on some robots and making sure that the acceleration that the robot is achieving is in line with the parameter being set (e.g. isn't substantially undershooting). Something like a robot from stand still to fill speed forward and checking the acceleration on the plot with the setting. Then change the settings of acceleration to a couple of different things just to make sure the trend holds

SteveMacenski, 2023-06-05:
Hey! Just following up here after some time, did you get a chance to test / verify?

oym1994, 2023-07-10:
> Hey! Just following up here after some time, did you get a chance to test / verify?

Hi, here is a comparison. The acc_lim_x are all set to be 1.0

[image]

For the original version(Show above), the real acceleration is about  3.25m/s^2, and the cmd_vel rapidly grow from 0.0 to 0.7 in about 0.5ms

[image]
For the revised version(Show above), the real acceleration is about 0.9m/m^2  which seems normal.

The max_speed is too slow(set to be 1.0m/s). I need to tune another parameter. But in my first experiment, the max speed is OK!

SteveMacenski, 2023-07-10:
Great! Can you submit the PR!

oym1994, 2023-08-05:
Ok, I will submit such PR at my convenient time. This is my first time to do that, any advice or notice?

SteveMacenski, 2023-08-07:
Just make sure to fill in the PR template when it pops up with all the details + make sure there's test coverage if any new code is added to make sure its exercising and working properly. Also, base against the `main` branch, and I'll backport if possible to existing distributions. That's really it :-)

SteveMacenski, 2024-09-12:
PR #4663 resolves and will be merged shortly with some documentation updates

closed 2024-09-12 by SteveMacenski