ros-navigation/navigation2 #3281: Ceres Solver rosdep does not cope well with colcon builds

roncapat, 2022-11-14:
Nav2 Constrained Smoother requires Ceres solver through this rosdep alias:

https://github.com/ros-planning/navigation2/blob/90ae5cc1d539b715a9d2cad77fd98a2f29aa5361/nav2_constrained_smoother/package.xml#L20

However, Ceres Solver is a Colcon package with its own package name. 

https://github.com/ceres-solver/ceres-solver/blob/37a3cb38414002c3dd8c76d152b483ce12f9611c/package.xml#L32

`<name>ceres-solver</name>`

So what happens is that, if I don't have yet installed Ceres in my own system and instead throw together navigation2 source folder and ceres source folder in the same colcon build session, colcon does not recognises that it has to build ceres first and provide to navigation2 visibility of the being built ceres package. 

At least, this is my experience. OF course, I could have installed ceres from repository or rosdep and everything would work out of the box. But maybe, if `libceres-dev` could be renamed in `ceres-solver` in the rosdep index and navigation2 package.xml files, this would cope well with such "from source" scenarios.

SteveMacenski, 2022-11-14:
The rosdep key is `libceres-dev` as you can see at: https://github.com/ros/rosdistro/blob/b92727cc08b7068982342202b5ad9eddb012a4cd/rosdep/base.yaml#L2983. I can't speak to that name provided in the repository, but that's not what's actually released in binary format for users to automatically install the package without being forced to build from source Ceres (and likely other dependencies).

I can understand your issue, but perhaps you should have an underlay workspace with ceres so that its already built and you can work with your main workspace freely.

SteveMacenski, 2022-11-18:
Closing due to non-response and doesn't seem to be a problem we can solve here

closed 2022-11-18 by SteveMacenski