ros-navigation/navigation2 #3285: Theta star planner fails to get path when goals falls in unknown space where tile value = NO_INFORMATION

pepisg, 2022-11-15:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 22
- ROS2 Version:
  - Humble
- Version or commit hash:
  - 0d86469 (latest humble branch)
  - cyclone DDS

#### Steps to reproduce issue
Launch the turtlebot3 simulation with these paramters:

```.yaml
amcl:
  ros__parameters:
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

bt_navigator:
  ros__parameters:
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.
    plugin_lib_names:
    - nav2_compute_path_to_pose_action_bt_node
    - nav2_compute_path_through_poses_action_bt_node
[... 267 lines cut ...]
    max_velocity: [0.26, 0.0, 1.0]
    min_velocity: [-0.26, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0
```

They are basically the default parameters but they use a rolling global costmap without a static layer (so we can easily send the robot to an unknown tile) and the defaults for the Theta Star planner.

Then try to send the robot to a place marked as unknown in the global costmap.

#### Expected behavior

The theta star planner generates a path to the requested goal even if it lies on a tile whose value is NO_INFORMATION,

#### Actual behavior

The theta star planner fails to compute a path and shows the following error message: `Either of the start or goal pose are an obstacle! `

#### Additional information

This is likely due to the way a [safe tile is defined in the theta star planner](https://github.com/ros-planning/navigation2/blob/main/nav2_theta_star_planner/include/nav2_theta_star_planner/theta_star.hpp#L92), where any cost above LETHAL_OBSTACLE (defined as 252) is treated as unsafe; howerver NO_INFORMATION tiles are set to 255, which is in fact greater than 252 but should not be considered as an obstacle.

SteveMacenski, 2022-11-15:
Sounds like this planner could use a `allow_unknown` option like NavFn and Smac Planners have! Sounds like a great contribution :smile:

closed 2022-11-16 by SteveMacenski

tosa-no-onchan, 2024-09-18:
Hi, I had same error.
But I copy https://github.com/ros-navigation/navigation2/tree/main/nav2_theta_star_planner describe.
It goes well.
``````
planner_server:
  ros__parameters:
    planner_plugin_types: ["nav2_theta_star_planner::ThetaStarPlanner"]
    planner_plugin_ids: ["GridBased"]
    GridBased:
      how_many_corners: 8
      w_euc_cost: 1.0
      w_traversal_cost: 2.0
``````