ros-navigation/navigation2 #3287: is with QOS profile

stevedanomodolor, 2022-11-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 docker
- ROS2 Version:
  - rolling
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue
``` 
colcon build --symlink-install
```

#### Expected behavior
I am currently using the nav2 docker images which has ubuntu jammy. 
```
root@stevedan-OMEN-Laptop-15-ek0xxx:/opt/overlay_ws# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

```
Tried the fix suggested by https://github.com/ros-planning/navigation2/issues/3212, still does work
This should build without error
#### Actual behavior
Nav2 utils is failing 
```

Finished <<< nav2_simple_commander [0.79s]
--- stderr: nav2_util                              
In file included from /opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /opt/overlay_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp: In instantiation of ‘nav2_util::ServiceClient<ServiceT>::ServiceClient(const string&, const SharedPtr&) [with ServiceT = lifecycle_msgs::srv::ChangeState; std::string = std::__cxx11::basic_string<char>; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>]’:
/opt/overlay_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:35:3:   required from here
/opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:49:15: error: cannot convert ‘rclcpp::SystemDefaultsQoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   49 |       rclcpp::SystemDefaultsQoS(),
      |               ^~~~~~~~~~~~~~~~~~~
      |               |
      |               rclcpp::SystemDefaultsQoS
In file included from /opt/ros/rolling/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /opt/overlay_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/opt/ros/rolling/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = lifecycle_msgs::srv::ChangeState; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<lifecycle_msgs::srv::ChangeState> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
  127 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
In file included from /opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /opt/overlay_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp: In instantiation of ‘nav2_util::ServiceClient<ServiceT>::ServiceClient(const string&, const SharedPtr&) [with ServiceT = lifecycle_msgs::srv::GetState; std::string = std::__cxx11::basic_string<char>; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>]’:
/opt/overlay_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:36:3:   required from here
/opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:49:15: error: cannot convert ‘rclcpp::SystemDefaultsQoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   49 |       rclcpp::SystemDefaultsQoS(),
      |               ^~~~~~~~~~~~~~~~~~~
      |               |
      |               rclcpp::SystemDefaultsQoS
In file included from /opt/ros/rolling/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /opt/overlay_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/opt/ros/rolling/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = lifecycle_msgs::srv::GetState; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<lifecycle_msgs::srv::GetState> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
  127 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
gmake[2]: *** [src/CMakeFiles/nav2_util_core.dir/build.make:104: src/CMakeFiles/nav2_util_core.dir/lifecycle_service_client.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:210: src/CMakeFiles/nav2_util_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav2_util [2.92s, exited with code 2]
```

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2022-11-16:
Are you using our docker images? If not, what's the difference? Is your Rolling up to date? 

In general though, lets follow up on ROS Answers. This isn't a bug with the project since our CI is turning over find in our docker containers on the main branch with ROS 2 rolling derived image. If you link your question here, I'll follow up with you there.

closed 2022-11-16 by SteveMacenski

jlblancoc, 2023-03-05:
@SteveMacenski 

I confirm this is still a real issue, as of today, on:
- Ubuntu 22.04 (host, no docker image)
- ROS 2 (Humble) installed via apt
- Nav2 cloned from git, `humble` branch.

Digging into it, I found the reason: this rclcpp change https://github.com/ros2/rclcpp/commit/f6056beaa09ec2c19a6cc3b74405e2b83948f161 has never been released to humble, this Nav2 commit https://github.com/ros-planning/navigation2/commit/2b40aaf99768859377dded3914a664fa1d57b82f in main fixes the rclcpp API change, but then at some point that commit was backported to the humble branch too, but for humble the API change never arrived...

Here's the relevant part of the header for rclcpp (for v16.0.3, the latest for humble):

https://github.com/ros2/rclcpp/blob/16.0.3/rclcpp/include/rclcpp/node_impl.hpp#L123-L137

You can see the `rmw_qos_profile_t`.

So, the key is: is there any reason rclcpp for humble is stuck at v16.0.3 (current is v19.3.0)?? 
1) It's not intentional: request rclcpp maintainers to make a new release.
2) It is indeed intentional (as I guess): then we should, IMO, fix the nav2/humble branch.

If we are in (2), I'll be glad to do the PR. 

However, I can't understand why the [builds pass](https://build.ros2.org/job/Hbin_uJ64__nav2_util__ubuntu_jammy_amd64__binary/31/consoleFull) in the ROS2 build farm, if in theory they have the same rclcpp 16.0.3 (!!).

Cheers,

TZECHIN6, 2023-03-06:
Faced a same issue with you @jlblancoc, nav2_util cannot be built due to the QoS.

chenzhiwo, 2023-03-06:
Having same issue.

stevedanomodolor, 2023-03-06:
The way i solved this was to update the docker I was using. That seemed to worked.

jlblancoc, 2023-03-06:
> The way i solved this was to update the docker I was using. That seemed to worked.

But that's in rolling, where the released version of rclcpp is much newer so it makes sense. 
The issue seems to apply to Humble only... except for the mystifying success to build in the build farm.

TZECHIN6, 2023-03-06:
How about not docker installation? Seem this only happens in humble, doesn't this should be a LTS release? I expect a stable release in this distro

momokang0614, 2023-03-06:
same issue，Ubuntu22.04 and use navigation2 on branch humble

momokang0614, 2023-03-06:
> @SteveMacenski
> 
> I confirm this is still a real issue, as of today, on:
> 
> * Ubuntu 22.04 (host, no docker image)
> * ROS 2 (Humble) installed via apt
> * Nav2 cloned from git, `humble` branch.
> 
> Digging into it, I found the reason: this rclcpp change [ros2/rclcpp@f6056be](https://github.com/ros2/rclcpp/commit/f6056beaa09ec2c19a6cc3b74405e2b83948f161) has never been released to humble, this Nav2 commit [2b40aaf](https://github.com/ros-planning/navigation2/commit/2b40aaf99768859377dded3914a664fa1d57b82f) in main fixes the rclcpp API change, but then at some point that commit was backported to the humble branch too, but for humble the API change never arrived...
> 
> Here's the relevant part of the header for rclcpp (for v16.0.3, the latest for humble):
> 
> https://github.com/ros2/rclcpp/blob/16.0.3/rclcpp/include/rclcpp/node_impl.hpp#L123-L137
> 
> You can see the `rmw_qos_profile_t`.
> 
> So, the key is: is there any reason rclcpp for humble is stuck at v16.0.3 (current is v19.3.0)??
> 
> 1. It's not intentional: request rclcpp maintainers to make a new release.
> 2. It is indeed intentional (as I guess): then we should, IMO, fix the nav2/humble branch.
> 
> If we are in (2), I'll be glad to do the PR.
> 
> However, I can't understand why the [builds pass](https://build.ros2.org/job/Hbin_uJ64__nav2_util__ubuntu_jammy_amd64__binary/31/consoleFull) in the ROS2 build farm, if in theory they have the same rclcpp 16.0.3 (!!).
> 
> Cheers,

Have you found a solution? Branch humble in Ubuntu 22.04

tonynajjar, 2023-03-06:
Until the reviewers can take care of it, for those of you building from source and wanting a quick fix, replace every instance of `rclcpp::ServicesQoS()` in Nav2 with `rmw_qos_profile_services_default`. Hope that helps temporarily

mini-1235, 2024-03-03:
Had the same issues, checking out to the humble branch and did some modifications to the Dockerfile solves :(