ros-navigation/navigation2 #3298: NAV2 Getting Started Example doesn't work for ROS Humble

Steven7099, 2022-11-23:
I'm following the NAV2 tutorial  step by step and can't get past the first example.
https://navigation.ros.org/getting_started/index.html 

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble

I have everything installed:
```
sudo apt install ros-humble-navigation2
sudo apt install ros-humble-nav2-bringup
sudo apt install ros-humble-turtlebot3*
```
Then I set the key environment variables:

```
source /opt/ros/humble/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
```

Then after running:

`ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False`

I rush to rviz, set the 2D pose estimation as quick as I can and end up with this in the terminal:

```
[component_container_isolated-6] [INFO] [1669122316.072762902] [amcl]: Creating bond (amcl) to lifecycle manager.
[rviz2-5] [INFO] [1669122320.120000768] [rviz2]: Setting estimate pose: Frame:map, Position(-2.97266, -0.58257, 0), Orientation(0, 0, 0.0273898, 0.999625) = Angle: 0.0547864
[spawn_entity.py-3] [ERROR] [1669122345.950231796] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-3] [ERROR] [1669122345.950698437] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-3]: process has died [pid 20529, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity turtlebot3_waffle -file /opt/ros/humble/share/nav2_bringup/worlds/waffle.model -robot_namespace -x -2.00 -y -0.50 -z 0.01 -R 0.00 -P 0.00 -Y 0.00 --ros-args'].
```

In RViz I see the [white map ](https://navigation.ros.org/getting_started/index.html#running-the-example) but it doesn't say shut down in the bottom left corner and theres also nothing in Gazebo.

I should be seeing the map and LiDAR in both [RViz and Gazebo.](https://navigation.ros.org/_images/navigation_with_recovery_behaviours.gif)

But this is what it looks like:

[image]

padhupradheep, 2022-11-24:
Just giving some hints on the issue..

On a quick dig I found that, for some reason the sandbox world model is not loaded.. I tried to spawn the robot urdf in the empty world it works fine.. 

I even tried the [turtlebot launch from the official repo ](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/humble-devel/turtlebot3_gazebo/launch/turtlebot3_world.launch.py) and it seems to not work as well..

AlexeyMerzlyakov, 2022-11-25:
I've reproduced the same (and not only) situations on my local Humble release, installed from the scratch yesterday.
The problems are most probably related to FastDDS known issues (#3033, #3014),  causing nodes to not bringup, or to miss some messages, which previously I've observed with it. [Switching to](https://docs.ros.org/en/humble/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html) CycloneDDS will resolve the situation:
```
$ sudo apt install ros-humble-rmw-cyclonedds-cpp
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

| Run | Status (FastDDS) | Status (CycloneDDS) |
| --- | ---------------- | ------------------- |
| 1 | Map FAIL | PASS |
| 2 | PASS | PASS |
| 3 | Controller FAIL | PASS |
| 4 | Map FAIL, Spawner FAIL | PASS |
| 5 | Spawner FAIL | PASS |
| 6 | Spawner FAIL | PASS |
| 7 | Spawner FAIL | PASS |

padhupradheep, 2022-11-25:
Hmm! There was a recent release for humble today, with all the new updates and changing to cyclonedds, does not resolve the problem for me atleast.. :cry:

VishalDhayalan, 2022-11-28:
The same happens to me (Gazebo is blank and there is no robot in RViZ) now and then (not rarely though). Stopping both processes and running it again makes it come up correctly (most times). I think there is not enough timeout or leeway with the defaults somewhere...

VishalDhayalan, 2022-11-28:
Update: I'm running into this issue 100% of the time when trying to run it with the SLAM toolbox...

In one terminal I run ros2 launch nav2_bringup slam_launch.py
and in another, I run ros2 launch nav2_bringup tb3_simulation_launch.py

but the second command shows either a blank gazebo or the gazebo world with no robot and RViZ just shows the white map with no robot... I'm constantly getting these error messages in the terminal where tb3_simulation_launch.py was run:

[image]

I'm new to ROS and Gazebo, so I'm not sure what's going on or how to fix this :(

AlexeyMerzlyakov, 2022-11-30:
From the logs, it is definitely Gazebo (or its related environment) problem. `gzserver` does not start well, so everything else will be failed:
```
[leha@leha-PC 2022.11.25_Humble_issue]$ cat run_github_fail.log | grep gz
[INFO] [gzserver-1]: process started with pid [20090]
[INFO] [gzclient-2]: process started with pid [20092]
```

But it should be:
```
[leha@leha-PC 2022.11.25_Humble_issue]$ cat run_local_pass.log | grep gz
[INFO] [gzserver-1]: process started with pid [12989]
[INFO] [gzclient-2]: process started with pid [12991]
[gzserver-1] [INFO] [1669353463.780707641] [intel_realsense_r200_depth_driver]: Publishing camera info to [/intel_realsense_r200_depth/camera_info]
[gzserver-1] [INFO] [1669353463.789291208] [intel_realsense_r200_depth_driver]: Publishing depth camera info to [/intel_realsense_r200_depth/depth/camera_info]
[gzserver-1] [INFO] [1669353463.791490745] [intel_realsense_r200_depth_driver]: Publishing pointcloud to [/intel_realsense_r200_depth/points]
[gzserver-1] [INFO] [1669353463.932283684] [turtlebot3_diff_drive]: Wheel pair 1 separation set to [0.287000m]
[gzserver-1] [INFO] [1669353463.932363457] [turtlebot3_diff_drive]: Wheel pair 1 diameter set to [0.066000m]
[gzserver-1] [INFO] [1669353463.938431238] [turtlebot3_diff_drive]: Subscribed to [/cmd_vel]
[gzserver-1] [INFO] [1669353463.961495468] [turtlebot3_diff_drive]: Advertise odometry on [/odom]
[gzserver-1] [INFO] [1669353463.986576876] [turtlebot3_diff_drive]: Publishing odom transforms between [odom] and [base_footprint]
[gzserver-1] [INFO] [1669353464.094009606] [turtlebot3_joint_state]: Going to publish joint [wheel_left_joint]
[gzserver-1] [INFO] [1669353464.094815859] [turtlebot3_joint_state]: Going to publish joint [wheel_right_joint]
[INFO] [gzclient-2]: sending signal 'SIGINT' to process[gzclient-2]
[INFO] [gzserver-1]: sending signal 'SIGINT' to process[gzserver-1]
[INFO] [gzclient-2]: process has finished cleanly [pid 12991]
[INFO] [gzserver-1]: process has finished cleanly [pid 12989]
```

If you are running the Gazebo first time, it could take for a long time (up to ~2-5 minutes) to start the Gazebo. During this time, Gazebo is downloading the models and could show empty or even blank screen. Once Gazebo has been downloaded all the models and started, you might close it and set environment variable `export GAZEBO_MODEL_DATABASE_URI=""` to empty; and then re-run nav2 stack demonstration. This will skip the downloading step every next time and will bring-up the Gazebo quickly.

Also, to get more information about the gazebo issues, you might pass the `--verbose` option to the gzserver, as follows:
```
--- /opt/ros/humble/share/nav2_bringup/launch/tb3_simulation_launch.py.orig	2022-11-30 13:14:22.254025224 +0300
+++ /opt/ros/humble/share/nav2_bringup/launch/tb3_simulation_launch.py	2022-11-30 13:15:53.137470923 +0300
@@ -159,7 +159,7 @@
     # Specify the actions
     start_gazebo_server_cmd = ExecuteProcess(
         condition=IfCondition(use_simulator),
-        cmd=['gzserver', '-s', 'libgazebo_ros_init.so',
+        cmd=['gzserver', '--verbose', '-s', 'libgazebo_ros_init.so',
              '-s', 'libgazebo_ros_factory.so', world],
         cwd=[launch_dir], output='screen')
```
Does it changes something in the output?

Also, did you check that `/opt/ros/humble/share/turtlebot3_gazebo/models` path you are referencing by `GAZEBO_MODEL_PATH` contains correct waffle models? Do you have fresh installation of ROS2 humble (w/o any residuals from previous ROS2 releases)? Also, please check the availability and relevance of all Gazebo packages in the system:
```
leha@leha-VB:~$ dpkg --list | grep -i gazebo | awk '{print $1,$2,$3}'
ii gazebo 11.10.2+dfsg-1
ii gazebo-common 11.10.2+dfsg-1
ii gazebo-plugin-base 11.10.2+dfsg-1
ii libgazebo-dev 11.10.2+dfsg-1
ii libgazebo11:amd64 11.10.2+dfsg-1
ii libignition-cmake2-dev:amd64 2.14.0-2~jammy
ii libignition-math6:amd64 6.12.0-3~jammy
ii libignition-math6-dev:amd64 6.12.0-3~jammy
ii ros-humble-gazebo-dev 3.7.0-1jammy.20221108.145559
ii ros-humble-gazebo-msgs 3.7.0-1jammy.20221108.173247
ii ros-humble-gazebo-plugins 3.7.0-1jammy.20221108.191431
ii ros-humble-gazebo-ros 3.7.0-1jammy.20221108.182840
ii ros-humble-gazebo-ros-pkgs 3.7.0-1jammy.20221108.201916
ii ros-humble-turtlebot3-gazebo 2.2.5-3jammy.20221108.201949
ii ros-humble-turtlebot3-gazebo-dbgsym 2.2.5-3jammy.20221108.201949
```

Steven7099, 2022-12-02:
I was able to get things to work by simple running Gazebo from the terminal.

 `$ gazebo`

Waited for it to boot up, then shut it down.

Then running:

`ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False`

Worked for me but still seems odd, it doesn't seem like a solution.

SteveMacenski, 2022-12-02:
This is a first time initialization thing for a gazebo, not related to navigation.

closed 2022-12-02 by SteveMacenski

padhupradheep, 2022-12-02:
Just an update from my side, it works without an issue after the recent release in humble..

animesh-singhal, 2023-06-14:
I am experiencing the same issue mentioned in this GitHub thread. Currently, the only workaround I have is to execute the "ros2 launch" command twice. It fails on the first attempt, but runs successfully on the second try. Is there a solution to make it work correctly on the initial run?
PS: Simply running 'gazebo' alone does not prevent the failure I encounter when executing the launch command for the first time.