ros-navigation/navigation2 #3304: AssistedTeleop Interface missing from Ubuntu Humble Nav2 Binary

austin-InDro, 2022-11-30:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 22.04
- ROS2 Version:
Humble - Binaries
- Version or commit hash:
ros-humble-navigation2  1.1.3-1jammy.20221110.233517
- DDS implementation:
CycloneDDS

#### Steps to reproduce issue
In Python3 cli
```
import from nav2_simple_commander.robot_navigator import BasicNavigator
```

#### Expected behavior
Import is successful
#### Actual behavior
```bash
>>> from nav2_simple_commander.robot_navigator import BasicNavigator
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/humble/lib/python3.10/site-packages/nav2_simple_commander/robot_navigator.py", line 26, in <module>
    from nav2_msgs.action import AssistedTeleop, BackUp, Spin
ImportError: cannot import name 'AssistedTeleop' from 'nav2_msgs.action' (/home/austin/Documents/2ws/install/nav2_msgs/local/lib/python3.10/dist-packages/nav2_msgs/action/__init__.py)
```
#### Additional information
It seems AssistedTeleop is currently missing from the interfaces.

Performing a `ros2 interface show nav2_msgs/` while autocompleting yields:
```bash
ros2 interface show nav2_msgs/
nav2_msgs/action/BackUp                   nav2_msgs/action/NavigateToPose           nav2_msgs/msg/CostmapMetaData             nav2_msgs/srv/GetCostmap
nav2_msgs/action/ComputePathThroughPoses  nav2_msgs/action/SmoothPath               nav2_msgs/msg/Particle                    nav2_msgs/srv/IsPathValid
nav2_msgs/action/ComputePathToPose        nav2_msgs/action/Spin                     nav2_msgs/msg/ParticleCloud               nav2_msgs/srv/LoadMap
nav2_msgs/action/DriveOnHeading           nav2_msgs/action/Wait                     nav2_msgs/msg/SpeedLimit                  nav2_msgs/srv/ManageLifecycleNodes
nav2_msgs/action/DummyBehavior            nav2_msgs/msg/BehaviorTreeLog             nav2_msgs/msg/VoxelGrid                   nav2_msgs/srv/SaveMap
nav2_msgs/action/FollowPath               nav2_msgs/msg/BehaviorTreeStatusChange    nav2_msgs/srv/ClearCostmapAroundRobot     
nav2_msgs/action/FollowWaypoints          nav2_msgs/msg/Costmap                     nav2_msgs/srv/ClearCostmapExceptRegion    
nav2_msgs/action/NavigateThroughPoses     nav2_msgs/msg/CostmapFilterInfo           nav2_msgs/srv/ClearEntireCostmap
```

SteveMacenski, 2022-11-30:
https://github.com/ros-planning/navigation2/releases/tag/1.1.3

I see the interface in the release source code and where it is installed in the cmake of the package. I'm not sure. Are you sure you have sourced the most recent `/opt/ros/...` directory in a clean terminal after you did the upgrade to 1.1.3? Maybe its a dirty environment thing? I'm looking at my install space you should be able to see references of the assisted teleop message in your `/opt/ros/humble/include/nav2_msgs/nav2_msgs/action` - do you not?

austin-InDro, 2022-12-01:
Strange, if I soure `/opt/ros...` from a clean terminal AssistantTeleop shows up. However, if I source from my workspace it disappears again. Reinstalling doesn't fix this problem.

I did check and AssitantTeleop is in `/opt/ros/humble/include/nav2_msgs/nav2_msgs/action`. 

As a test, I created another fresh workshace and `install/setup.bash` in both are identical.

It seems cleaning out my workspace and recompiling everything did the trick though. Seems it was some weird cache issue! Thanks Steve.

closed 2022-12-01 by austin-InDro