ros-navigation/navigation2 #3316: Information regarding the nav2 stack upgrading to behavior tree cpp 4.0 and later versions

Naraharirahul, 2022-12-06:
## Feature request
Compatibility issues of the ROS2 Humble's navigation stack with behavior tree  v4.

**Required Info:**

- Operating System:
  Ubuntu 22.04 
- ROS2 Version:
- ROS2 Humble
- Version or commit hash:
  - 
- DDS implementation:
  Cyclonedds

#### Feature description

My current development setup is using ROS2 Humble with behavior tree 3.8.x version. My behavior tree xml would consist of plugins developed by me and a few from the navigation2's plugins. 
I recently came across behavior tree 4.0 version which has scripting feature which I would like to use. I have noticed that some of the class names in the behavior tree 4.0 has been changed and also the package version itself is different.  This would mean that I cannot have a tree which uses behavor tree 4.0 + nav2 together.

I was hoping to know is there any migration planned ahead with this being an issue? I have raised an issue at Behavior tree CPP repo where I was told to ask here. https://github.com/BehaviorTree/BehaviorTree.CPP/issues/481

SteveMacenski, 2022-12-06:
I have not looked in great detail the changes between V3 and V4. Are there any substantive changes other than naming to the BT nodes and XML files? I'm wondering if its possible for the XML files to be backward compatible or if those also need to all change during the transition. I'd like as much as possible for all the BT XMLs to work on both. 

I have no plans at this very moment to move to v4 but that's mostly lack of resources and I figured a user would come along and want to use V4's features and be open to contributing the Nav2 port. I'd be happy to have a PR and updated dependencies to use V4 instead (@facontidavide are binaries for V4 released on Rolling?)

facontidavide, 2022-12-06:
I believe that it IS a good idea to migrate to BT.CPP 4 in ROS Iron... I also understand your point of view (it won't have a measurable benefit for us and we don't have resources).

**The only thing I can offer is to be the one that submits a PR with the code migration.**

But the dilemma is that we have two opposite groups of users:

- Those that have developed their own Nav2-BT plugin, will be obliged to migrate their code too (and maybe, they don't want to).
- Those that DO want to migrate to version 4.

https://www.behaviortree.dev/migration

As a reminder, version 3 and 4 are:

- Not binary compatible, that is the reason why we need to wait for ROS Iron.
- at the C++ level, they are mostly API compatible (given a simple renaming).
- XML can be auto-converted with a python script, to some extent.

SteveMacenski, 2022-12-07:
Is there V4 binaries for Rolling though?  You could (and I assume maybe did?) release V3/V4 in parallel under different release names. We cannot consider migrating Nav2 until there is a binary release for the dependencies available so that the build farm can turn over. 

That would also allow us to do a more careful transition if in Rolling both are available to do that migration leading to Iron. I support a V4 transition and I don't think that @facontidavide you have to be the one to do it (unless you want), it just wasn't on my TODO list since I knew someone else would come along and want it (and we have time before Iron).

facontidavide, 2022-12-07:
Ok, I guess we have a plan. I am currently working on version 4.1 and push that to Rolling at the end of December.
We will continue from there.

SteveMacenski, 2022-12-07:
Sounds good!

Naraharirahul, 2022-12-07:
Thanks @SteveMacenski & @facontidavide for this.

SteveMacenski, 2022-12-07:
It’s certainly possible that you could start a V 4 migration now and open a pull request. Then we can look at reviewing and merging it once the V4 binaries are available

Naraharirahul, 2022-12-08:
Okay. I will work on that and shall get back to you.

doisyg, 2022-12-15:
Jumping in the issue, @facontidavide do you think you could release it for Humble too ? `ros-humble-behaviortree-cpp-v4` (apt) / `behaviortree_cpp_v3` (rosdep key)

doisyg, 2023-03-12:
Friendly ping @facontidavide , do you think you could release `ros-humble-behaviortree-cpp-v4` and `ros-rolling-behaviortree-cpp-v4` ?

doisyg, 2023-05-16:
Hello @facontidavide, with Iron release approaching, I was thinking about jumping on the migration to BT v4. Do you still plan a ros release of v4 ?

ksitterle, 2023-10-30:
Hi all, any update on releasing `ros-humble-behaviortree-cpp-v4` and `ros-rolling-behaviortree-cpp-v4` ? I'm not seeing them

mitch-adams, 2023-10-30:
To clarify, are you planning on backporting this to Humble/Iron, or will Nav2 only be compatible with BTv4 starting with Jazzy?

SteveMacenski, 2023-10-30:
Jazzy and newer

SteveMacenski, 2024-01-24:
https://github.com/ros-planning/navigation2/issues/4059 supersedes and of the same content (migration to BT.CPPv4) Moving conversation there