ros-navigation/navigation2 #3318: last goal NavfnPlanner

dujeong, 2022-12-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - `0d86469f1c31708117e68f3f44ad5392347e9a5d`
- DDS implementation:
  - cyclonedds

#### Steps to reproduce issue
publish `2D Goal Pose` in rviz2.
planner plugin: **nav2_navfn_planner/NavfnPlanner**

#### Expected behavior
robot rotates to the goal angle which I set by `2D Goal Pose` in rviz2.

#### Actual behavior
last goal angle in path is different from the goal angle I set by rviz2.
The gif below is a tool I made to test the path planning, and the same result is obtained even if it is operated in the actual "2D Goal Pose". The planner I used is "**nav2_navfn_planner/NavfnPlanner**". 
("use_final_approach_orientation" parameter is also set to true)
[image]

SteveMacenski, 2022-12-07:
use_final_approach_orientation Should be false if you want to use the final orientation sit by your request. That parameter allows you to use the final orientation on approach to the goal instead. I believe maybe you’ve misunderstood its meaning.

dujeong, 2022-12-08:
I misunderstood the parameter meaning,, after changing "use_final_approach_orientation" to false, it works well. Thank you!

closed 2022-12-08 by dujeong

SteveMacenski, 2022-12-08:
No worries, happy it was something easy :-)