ros-navigation/navigation2 #3322: Navigation not working after sending global_costmap service call to disable costmap.

kamakshijain, 2022-12-12:
## Bug report
Problem: nav2_param.yaml has costmap plugins enabled, on sending goal, turtlebot will start navigation. Let's say we send a service call `ros2 service call /global_costmap/global_costmap/set_parameters rcl_interfaces/srv/SetParameters "{parameters: [{name: "static_layer.enabled", value: {type: 1, bool_value: False}}, {name: "obstacle_layer.enabled", value: {type: 1, bool_value: False}}, {name: "inflation_layer.enabled", value: {type: 1, bool_value: False}}]}"` to disable costmap, we then send goal pose again. This time controller does not generate a path to the goal and it gets stuck. 

While waiting if I sent `ros2 service call /global_costmap/global_costmap/set_parameters rcl_interfaces/srv/SetParameters "{parameters: [{name: "static_layer.enabled", value: {type: 1, bool_value: True}}, {name: "obstacle_layer.enabled", value: {type: 1, bool_value: True}}, {name: "inflation_layer.enabled", value: {type: 1, bool_value: True}}]}"` to enable the costmap back, path to the same (above goal) is suddenly generated and turtlebot starts navigating.

**Required Info:**

- Operating System:
  - 
  -  Ubuntu 22.04
- ROS2 Version:
  - 
  - Humble
- Version or commit hash:
  - 1.1.2
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

```
Step 1: initialize setup
Step 2: send a small goal pose from rviz
Step 3: `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
Step 3: send `ros2 service call /global_costmap/global_costmap/set_parameters rcl_interfaces/srv/SetParameters "{parameters: [{name: "static_layer.enabled", value: {type: 1, bool_value: False}}, {name: "obstacle_layer.enabled", value: {type: 1, bool_value: False}}, {name: "inflation_layer.enabled", value: {type: 1, bool_value: False}}]}"`
Step 4: send again goal pose on rviz
```

#### Expected behavior
1 if the path is clear, turtlebot should generate a path and start navigating
#### Actual behavior
1 there is no path generated, it gets stuck.
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2022-12-12:
That seems like largely expected behavior to me. If you disable all your plugins so that there is nothing in your costmap, then there's nothing to size it or be able to plan within it. If you have no static map, how can you set the size of the costmap to begin with in order to plan through a space? It has nothing to size the costmap if your settings don't have a size nor a layer which impacts the size like the static layer, for example. 

But _how_ it fails for you is what I think is problematic. The planning request should fail, which should result in the behavior tree ultimately failing after all the recoveries fail. It should not do what you see where it succeeds.

SteveMacenski, 2022-12-12:
What I suspect is happening is that https://github.com/ros-planning/navigation2/blob/humble/nav2_planner/src/planner_server.cpp#L460-L462 is triggered which `returns` without setting the action as failed. We resolve this in Rolling with the piping PRs which return error codes on contextual failures returned to the behavior tree server. Or could be that the path isn't empty but a single-point value. Some logging here would be interesting and if the `main` branch resolves the failure problem.

kamakshijain, 2022-12-13:
Other tests that I ran
Test 2: static_layer remains enabled i.e static_layer.enabled: true and using the srv I set obstacle and inflation layer as false. This also results in path not getting generated after the srv.

Test 3: static_layer and obstacle_layer remians enabled, I only change the inflation layer again, no path generated same behavior.

Test 4: in my yaml file, i.e nav2_params.yaml (given below), all 3 layers are set to false. I double check using get params command, layers are set to false. I give a goal, path is generated. 

This is opposite to the behavior we observed when we are setting the parameters as false using srv call. I am double checking parameter value after service call to check enable switching between true and false.
````
global_costmap:
  global_costmap:
    ros__parameters:
      footprint_padding: 0.03
      update_frequency: 1.0
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_footprint
      use_sim_time: True
      robot_radius: 0.22 # radius set and used, so no footprint points
      resolution: 0.05
      plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: False
        observation_sources: scan
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        combination_method: 1
        scan:
          topic: /scan_multi
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          inf_is_valid: false
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
        enabled: false
        subscribe_to_updates: true
        transform_tolerance: 0.1
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        enabled: false
        inflation_radius: 0.55
        cost_scaling_factor: 1.0
        inflate_unknown: false
        inflate_around_unknown: true
      always_send_full_costmap: True
```
`

SteveMacenski, 2022-12-13:
Please do a bit more debugging into the code - I can't answer for you exactly this behavior without some more information.

SteveMacenski, 2023-01-12:
https://github.com/ros-planning/navigation2/pull/3356

I believe this resolves it

closed 2023-01-12 by SteveMacenski