ros-navigation/navigation2 #3325: Mixing Wall time with simulation time in Costmap2DROS::mapUpdateLoop

AzaelCicero, 2022-12-14:
## Bug report

**Required Info:**

- Operating System: 
  - All
- ROS2 Version: 
  - All
- Version or commit hash:
  - All 
- DDS implementation:
  - All

#### Steps to reproduce issue

Just use costmap_2d node with simulation time, which has very different rate than wall time. You will observe that HZ of map update and map publish is not reaching the expected value and are far off the target. 

#### Expected behavior
Configured rates of update and publish should be achieved in respect to simulation time.

#### Actual behavior
Map update is reaching expected rate in respect to wall time. Publish rate is bounded to the update rate, as it can only occur during the update, yet it is triggered in respect to simulation time. Depending on the mismatch of the rate between simulation and wall time you will observe different HZ of update and publish. 

#### Additional information
Code should be written in respect to one time source, in this case I think that simulation time is better suited. 

https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/costmap_2d_ros.cpp#L461-L498

SteveMacenski, 2022-12-14:
@AzaelCicero A PR is always appreciated. 

> For local costmap, updateMap() execution time took ~1.0 sec, for global costmap - ~1.5 sec.

If the issue is that the actual updating of the costmap is too long, that's not a problem with clocks / timers. That's trying to do too much "stuff" to be faster than real-time algorithmically. 

> sleep_time = 1/frequency - (cycle_end - cycle_start)

The ROS rate should be doing that for us to keep a regular frequency - unless there's something wrong with that.

AlexeyMerzlyakov, 2022-12-15:
> The ROS rate should be doing that for us to keep a regular frequency - unless there's something wrong with that.

Indeed the `WallRate` already does the same logic I've supposed in previous message. From [rate.hpp](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/rate.hpp):
```
  virtual bool
  sleep()
  {
    // Time coming into sleep
    auto now = Clock::now();
    // Time of next interval
    auto next_interval = last_interval_ + period_;
    ...
    // Calculate the time to sleep
    auto time_to_sleep = next_interval - now;
    ...
    // Sleep (will get interrupted by ctrl-c, may not sleep full time)
    rclcpp::sleep_for(time_to_sleep);
    return true;
  }
```
So, that is my oversight.

@AzaelCicero, could I ask you to describe the test steps to reproduce the problem in more details?

AlexeyMerzlyakov, 2023-02-09:
Just for noting:
On the days I've got some strange effect for both proposed in this ticket approaches (switching to `rclcpp::timer`; or making our own Nav2's `ROSRate` instead of `rclcpp::WallRate`). The effect did not appear if we are slowing down the simulation in Gazebo: everything is working as expected. But while we are increasing Gazebo `Physics -> real time update rate` to 1500Hz or higher, both ROS and real measured times between two consequent costmap publications became to be unstable and shown incorrect jumping values.
After some investigation, I've found that the reason of this issue is that Gazebo simulation time has some quantum of ROS-time: minimal value of ROS time to be increased on the simulation step. This step in default Gazebo configuration is `100ms`. So high value is giving us a instabilities on high simulation frequencies.
This ROS-time minimal increase value is placed in the `gazebo_ros` package and utilized as input frequency for `Throttler(const double _hz)` constructor. Default value passed to this constructor is being handled by `publish_rate` ROS-parameter of Gazebo. It is being set by-defalt to `10Hz` (`100ms`), which causes the problem on higher rates.

Adding `publish_rate:=100.0` (`10ms` ROS-time quantum) parameter while calling `gzserver` drastically improves the situation on high (1500+) gazebo `real time update rates`:
```
diff --git a/nav2_bringup/launch/tb3_simulation_launch.py b/nav2_bringup/launch/tb3_simulation_launch.py
index b4dfd3bd..7fdf35e5 100644
--- a/nav2_bringup/launch/tb3_simulation_launch.py
+++ b/nav2_bringup/launch/tb3_simulation_launch.py
@@ -160,7 +160,8 @@ def generate_launch_description():
     start_gazebo_server_cmd = ExecuteProcess(
         condition=IfCondition(use_simulator),
         cmd=['gzserver', '-s', 'libgazebo_ros_init.so',
-             '-s', 'libgazebo_ros_factory.so', world],
+             '-s', 'libgazebo_ros_factory.so', world,
+             '--ros-args', '--param', 'publish_rate:=100.0'],
         cwd=[launch_dir], output='screen')
 
     start_gazebo_client_cmd = ExecuteProcess(
```

AlexeyMerzlyakov, 2023-02-09:
The above observations makes me think about the accuracies of both solutions from this ticket (switching to `rclcpp::timer` vs. having our own `nav2_utils::ROSTimer`). I've made some small investigation by measuring the time intervals (in ROS and real time) passed between two subsequent `mapUpdateLoop()` cycles. The measurements are summarized by average time error (from expected time) and standard deviation of this time. This was made for ROS time quantum  (`publish_rate` Gazebo ROS-parameter) to be equal to `10ms` and `2ms`. For each ROS time quantum value, the simulation speed was varied by changing `real_time_update_rate` (https://classic.gazebosim.org/tutorials?tut=physics_params) in `200Hz..10KHz` range. The summarized results are presented in the table below:
[image]

As follows from the results, both approaches (`rclcpp::timer` and `nav2_utils::ROSTimer`) are having comparable errors of real measured times between `mapUpdateLoop()` cycles on low and high `real_time_update_rate` frequencies (and thus on different CPU loads); when `nav2_utils::ROSTimer` is slightly overcomes `rclcpp::timer` approach.

However, when comparing the time errors of measured ROS-time between `mapUpdateLoop()` cycles, which we are initially targeting to hit in this issue, the undisputed leading method in this case is `rclcpp::timer` approach. It shows ~33x-130x lower time errors and notably better deviations (better results stability) comparing to `nav2_utils::ROSTimer` approach. Here is the case of most CPU-loading case (`ROS quantum time = 2ms`; Gazebo `real time update rate = 1KHz` which is 10x faster than real world speed) for both approaches, depicted below:
[image]

Which makes `rclcpp::timer` approach the best for selection. I'll update #3327 with all review items from there (it seems the PR was abandoned for a long time) and prepare the final patch shortly.

SteveMacenski, 2023-08-31:
Yay! rclcpp Rate merged in!

SteveMacenski, 2023-11-27:
@AlexeyMerzlyakov what's the story with this ticket now? With the new Rate updates in rclcpp, can we use that now in the costmap thread and close out this ticket?

AlexeyMerzlyakov, 2023-11-30:
@SteveMacenski, the scope of related works are reported in https://github.com/ros-planning/navigation2/issues/3303#issuecomment-1833302021. This issue is covered by #3303, as there are already discussing all the places need to be ROS-time respective, incl. new `rclcpp::Rate` API

SteveMacenski, 2023-12-01:
From @AlexeyMerzlyakov translating to this ticket:

More general, by grepping where we are using `std::chrono`'s now() and `rclcpp::WallRate` (the same problem caused by rclcpp WallRate, described in #3325), we have the following places where ROS timings are being ignored (tests were excluded):

| Where | Type | Comment |
|-------|------|---------|
| nav2_behavior_tree/plugins/decorator/rate_controller.cpp | std::chrono | |
| nav2_behavior_tree/src/behavior_tree_engine.cpp | rclcpp::WallRate | |
| nav2_behaviors/include/nav2_behaviors/timed_behavior.hpp | rclcpp::WallRate | |
| nav2_behaviors/plugins/wait.cpp | std::chrono |Discussed by #3303 |
| nav2_controller/src/controller_server.cpp | rclcpp::WallRate | |
| nav2_costmap_2d/src/costmap_2d_ros.cpp | rclcpp::WallRate | Covered by #3325 |
| nav2_smac_planner/src/a_star.cpp | std::chrono | |
| nav2_theta_star_planner/src/theta_star_planner.cpp | std::chrono | |	
| nav2_theta_star_planner/src/theta_star_planner.cpp | std::chrono | |	
| nav2_waypoint_follower/src/waypoint_follower.cpp | rclcpp::WallRate | |

SteveMacenski, 2023-12-04:
@AlexeyMerzlyakov is https://github.com/ros-planning/navigation2/pull/3404 good to go with the rclcpp merge?

AlexeyMerzlyakov, 2023-12-06:
@SteveMacenski, please refer to the https://github.com/ros-planning/navigation2/pull/3404#issuecomment-1842776234 which explains the intention

SteveMacenski, 2023-12-06:
OK, so I think we're all squared away on this issue. I have decided that things like Costmap's loop rate, waypoint follower's loop rate, and the controller's loop rate should be tied to the system clock. Internal timings of systems are using ROS time when in simulation, but module-level operations should remain as tied to wall time.

This is because systems like the costmap and controller server are **not** systems that can be simulated at 2x-3x-10x speeds due to their computational complexity. The added variation due to the simulation clock is not worth the advantages and results in a poorer quality and explainability of navigation behavior at those higher speeds since you'll see problems not seen at 1x speeds. I'm not so dead set on this that I'm not open to considering this change in the future, however today I think it is not a worthwhile trade off. It also created a number of serious issues breaking testing behavior which given the above statements, after about half a day of debugging I ultimately decided wasn't worth continuing to address. 

The behaviors (timed behavior, spin, drive on heading, backup, wait, etc), planners planning time, smoothers smoothing time, and rate controller have all been updated to use simulation time from the clock as internal behavioral matters. Costmap's & waypoint follower's loops; controller will remain tied to walltime since those are system level behaviors where changes in timing are important. 

You can see that in the PR https://github.com/ros-planning/navigation2/pull/4000

closed 2023-12-06 by SteveMacenski