ros-navigation/navigation2 #3335: Pluginize nav2_bt_navigator

redvinaa, 2022-12-21:
## Feature request

#### Feature description
I'm planning to extend the bt_navigator package to be plugin based, after the [discussion on Discourse](https://discourse.ros.org/t/make-bt-navigator-extendible/28570). The navigators would be plugins, loaded into the bt_navigator node at runtime. This would make it easier to write new navigators in the case of custom behavior trees needing parameters different than the NavigateToPose or NavigateThroughPoses actions.

#### Implementation considerations
As the other plugin-based packages, the implementation would use the pluginlib library. As for the base class, it's a question to be answered. The navigator base class currently used for the navigators is actually a class template (the template type is the type of the action server that the navigator implements), which cannot be used as a base class as is.

A possible solution would be to separate the action server from the behavior tree engine, and only include the latter in the plugins' base class, although I understand the best would be to include both. @SteveMacenski do you have a better idea how to implement this?

mradams-bastian, 2022-12-21:
There might be some overlap here with [this](https://github.com/ros-planning/navigation2/pull/3068) PR

SteveMacenski, 2022-12-21:
This is the [base navigator class](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/include/nav2_bt_navigator/navigator.hpp) 

The biggest question I would propose from looking at this is if every navigator type would be necessarily going to use our BT Action Server (https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/include/nav2_bt_navigator/navigator.hpp#L159-L167). Everything else here looks sufficiently general for use as a plugin base class. The `bt_action_server_` is the root of the BT behavior, but conceptually someone could modify that (but I would hope they'd contribute it back). It also is the main operation that we play with in the `on_*` functions to remove the boilerplate of calling the object's methods so that the implemented plugin can focus only on what it needs to configure/activate/deactivate/cleanup. 

I think this is good as-is, but that's a point for discussion.

redvinaa, 2022-12-23:
> There might be some overlap here with https://github.com/ros-planning/navigation2/pull/3068 PR

That PR solved the above problem by creating a virtually empty `nav2_core::Navigator` class as a base class and that doesn't enforce the plugins to be BT related at all.

> The biggest question I would propose from looking at this is if every navigator type would be necessarily going to use our BT Action Server

I think alleviating that assumption sounds reasonable. For example the `/goal_pose` topic coming from RViz is already a workaround. So the base class would include the BT nature but not the Action nature.

SteveMacenski, 2023-01-03:
Starting on this this week

SteveMacenski, 2023-01-03:
https://github.com/ros-planning/navigation2/pull/3345 implements, will merge when I get some documentation updates completed (e.g. updates to describe feature + a tutorial on it since this is an important new plugin type)

closed 2023-01-03 by SteveMacenski

AntoBrandi, 2024-01-23:
Are there any plans of having this for humble?

SteveMacenski, 2024-01-23:
It is non backportable. Its in iron and newer