ros-navigation/navigation2 #3338: Navfn makes weired last pose of the path

WonsokYoo, 2022-12-23:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04 (Docker container)
- ROS2 Version:
  Humble (but also Foxy)

#### Steps to reproduce issue
  Run any Gazebo environment and runs navigation2 with Navfn planner.

#### Expected behavior
  Make smooth and continuous path
#### Actual behavior
  Last pose of the path is weired. The last pose is out of path's smooth line.
[image]
[image]

#### Additional information
   If I search on google, I could see same phenomenon (Of course I cannot know exact configuration of them)
      ex> https://index.ros.org/p/nav2_rotation_shim_controller/

SteveMacenski, 2022-12-23:
The planner does not generate orientations. So the last points on the path has its orientation set as the goal orientation requested as a holonomic search based planner. 

The orientation of the last point is used by local trajectory planners in order to achieve the requested orientation of the navigation goal.

WonsokYoo, 2022-12-23:
@SteveMacenski Thanks for your reply but could you explain more precisely? My question is actually regarding x, y position not orientation. In my opinion, the red shortest path should be generated like below:
[image]

AlexeyMerzlyakov, 2022-12-23:
The effect has two reasons:

R1: The last piece with goal position and orientation added by `smoothApproachToGoal()` routine. It clips the path (if necessary) and inserts the goal pose as the last part of produced by NavFn path. In some cases, it might generate the jumps at the end as depicted below:
[image]
The jumps in this case will be less than one grid cell. However, for the precision reasons, it could not be disabled in NavFn.

R2: NavFn planner due to its A* algorithm nature, which calculates the potential through the cells' grid, might produce far from true optional path, as follows at the example below (`smoothApproachToGoal()` routine was disabled):
[image]
More about this effect you can find in https://news.movel.ai/theta-star. As well as there, I would like to suggest to use the "ThetaStar" planner which was designed and using to avoid such effects in the end or during the path.

WonsokYoo, 2022-12-28:
@AlexeyMerzlyakov Thanks for your kind and precise explanation. I could understand more about NavFn and Theta*. Now I have one more question about NavFn. You have mentioned above "NavFn planner due to its A* algorithm nature, which calculates the potential through the cells' grid", and I have the same theoretical background about basic A* also but for navigation2 implementation as you can see below (both images with or without smoothApproachToGoal) there I can see under resolution results which means less than single grid for the poses on paths
[image]
Based on above result, I thought A* of navfn have some post processing so in my opinion the R2 case should be like below (red line path)
[image]

Do you have any ideas about this?

AlexeyMerzlyakov, 2022-12-29:
@WonsokYoo, thank you for good questions on NavFn internals

> there I can see under resolution results which means less than single grid for the poses on paths

As I've mentioned, the potential field grid is being quantized by map cells. However, the path shifted by potential grid in the best direction is being calculated as [linear interpolation](https://github.com/ros-planning/navigation2/blob/main/nav2_navfn_planner/src/navfn.cpp#L874-L880) of potential by pose shift _between_ cells. Pose shift is [being normalized](https://github.com/ros-planning/navigation2/blob/main/nav2_navfn_planner/src/navfn.cpp#L899-L901) by `pathStep` value, which is equal to `0.5`. This is acually what you've observed on upper picture. The last pose shift (near the goal) in algorithms is actually the first iteration in `NavFn::calcPath()`, since the propagation is being spread from the last down to the first pose. In this case, at the very first iteration `potarr` will have its `POT_HIGH` value and we fall into the first [if-condition](https://github.com/ros-planning/navigation2/blob/main/nav2_navfn_planner/src/navfn.cpp#L819) in the algorithm where the pose will be shifted on one cell `(±1.0, ±1.0)` value, which is also observable on the upper picture.

> Based on above result, I thought A* of navfn has some post processing so in my opinion the R2 case should be like below

For this we have post-porcessing path smoother module (https://navigation.ros.org/tutorials/docs/adding_smoother.html) which will allow to make the path as you've shared on bottom picture. I'd suggest to consider Simple or Savitzky-Golay smoother to try to.

SteveMacenski, 2023-01-02:
Hi,

Sorry - Holidays and all that. 

> Thanks for your reply but could you explain more precisely? My question is actually regarding x, y position not orientation. In my opinion, the red shortest path should be generated like below:

Ah, this is due to the gradient potential field descent happening. NavFn does grid search to generate a potential field to the goal, then uses gradient descent to actually find the route, which given the discretized nature of a grid-map means that odd little artifacts like that can occur. Its not usually a problem, functionally speaking, it might just not look very good to the human eye but doesn't affect robot performance. 

But also ditto what @AlexeyMerzlyakov says about `smoothApproachToGoal`, who's aim is to resolve one of the more annoying artifacts on approach to the goal pose with the discrete potential field gradient descent. It can make small gaps in exchange for not jerking around. 

If you're looking for a "pure" A*, that would be Smac Planner 2D. But honestly NavFn is very fast and has been used in production by alot of companies for over a decade.  Its still a good choice

closed 2023-01-02 by SteveMacenski